jbruce12000/kiln-controller #3: SPI support for sensor

Steve1445, 2019-03-03:
After not being able to get a temperature out of the sensor, I have switched over to SPI, through this I can confirm (simpletest.py) temp is being reported.

However, after changing the gpio defaults in the config.py to the new pins and starting to launch the interface I get two error messages (see below).

pi@raspberrypi:~/kiln-controller $ virtualenv venv
Running virtualenv with interpreter /usr/bin/python2
New python executable in /home/pi/kiln-controller/venv/bin/python2
Not overwriting existing python script /home/pi/kiln-controller/venv/bin/python (you must use /home/pi/kiln-controller/venv/bin/python2)
Installing setuptools, pkg_resources, pip, wheel...done.
pi@raspberrypi:~/kiln-controller $ source venv/bin/activate
(venv) pi@raspberrypi:~/kiln-controller $ source venv/bin/activate; ./kiln-controller.py
2019-03-03 17:37:59,298 INFO kiln-controller: Starting kiln controller
2019-03-03 17:37:59,571 INFO oven: import MAX31855SPI

Traceback (most recent call last):
  File "./kiln-controller.py", line 33, in <module>
    from oven import Oven, Profile
  File "/home/pi/kiln-controller/lib/oven.py", line 24, in <module>
    if config.gpio_air in spi_reserved_gpio:
AttributeError: 'module' object has no attribute 'gpio_air'

(venv) pi@raspberrypi:~/kiln-controller $ 

Any insight would be appreciated.

jbruce12000, 2019-03-03:
Hi Steve,

Those bugs are fixed.  It was due to some leftover code from the original codebase.  It had control pins for a cooling fan and a door switch.  removed.  Do a git pull and you should be good.

I've never used the SPI interface, so you're likely to run into more problems.  I'll help as much as I can, but without actually using that interface, I may not be able to help much more.

jbruce12000, 2019-03-03:
Looks like you already figured this out, but I had to pip install python-dev to get the adafruit spi module to install.  I added this to the README to make it easier for the next spi user.

closed 2019-03-03 by jbruce12000

Steve1445, 2019-03-04:
Hi Jason,

thanks for the quick response. I have done a fresh clone and updated the
Adafruit MAX31855 library. I'd rather not use the SPI interface but just
can't seem to see the sensor data. Been back and forth and even replaced
the 31855 board, but nothing.  Moved over to SPI to prove to myself temps
are being measure, I can view it in real time so that helps.  But even
after all this, still getting the same errors.

Steve

Steve1445, 2019-03-04:
Hi Jason, I realized my mistake this morning when I told you last night that your SPI code fix had no effect. I corrected and tried running, the previous errors are now gone and this one below shows itself.

  File "./kiln-controller.py", line 8, in <module>
    import bottle
ImportError: No module named bottle

Steve1445, 2019-03-04:
So worked my way through this error and a couple of others where it was not seeing modules being installed.   I think in my environment while it looks to me this was complied under Python 2.x I am running Python 3.x, consequently needed to adjust any Python command, i.e. Python3. At this point nothing for you to address. This is just an FYI.

jbruce12000, 2019-03-04:
If you get python 3.x to work, post a PR and I'll check it out and get it merged if everything works.

jbruce12000, 2019-03-04:
Do you have your MAX31855 wired up like my schematic shows? Have you made sure the GPIO pins on your pi work as expected? https://github.com/jbruce12000/kiln-controller/blob/master/docs/troubleshooting.md

jbruce12000, 2019-03-04:
There are two ways that I know of to get python3 virtualenv to run.

1. sudo apt-get install python3-virtualenv
2. python3-virtualenv ./venv

or 

1. python3 -m virtualenv ./venv

there is no symlink created to python2 virtualenv in either case.

Steve1445, 2019-03-04:
Commenting on your last three posts.

This Python and Pi thing is altogether new to me, so I doubt I would be the one to get it to run on 3.x but you never know. I don't know what a PR is. 

I wired the Pi and the MAX31855 exactly like your schematic, being new to this, I was confused by the whole BitBang, GPIO and physical interface thing, but I muddled through and think I got it right.  However, when I started asking questions of Adafruit, they led me down the SPI configuration, duh who was I to know the was a difference.  Although I could swap out now in a heartbeat.  The one thing I do like about the SPI is being able to see the thermocouple working without having the kiln interface up and running. This turned out to be important and I could not shake "vcc to ground" and "vin to ground" errors when troubleshooting with Adafruit.  They thought my twisted pair thermocouple wire and thermocouple was acting as an antenna for stray interference. Which is was not. Turned out the grounding was snakey. 

I will look at your last post on the Python3.

I am at the point now where upon launch of the app, its telling me it cannot find the adafruit spi module. And as far as I can tell when I do a pip install there isn't one out there.  Since I have SPI enabled in the Pi, my next thing is going to see if I can just comment that out as when the web app launches (at this point it's all messed up) but it does show the temp, which is something I never had before.

jbruce12000, 2019-03-05:
No worries about python3.  You should not be using it with this software.  Check your version with...

python --version

A "PR" is a pull request. It is used on github to suggest changes a developer should make to his code.  You would submit it.  I would approve and merge it. You are new to python and git, so you should probably not try to do this.

Here is what my output looks like when I install Adafruit-MAX31855...

pip install Adafruit-MAX31855
Collecting Adafruit-MAX31855
  Using cached https://files.pythonhosted.org/packages/8b/e9/beacb8ca965fe758e131da06d7bdadcc9cdaa167228f1ac1e7d8d64c1bfd/Adafruit_MAX31855-1.6.1-py2-none-any.whl
Collecting Adafruit-GPIO>=0.6.5 (from Adafruit-MAX31855)
Collecting spidev (from Adafruit-GPIO>=0.6.5->Adafruit-MAX31855)
Collecting adafruit-pureio (from Adafruit-GPIO>=0.6.5->Adafruit-MAX31855)
Installing collected packages: spidev, adafruit-pureio, Adafruit-GPIO, Adafruit-MAX31855
Successfully installed Adafruit-GPIO-1.0.3 Adafruit-MAX31855-1.6.1 adafruit-pureio-0.2.3 spidev-3.4

When I run **pip freeze** here are the modules I get that have been installed...

Adafruit-GPIO==1.0.3
Adafruit-MAX31855==1.6.1
Adafruit-PureIO==0.2.3
bottle==0.12.16
gevent==1.4.0
gevent-websocket==0.10.1
greenlet==0.4.15
pkg-resources==0.0.0
spidev==3.4

jbruce12000, 2019-03-05:
huh.  looks like support for python3 is pretty trivial.  I made changes to get the main interface up-and-running in about 15 minutes. I can push this if you like. lmk.

Steve1445, 2019-03-05:
Hi Jason,

After messing around with SPI and not being able to find the adafruit.spi
module I decided to start over fresh. I had used your troubleshooting doc
when I was initially putting this thing together, so I was pretty confident
with all the ways the pins should fire and their purpose. But I got to
thinking how when I was setting this up I was plagued by the constant error
messages coming from the MAX31855, no matter what I checked, or double
checked I could not shake them.  Adafruit eventually sent me a new board as
it was the only thing left to look at that point. Not having an O-scope to
look at noise on the T-couple line, I instead injected 1.4 vdc into the
chip to simulate temp. Still the errors persisted and removed the T-couple
and noise as a suspect.

Sometimes you just way overthink things.  I noticed in one of your posts
you mentioned checking voltage to ground, as I looked at the schematic, I
realized it shows one leg going to ground (the kiln power supply or
essentially earth) it dawned on me while I observed ground with the
MAX31855 and the Pi, I did not ground to earth the whole. Once I did that,
my errors went away. Duh! Furthermore, I realized that this might also have
something to do with not pulling a temp off the T-couple. So with that in
mind, I wiped the SD and started fresh, and went back to your original
instructions and not on my wayward SPI path.

Long story short, seem to be in business.  Reading off the t-couple, ran a
small test, but my temp went rocketing past the first breakpoint (180) and
I stopped it at 278. So I have some more stuff to work through, but at
least I'm at a good point to quit for today.

As far as the Python3, seems like it should be posted regardless since
support for Python2 runs out this December.

One other thing, log file somewhere that records the firing? Nice to be
able to catch the history for later review.

I have a bunch of leather ware saved up just waiting for bisque firing,
really psyched about the possibility of  maybe I am getting to the point
where I can run a profile, stuff will come out so much better than timing
my steps and hoping for the best at what is going on in the kiln.

Thanks for your help

Steve

jbruce12000, 2019-03-06:
Congratulations!  You're the first to get things working.  Nice job.

Logs from the app on the pi go to /var/log/daemon.log and look like this...

Jan 21 06:25:40 raspberrypi python[286]: 2019-01-21 06:25:40,390 INFO oven: temp
=1092.4, target=1093.2, pid=1.000, heat_on=2.00, heat_off=0.00, run_time=15993, 
total_time=48780, time_left=32786

temp = temperature read by thermocouple
target = target temperature
pid = pid value for that 2s
heat_on = number of seconds the elements were on
heat_off = number of seconds the elements were off
run_time = seconds since start
total_time = total seconds for run
time_left = seconds left till the end

It's trivial to convert the data to csv... 

grep "INFO oven" daemon.log|sed 's/temp=//'|sed 's/target=//'|sed 's/heat_on=//'|sed 's/heat_off=//'|sed 's/run_time=//'|sed 's/total_time=//'|sed 's/time_left=//'|sed 's/pid=//'|sed 's/.*: //' >out2.csv

Here is a slow glaze firing imported into google docs. Make sure to check out the **report** tab.

https://docs.google.com/spreadsheets/d/1Lcp88P1cNNzYWgKDfnd5UaPVqLuBdAT3lkpfcZMKEBM/edit#gid=2116406322

Next you need to tune the PID params to your kiln. You can try my values, they might work, might not. BTW, I have never had that much overshoot.  You can take a look at the spreadsheet I posted above, one sample of temp every 2s.  See row 365 where the max overshoot is 12F.  You can see the average error (difference between set temp and actual temp) is 1.78F over the entire 13 hour run.  Once you get tuned, you can count on the control being nearly perfect.

My guess is that your elements are always on.  Not sure why.  You'll need to investigate.

Steve1445, 2019-03-06:
I need a note of clarification; the schematic shows the output to the base
of the transistor as GPIO23 which (in my Pi) is pin 13 (BCM), however it
looks like you have listed the physical pin numbers in some cases (1, 2, 9
) while using BCM in others (17, 22, 27) where there is a listing of GPIO
17, 22, 27 on the schematic but don't exist in the Pi.

My only confusion is for the GPIO23, on physical pin 33 (which is BCM pin
13) or should it be BCM pin 23 GPIO4. Not knowing about the differences
between RPi, the schematic may be based upon what you built on and not what
I am building on?  The main point would be that GPIO4 does go high and low
as directed by the Pi.

Steve1445, 2019-03-06:
and checking that log file, I don't find any mention of Temp, target, etc.
basically any of the parameters you mentioned

jbruce12000, 2019-03-06:
Yes, pinout on Rpi is confusing and I did not help matters at all with my schematic.

3.3V, 5V, and ground all just list the physical pin because they don't have BCM pin numbers

GPIO inputs use BCM pins 17,27 and 22 which correspond to physical pins 11,13, and 15 on my raspberry pi 3. These physical pins are mislabled on my schematic. These correspond to GPIO 0, 2, 3 which are all mislabeled on my schematic.

GPIO output uses BCM pin 23 which corresponds to physical pin 16 on my pi.  This physical pin is mislabled on my schematic. BCM 23 corresponds to GPIO4 which is mislabled on my schematic.

Wow...Broken.  I will put in an issue and try to fix my schematic this week, but I'm pretty busy.

When in doubt, map GPIO 0,2,3,4 from the **Name** column to the **BCM** column to the Physical Pin column in the output of **gpio readall**.  The BCM pin is the pin to use in config.py.

As to logging... Is this running on the PI? Did you run **start-on-boot** so everything starts automatically? or are you logging in and running kiln-controller.py manually?

if you ran **start-on-boot**, systemd should be redirecting stdout and stderr to /var/log/daemon.log
if you're running it manually, logs go to stdout.

Steve1445, 2019-03-06:
don't worry about the schematic, it's all good. I puzzled it out and am
running through the PID tuning now.

I start the app manually, I tried the start-on-boot and never saw it run
any apparent result, i.e. connecting to the kiln controller.

jbruce12000, 2019-03-06:
Hmm.  **start-on-boot** is a simple shell script that copies a file from the checked out repo to /etc/systemd/system and then runs another command.  I think the only way this could break is if you're not logging in as the user **pi**. maybe you created an account and login using that? Anyway, checkout start-on-boot because you can probably just change the path the first command it runs and it should work.

It's worth it because you can just power on your pi and connect to the kiln web interface 15s later.

Steve1445, 2019-03-06:
Is it possible systemd only runs on jessie and I am on stretch?

Steve1445, 2019-03-07:
try this on for size
[image: kiln controller.jpg]

jbruce12000, 2019-03-07:
systemd is the default service manager for both Jessie and Stretch...
https://wiki.debian.org/systemd

The schematic image you uploaded did not show up in comments for me, but I made the changes a few minutes ago and uploaded a new image.  Thank you for taking the time to create that if you did so.

jbruce12000, 2019-03-07:
To tell if you have systemd support, type **systemctl** and it should show you a list of services that are managed.

Here is the version of Raspbian I'm running from NOOBs...
$ cat /etc/issue
Raspbian GNU/Linux 9

Steve1445, 2019-03-07:
so running systemctl after boot shows:

● kiln-controller.service      loaded failed failed    kiln-controller
however, if I manually run the line

systemctl start kiln-controller.service   I am prompted to enter the
default password for user Pi.  Which is the user I am logged in as and has
not been changed.  Of course I can sudo it and then it will not prompt but
start.

kiln-controller.service       loaded active running   kiln-controller

Steve1445, 2019-03-08:
resolved auto start of app by adding a line to the end of /etc/profile
pointing to a script that starts the service.