jbruce12000/kiln-controller #5: Thermocouple short to ground & short to VCC

virtualbuck, 2019-06-12:
Hello.  I first downloaded and installed this software in March.  Everything worked great (thank you!) for several runs as I tested out a newly built wax burnout kiln.  After initial testing, I had to put the project aside for a while.  When I came back to it, I ran the software and would get an occasional Thermocouple Short to Ground error or a Short to VCC error in the console.  The next time I ran it...I got the same errors.  I was concerned that I got some debris into breadboard or that some wiring had come loose so I rewired the circuit.  Did not fix.  I built and installed a new MAX31855.  Did not fix.  I got a new RPI and soldered the circuit.  I moved the flash to the new RPI and booted it.  Same errors.  Now everything is new/different except the flash and the errors are still there.  Later today I will start with a new flash and reinstall all the software.  I suspect and hope that this will resolve the issue.  

Have you had anything like this happen before?  Is there any easy fix?

Also...when I changed the RPI, I went from 3B+ to zeroW.  I have seen a few oddities that may be manifestations of the same problem that results in the Thermocouple Short errors.  There is a pull down list that shows the saved profiles.  It does not get populated with the existing saved profiles until you add a new profile and save it...then all the previous profiles show up in the list.  Also, the controller was set to degrees F in March but now the units are degrees C but the value is correct as if the units are in F.   Very strange...seemingly non-deterministic behavior.

FYI...I have been an Arduino fan for many years and I am accustomed to its robustness.  Maybe RPI is more easily corrupted due to incorrect shutdown?  I did have a test running when the electrician shut off a breaker to work on a group of circuits and my RPI Kiln Controller was running on one of them.  I thought I had a successful run after that but now I am not so certain.  

Thanks for the work you are doing on this project.   Let me know if you would like me to look at something on the flash.  I will not reformat that flash....I will start with a new one...so it will be available to you for a few weeks at least.  

[image]

virtualbuck, 2019-06-13:
I set aside the previous SD card and started with a new card.  Using the RPI3B+, I reinstalled Rasbian and followed all the directions to install the kiln-controller.  I launched the kiln-controller server and connected to the UI via browser.  As soon as I pinched the thermocouple tip between my fingers to warm it up, I got the Thermocouple Short to Ground error followed a moment later by the Short to VCC error.

So now I have swapped out the entire system and the problems are persisting.  

Anybody have any thoughts on where to go from here?

Thanks!

murraycs, 2019-06-13:
This may sound silly, but double check the pin numbers on the Pi. The different revisions have different numbers than each other, so it's quite simple to plug the wrong wire in if you are inadvertently looking at the wrong diagram. I had to quadruple check when wiring mine together, and ultimately had to edit the pin numbers  in the code to make everything work logically.

virtualbuck, 2019-06-13:
Thank you.  If the physical pins of the schematic provided with this project are correct for the RPI3B+, then I have the pinouts correct.

murraycs, 2019-06-13:
The schematic shows just PIN numbers, not location. Just in case, here is a label for the pins:  https://www.raspberrypi-spy.co.uk/2014/07/raspberry-pi-b-gpio-header-details-and-pinout/

You can also run pinout from the command line. If it is not the pin, you may have a bad thermocouple. I’ve been running this software with no issue for 4 months now, so I’m fairly certain that you have a hardware issue rather than software.

virtualbuck, 2019-06-13:
Thank you Murray.  I rechecked everything.  Ran pinout.  Everything is wired correctly.  

I am the studio tech in the art dept at a college.  I have many thermocouples and have tried 4 different ones (2 new) with the same results.  I have two completely different hardware setups...the 3B+ with a breadboard and the zero with a soldered circuit.  same results.

When I first wired the breadboard, the controller ran like a champ...except I did see spurious spikes reported by the thermocouple...temperatures would be reported that were well outside the range of what was expected.  It would cause spikes on the graph.  But I never noticed errors in the console until recently.  

All this kind of points to the wiring to the max31855 breakout board or the board itself.  I bought both these boards in the last 3 months...maybe there is a bad batch?

murraycs, 2019-06-13:
So weird! Looking online, it the Max31855 has previously had bad batches, so that's probably a good theory. Please update us when you learn more.

virtualbuck, 2019-06-13:
I have taken the problem to Adafruit.  There are posts on the support pages that are very similar to what I am seeing.  One fellow said he resolved the problem by grounding everything to his kiln ground. Bizarre.

We will see where this leads and I will post an update when resolved.

virtualbuck, 2019-06-14:
I am working with Adafruit.  I wanted to cut and paste a few things from that post for posterity:

Here is the last comment from a fellow that was having same exact symptoms I am seeing:

"Got the new board, once I grounded out the board to the actual ground (of the kiln to ground) not just Raspberry Pi, the errors stopped. So thank you for sending along the hardware."

virtualbuck, 2019-06-14:
Me:   I can't ignore an imminent failure in a component in a kiln when I can't be certain that the failure won't lead to overheating.

Adafruit:For industrial controls, we address risks like that in a few ways:
1) Avoid single points of failure. In some cases, this means redundant sensors.
2) Scheduled preventive maintenance - components with finite lifetimes (i.e. thermocouple probes) are replaced regularly.
3) Design the system to 'fail safe'. In the case of temperature controls, in the absence of credible feedback, you bring the system to a safe state. Electric kilns are fairly straightforward systems in that regard. Things get much trickier when you are dealing with combustible gasses.

me:    are you saying that noise can cause these errors to report?

Adafruit: Yes, noise can cause these types of errors. The sensor is simply looking at voltages on pins. Errors are generated when certain thresholds are exceeded. The 'human antenna' test of pinching the probe tip demonstrates that nicely,

Me:    Why would OP's problem resolve by grounding the breakout board to the kiln? 

them:  Noise couples into systems in a variety of ways - some of which are not at all obvious. For example: if noise radiated from the kiln is picked up inductively by the probe, there will be voltage fluctuations at the inputs to the amp. If you ground the amp to the source of the voltage fluctuations, the signal are measured relative to that - cancelling out the noise.

me:  Is your recommendation to ground the breakout board to the kiln? My kiln's chassis is grounded to earth via mains ground. How would I ground the breakout board to the kiln? Just solder another lead onto the GND pin on the breakout board and take it to the kiln chassis...at the same time leaving the lead between GND on max31855 and the RPI header ground?

Or is you recommendation to ignore the errors?

them:   no reply yet.

aside from me...It is my understanding that taking a floating ground and wiring it to earth ground is not necessarily a good and proper thing to do...sometimes dangerous.  Interested to see what they recommend!!

virtualbuck, 2019-06-14:
so here is a question for everyone that is successfully running this project.  Did any of you add any extra ground to your circuit which is not shown in the schematic?  Is there such a thing as a 3 prong ac to dc (USB) adapter that uses the mains earth ground as the DC ground so that the RPI ground pins are actually grounded to earth through the power supply?  Is that a real thing?

virtualbuck, 2019-06-14:
Here is the Adafruit reply to my last two questions:

me:      Is your recommendation to ground the breakout board to the kiln?

AF:  Unfortunately, there is no one-size-fits-all solution for EMI problems. Even full-time EMI specialists will admit that a large part of their job is trial-and-error.

Grounding to the kiln is certainly worth trying. Many modern switching power supplies are isolated. So their 'ground' will be floating relative to the mains ground. And in older 2-wire system where ground and neutral were combined, current flow will cause voltage drops in the ground, so the ground potential can vary on different branches of the circuit.

me:    Or is you recommendation to ignore the errors?

AF:  The goal of course is to minimize the interference so you have fewer errors to ignore and more confidence in the overall quality of the data.

virtualbuck, 2019-06-14:
Before I added the earth ground, I verified twice that I could throw a Short to Ground (or VCC) error by pinching the thermocouple tip.  

To make the earth ground, I took a three prong ac plug and only wired the ground terminal.  I attached a lead between the ground bus on my breadboard and the ground-only ac plug.  Then I plugged in the AC ground plug in a normal, grounded receptacle.  I was  NOT able to reproduce errors by pinching the thermocouple.  I unplugged the ground again and verified that the errors were again produced as before.  

I am going into school in a little bit and I will try my grounding plug with my kiln.  I will update later.

virtualbuck, 2019-06-14:
I am back.  Great success.  Short to Ground and VCC errors were easy to recreate with my kiln but stopped as soon as I added the earth ground to my project.  I started a 24 hour, 200F soak on the kiln and I hope to see that all the anomalous temperature spikes are not on the graph tomorrow morning.

A couple of points that I picked up during my research.  The Raspberry Pi could be connected to an earth ground through the HDMI port.  If the display has an earth ground, your Pi likely does too.  Mine did not...just a 2 prong AC plug to a DC power supply.

Also, YMMV depending on the power supply you use with your Pi.  I was just using a crappy little wall wart.

Closing this ticket.   Have a nice day!!  Buck

So

closed 2019-06-14 by virtualbuck

jbruce12000, 2019-06-16:
glad you solved it. sorry I did not notice this earlier. yes, I had the same problem but only at very high temperatures... right at cone 6 I would get the same thing... an increase in errors for shorts to ground and vcc. they were intermittent so I had the software take 5 readings from the thermocouple. I suppose it's time to check my earth ground!

seesoe, 2020-07-21:
@jbruce12000 I'm having the same high temperature issues on my esp32 setup. I tried looping detect 5 times before aborting like you mentioned, but didn't help. Did you end up resolving your issue with earth ground or some other method?

tonymoon61, 2021-07-08:
> To make the earth ground, I took a three prong ac plug and only wired the ground terminal. I attached a lead between the ground bus on my breadboard and the ground-only ac plug. Then I plugged in the AC ground plug in a normal, grounded receptacle.

Brilliant! That's such a good idea. I ended up building my kiln controller inside of an old computer power supply. I used the power supply's original power jack. I spliced the ground wire in to the ground wire going between the MAX and the raspberry pi with great results. I've had no issues since doing that, but I also ordered a new MAX.

To anyone else experiencing this, look carefully at the pins. Some of mine had some solder crossing bridging two pins. Cutting those helped as well. There is some reason they come that way, but I am struggling to find the link.

amahpour, 2023-11-17:
@seesoe @jbruce12000 I'm experiencing the same issue at high temps. I designed my own HAT to go on top of my RPi but, foolishly, left out the ferrite beads. I'll probably redesign the board and add them back in.

One thing I didn't understand was the ground piece from @virtualbuck. Is @virtualbuck implying that I attached the AC ground to the RPi ground?

amattmiller, 2024-07-09:
Late to the party here, but I found this thread while trying to measure temperature on a noisy motor. I was able to make it work by adding a 100pF cap across the thermocouple inputs to the breakout board. If that works for you it should be easier than trying to ground the thing you are sensing.

Also, fwiw, this does seem like a noise issue. In that case, your pi or esp32 ground would be what really cares about noise. Beefing that up, adding bypass caps, and adding ferrite beads all sound like good things to try.  It could also be that the thing you are testing is loosely tied to some potential whether you realize it or not. You are putting your thermocouple onto what your system sees as a random potential. The disparity between that potential and your system potential may allow noise to get on the thermocouple lines. Since the your controller is floating, connecting the thing you are measuring back to system ground (pi's or esp32 or whatever) could help. That might be why Adafruit recommends giving it a try.