jbruce12000/kiln-controller #6: temp priority

sui001, 2019-07-09:
Hi,

is there a setting to make it temp priority that I have missed?

At the moment if I set the temp increase above the rate of the kiln it will just jump to the next segment after the time. I'd like to be able to have it temperature set.

thanks

jbruce12000, 2019-07-09:
I think the feature you're asking for would do this:

In a schedule, if a specific temperature is not reached, pause the schedule until the temperature is reached.  Is this just for heating? or for cooling too?

I envision this feature being complex to implement and causing lots of problems.

jbruce12000, 2019-07-09:
A good example of a problem it could cause...  If a kiln element goes bad during a firing, the kiln would just keep trying to reach temp forever.  So, for this specific case, it might make sense to have a configurable max time for a firing.

Another example.  Let's say you're in a cooling phase and your SSR goes bad.  These usually go bad with a short circuit meaning that all elements are on all the time.  The cooling temp is never reached because the kiln is running full blast.  Makes sense to use the max_temp and potentially have a configurable max firing time.  complicated.

sui001, 2019-07-09:
I use this a lot in firing schedules. For eg.
50°/hr till 550°
999° till 1100°
Doesn’t matter if kiln can’t keep up, just goes till it hits those temps. I check it regularly so any accident will get picked up, I never fire unattended. 

Cooling down isn’t an issue, because longer is usually preferable.

Lack of this feature isn’t a deal breaker :) 
It’s amazing to have, and I’m super keen to change my other kilns to this. 
Thanks

murraycs, 2019-07-09:
Max firing time on the controller level wouldn’t work if the SSR is shorted
out. I agree it would be a good feature, but one should have a physics fail
safe. The kiln sitter is one (max temp), and an upstream physical timer is
another. Some of the kiln sitters include one, but should be fairly easy to
add if you don’t have one. It’s actually a good reminder to me to go pick
one up.

All that being said, the ultimate fail safe is to monitor your kiln, and
you should be checking in periodically. It’ll be very apparent if one of
these failures happens and you can go manually turn it off.

jbruce12000, 2019-07-12:
@murraycs yeah, you're right.  I did not think that through fully.  A timer would not help if there was a short.  Power would continue to be applied through the SSR to the kiln.

@sui001 this feature does not exist today.  I'll take a look and see what I can do and where the change would need to be applied.

murraycs, 2019-07-12:
The simplest way is just to put a timer upstream of the elements, even
upstream of the SSR. Time runs out and the power is cut physically. Not
saying a software timer wouldn’t be useful - redundancy is good after all.

jbruce12000, 2019-07-12:
@murraycs In my case, I use an old kiln with a kiln-sitter.  This provides me the safety of shutting off should the kiln get too hot.  I'm constantly monitoring the kiln through the web interface and regularly through peep holes to make sure nothing has gone astray.

sui001, 2019-07-14:
hey - seems this  iteration might solve the issue: https://github.com/NickSifniotis/kilnController
thanks for the help

closed 2019-07-14 by sui001