jbruce12000/kiln-controller #7: Proposal

Hansuitherlev, 2020-04-20:
Great project; my wife is enjoying it a lot with her glass kiln!

I made your project by following the instructions. However, I am not a true programmer. Now there is one wish from my wife. It would be great if a profile could have a command where the temperature drops untill it reaches a certain temperature and then would maintain that temperature for a certain time period. I.e. "upon reachint Target temp, maintain target temp for xxx minutes".
Now one has to guestimate how long it will take for the kiln to coold own to the target temp.
Thanks again for a truly useful project with a truly usable instruction file!

jbruce12000, 2020-04-20:
In the industry I think this is called a hold. The kiln runs forever until it reaches a specific temperature (either going up or going down) and then it continues the rest of the schedule after that specific temperature is reached.  Today, with this code, the only way to achieve this is to set a cooling rate  that her kiln can actually attain (like 30/degrees per minute) and then have it cool there as long as she wants.  So to reiterate, the current software does not support the first part - achieve temperature X as soon as possible (but no time specified)... but the last part is supported (hold a temperature for X minutes or hours or days). 

You might wanna check this version out. I think it has this implemented...
https://github.com/NickSifniotis/kilnController

jbruce12000, 2021-03-18:
hmmm. I made this change for heating. It would probably not be hard to do it for cooling too.

jbruce12000, 2021-03-22:
done for cooling and heating. the schedule gets shifted if temp is more than N degrees from setpt.

closed 2021-03-22 by jbruce12000

Hansuitherlev, 2022-02-28:
Dear Jason, still we are using hour kiln controller a lot. Just installed it on kiln nr 2, however we noticed a major issue. When running a profile where a peak temperature is followed by a cooling down to a certain temperature the program doesn t pick up when reaching that temperature; it just keeps on cooling. 

The old version of the software dis not have this issue. What to do?

Best regards,
Hans van den Brink

jbruce12000, 2022-03-02:
Can you create an issue for this and post the log so I can see what's happening? I also need the profile which would be located in storage/profiles/whateveryoucalledit.json.

Hansuitherlev, 2022-03-03:
Thanks Jason, Will do so. Sendt fra Mail til Windows Fra: Jason BruceSendt: 2. marts 2022 23:36Til: jbruce12000/kiln-controllerCc: Hansuitherlev; AuthorEmne: Re: [jbruce12000/kiln-controller] Proposal (#7) Can you create an issue for this and post the log so I can see what's happening? I also need the profile which would be located in storage/profiles/whateveryoucalledit.json.