jbruce12000/kiln-controller #13: MAX31855 Measurement Instability

haydos, 2021-02-06:
Hi,

During testing I have noticed instability of the temp measurements from the MAX31855 board when observing the GUI.
The average temperature intermittently jumps +/- 40deg C, usually ~2mins between changing the average temp.
I connected the MAX31855 to an Arduino and have confirmed that stable measurements are being obtained from this board so it seems to be the Pi.
Can you provide any advice regarding what might cause this issue?

Thanks, Shannon

haydos, 2021-02-08:
I ended up using the Arduino as an interface to the MAX31855 to provide stable temperature readings.
I've modified the oven.py file:
- Added a tempSensor option for "Arduino"
- Imported serial library
- Obtain temperature measurements by serial comms using the Raspberry Pi USB connection
Happy to share the code if you're interested

jbruce12000, 2021-02-19:
Interesting. It sounds like a possible noise problem. I had a similar problem where simply twisting the leads to the thermocouple fixed it. Shielded cable is also an option, but often the leads are made of a specific material that cannot be changed.

Glad you came up with a work-around.

closed 2021-02-19 by jbruce12000

Giterd, 2021-04-13:
I know this is closed, but thought I would add a comment.  I use a Pi 4B and noticed quite a lot of temperature noise when connecting via wifi.  If I turn off bluetooth and wifi, and connect with an ethernet cable, the noise is almost entirely eliminated.  The thermocouple lead is wrapped around a ferrite core.