jbruce12000/kiln-controller #17: emergency_shutoff_temp doesn’t currently shut off heaters?

tanaes, 2021-04-12:
Amazing project! I'm thinking about implementing this for one of the kilns at our maker space. 

I was browsing the code, and noticed that although there is a `emergency_shutoff_temp` wisely defined with a note that SSRs can fail closed, it looks like all this does when tripped is to reset the Pi, which if I understand correctly would leave heater energized via the failed SSR. 

If this is indeed the case, what about adding a failsafe mechanical NO relay that is controlled from the RPi via a GPIO pin only enabled on successful startup?

If I'm missing something and this would be redundant let me know, otherwise I'm happy to submit a PR with the functionality if we implement it. 

Thanks for sharing this great project!

Giterd, 2021-04-12:
Related to this request, to avoid the "run forever" scenario, can a timeout feature be included?  My Paragon kiln controller does this, from its manual:

FtL / Fired Too Long
This message appears when both of the following conditions are met:
■ The temperature rise is less than 27°F / 15°C per hour.
■ The firing is 2 hours longer than programmed.

From config.py:

# If the kiln cannot heat or cool fast enough and is off by more than 
# kiln_must_catch_up_max_error  the entire schedule is shifted until
# the desired temperature is reached. If your kiln cannot attain the 
# wanted temperature, the schedule will run forever.
kiln_must_catch_up = True
kiln_must_catch_up_max_error = 10 #degrees

murraycs, 2021-04-12:
The optimal solution is to have a physical backup that is unrelated to the Pi, similar to what some Kiln Masters include. Twist a dial timer, and when it ends the kiln shuts off. Additionally, having a cone set on the physical switch is also advisable. Relying entirely on software for a fire hazard is risky at best.

Giterd, 2021-04-12:
I agree that a hardware shutoff is good practice.  I would still like the software timeout, to provide redundancy.

tanaes, 2021-04-12:
Thanks for the great ideas! The advantage of having a normally open mechanical relay 'deadman switch' is that it must be *actively* engaged by Pi, so in the event of either a failure detected by the controller (e.g. SSR stuck on) or a failure of the controller, it will not energize the heating circuit. (I suppose that it is technically possible that the transistor controlling the relay and the SSR would simultaneously fail closed, but that's rather far down my list of concerns!) 

Mostly, I'm a bit worried about the documentation *implying* a safety behavior that doesn't actually seem to be provided by the current code:

https://github.com/jbruce12000/kiln-controller/blob/df4d3bc91c2db2577a7d4d5d89a154c34bc15753/config.py.EXAMPLE#L86-L90

@Giterd the software time out is also a great idea, I can implement this as well.

@murraycs I'm afraid I'm on the electronics side of things at our space rather than the ceramics side. :D What do you mean by 'having a cone set on the physical switch'? Is it possible to have high cone positioned in such a way that, if it were to melt, would interrupt a physical circuit?

murraycs, 2021-04-13:
@tanaes this video of setting up a Kiln Sitter gives a better visual explanation than I could write out in text: https://youtu.be/8wo5W3vI8Ec

Timecode 1:20 shows the piece I’m referring to. 
Since Cones are a function of temperature **and** time, this should also be a backup even if the kiln never reaches full temp. Make sure if you have any holds that you have a high enough cone it won’t cause a premature shutdown. 

Some models of Kiln Sitters also include the mechanical timer I mentioned, going up to 24 hours. If for some reason you need longer you just twist the dial partway through.

tanaes, 2021-04-13:
Oh very cool! Thanks for linking that video. Serious Marty Stouffer vibes from that narrator!

Our ceramics area captain informs me that the kiln already has a kiln sitter, so we are indeed backed up. :)

jbruce12000, 2021-04-14:
If the temperature in the kiln exceeds emergency_shutoff_temp, a few things happen and none of them would kill the power to a potentially shorted SSR. To be clear, the PI is NOT RESET, but the current profile is stopped and messages are logged.

I have had requests for a safety relay in the past, but I don't have one so I can't test that. I have a kiln-sitter, so that is my failsafe. If you want to add settings and code for a safety relay, submit a pull request and I'll check it out.

jbruce12000, 2021-04-14:
As far as a firing timeout, this should be pretty easy to implement. I think I'd implement that as a timeout in total hours instead of hours past the length of firing profile. what do you think?

jbruce12000, 2021-04-14:
I see what you're saying about the wording of "emergency shutoff". Maybe this should change, but I think what folks should really keep in mind is... if there is a short, all control is lost.

jbruce12000, 2021-04-14:
Mostly, I'm a bit worried about the documentation implying a safety behavior that doesn't actually seem to be provided by the current code

@tanaes how would you change the wording? remember that this feature does do what it says, but if it cannot control the SSR because it failed, all bets are off.

tanaes, 2021-04-14:
I would probably just add what you said above! That if this temperature is reached, the profile is turned off. BUT if the reason the temp was reached is because the SSR shorted closed, that it can not do anything to cut power to it, and so it's a good idea to have a redundant backup of some sort.

Thanks! I will submit a PR if I can get the safety relay implemented successfully.

jbruce12000, 2021-04-14:
ok, I'll change that wording.

Giterd, 2021-04-14:
> @Giterd As far as a firing timeout, this should be pretty easy to implement. I think I'd implement that as a timeout in total hours instead of hours past the length of firing profile. what do you think?

Different profiles have different run times, so personally I would prefer a timeout set as hours past the current profile total time.  However, I can see that this might not be ideal if there is an extended controlled cooldown at the end. But a simple, fixed (in config.py) time will also work and I expect easier to implement.

jbruce12000, 2021-04-14:
I don't think implementation would be too hard either way. ok, I'll add the firing timeout sometime this week.

kayak01, 2021-04-14:
Might be worth adding a contactor inline with the ssr as a backup? If the kiln is not responding open the contactor?

tanaes, 2021-04-14:
> Might be worth adding a contactor inline with the ssr as a backup? If the kiln is not responding open the contactor?

Yup that's my plan! I'll have it wired to the normally open terminal, and add an enable output from a RPi GPIO under the run loop. That way if the loop breaks due to an over temp condition, or the Pi looses power or something, it cuts power to the SSR.

kendrick90, 2021-05-30:
Any progress on this? I'm working with a different kiln controller that uses an ESP32 but I'm interested in trying out this project too. I would like to include an electro-mechanical relay like [this one](https://www.amazon.com/gp/product/B07DSZFLMQ/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1) as a fail safe. The whole reason I'm looking into these things is that my kiln was having the SSR stuck on problem. It's scary to see it much hotter than it's supposed to be! Not to mentioned the ruined work... My kiln also has a switch attached to the door that opens the EMR when the door is opened.

tanaes, 2021-05-30:
Yup I actually have the code written, and need to assemble the hardware
to test it. Hopefully soon. 😅

—
<https://github.com/jbruce12000/kiln-controller/issues/17#issuecomment-
auth/AB7ISACE5SBN2HDEOANK7U3TQHHI7ANCNFSM42YM72FQ>.

marktilles, 2021-06-09:
Regarding an emergency overheat shut off, I simply took a cheap Rex-100 style PID controller and solid-state relay, and tied these in to the original physical relay which I left in my kiln.  I can set this device easily to any temperature, so if overheating ever exceeds my desired maximum temperature that separate device will shut down the physical relay. I also require a push button to initiate the circuit, as this makes sure that a power outage in my house will never let anything start up in an unknown state.  I did this by looping one terminal of the relay coil through the extra unused pair of control terminals on the relay, and having a separate momentary pushbutton trigger the relay to the on position. Problem solved.

marktilles, 2021-07-24:
> Different profiles have different run times, so personally I would prefer a **timeout set as hours past the current profile total** time. However, I can see that this might not be ideal if there is [...]

In my case, my large Chematex oven can't keep up with the profile curve we enter at the top ends of our firings.  I therefore use the new "catch-up" feature to ensure our firings actually reach the top temperature. However, I need to add a few minutes (ca 5 extra minutes) of flat temperature time into the top end of the curve to allow the kiln to actually catch up to the top temp, since the profile would otherwise stop when the target temp reaches its value, regardless if how many catch-up degrees the actual sensor temp lags behind.  My firings thus routinely take longer than the estimated firing time when the profiles are started. Sure, I can always go in and adjust my curves to more closely match the heating capacity of the oven at its higher temps, to ensure the estimated time more closely matches the actual. But unless one does this then deciding on a "fixed" amount of "too much runtime" for any particular kiln and firing curve, well in my mind, this could add yet another variable that would require even more explanation to the user. 

_One idea I have to improve the overheating safety options is as follows (I am already doing something similar with a separate REX-controller, with its very own thermocouple, but the Raspberry Pi could perform this function as well):_  
A Raspberry Pi based overheating device interlock circuit could be controlled using a GPIO as defined in the config.py file. This circuit could control a dedicated SSR that feeds A/C power to a physical contactor that all the heater SSRs are routed through. So, when an overheating condition occurs not only would the profile be canceled but this GPIO would shut down the contactor circuit as well.

felixn-unity, 2021-08-03:
Hi

I don't see any "emergency_shutoff_temp" in the latest code?

Thanks

marktilles, 2021-08-05:
It’s actually in there, in oven.py - and seems to work when testing - but its hard coded In picoreflow so the web interface doesn’t reflect real life operation. I’m hacking my own changes to the system but need to clean up my notes and document my reasons for my changes in a file. 
Mark

felixn-unity, 2021-08-06:
Thanks I am working with master branch and have not yet tried it on real Kiln, but thermocouple seems til work. I would be interested to see what changes you made. I would like to get the opendoor to work.

I will try it out on my sisters Kiln in a few days - and crossing fingers :)

Thanks

marktilles, 2021-08-06:
I have started a text file with some descriptions in the docs folder, but
its not done yet.  I have also placed remarks with my full name in caps in
most places in the code files where I have made changes. Have a look if you
like.

marktilles <https://github.com/marktilles>/kiln-controller
<https://github.com/marktilles/kiln-controller>

FGMarshall, 2021-10-04:
Since I have cheaper power at night I would like to set a delay period between when I load the kiln and the power on time. It would help to have a box on the web page where I could set a delay before the schedule starts.
I see I can set a dead delay at the start of each stored schedule but that takes longer than just filling in a box on the home page.
Where in the code can I add this please?

marktilles, 2021-10-04:
> Since I have cheaper power at night I would like to set a delay period between when I load the kiln and the power on time. It would help to have a box on the web page where I could set a delay [...]

Hi Marshall,

You know, I am a hacker type of user and I haven’t interfaced with Jason’s code base. I have my own branch of his code hanging out there under my own login, and I just last night got that feature working. My interface is a hell of a lot more feature rich than the original, but the only way that I could say right now that you could get it going would be to download a copy of my code and talking with me. Like I said, I’m not a true programmer but I managed to get things to work the way I like them too. I would be happy to hook you in online temporarily to my oven so you could see these different features I put in. You can email me at Markt3@tilles.net or text me on iMessage or WhatsApp at +46730264112. I am in Stockholm Sweden, so a bit different time zone then maybe you are in.

tanaes, 2021-10-04:
I've been working on an add-on board for a Raspberry Pi to simplify the retrofitting of our makerspace's kiln: https://github.com/tanaes/KilnHat

Now that the hardware prototype is together, I'll be working on adding some of these features to my fork of this repository. @marktilles I'd be interested in talking to you about some of the features you've added! Maybe I can help to merge them into the main branch.

marktilles, 2021-10-04:
Hi John, just give me a call, get in touch. I can open up my firewall and
let you see a demo anytime.

marktilles, 2021-10-04:
That’s a beautiful card you’re designing.

iog-creator, 2021-10-05:
I'd love to see if we can work together on this. I'm building a
temperature controller and using this setup. Do you by chance know how to
go about changing the interface?

Bryon McCoy

203-989-0875

marktilles, 2021-10-05:
Bryon,
Are you related to John Sanders as well? I have two of you that are working
on this board or something. I’m confused!

marktilles, 2021-10-05:
Bryon, I have managed to hack changes in the interface for my own use, but there’s not a chance in the world I could explain the programming changes. Better yet you use GitHub and look through my code. I still don’t even fully understand it! Just happy that I managed to hack it to work.

marktilles, 2021-10-21:
> wow. ok. lots of changes... way to many to accept into master all at once. I need changes in reviewable chunks that are logically ordered. I really like the work you've done and would love to incorporate the changes. Can you submit PRs for this?
> 

Hi Jason, I’m glad you like what I’ve done, but just as I’ve mentioned before I just hacked my way through it to get what I want to work – work. I’ve got a programming buddy that’s eventually going to help clean up my code, and maybe then figure out how to put my changes into small sections for updating your mainstream repository. The first thing he’s going to do is push my delay timer into the backend, and then I will look at taking the time to go back and figure out all the small changes. Were I a true programmer, I would have hung off of your code and done it the right way.

marktilles, 2021-11-14:
@jbruce12000 
Hi again, my buddy Marko B. has just today begun to go through my code to assist me to clean up and correct its code. He will follow standard procedures to propose my improvements and changes to your main branch by performing pull requests, one for each feature to add or change. Please stand by, I’m a creative thinker and think much of what I have to offer will be of interest to the kiln community.

tanaes, 2021-11-14:
> @jbruce12000 @adq @tanaes @iog-creator @elixn-unity
> 
> 
> 
> Hi again, my buddy Marko B. has just today begun to go through my code to assist me to clean up and correct its code. He will follow standard procedures to propose my improvements and changes to your main branch by performing pull requests, one for each feature to add or change. Please stand by, I’m a creative thinker and think much of what I have to offer will be of interest to the kiln community. 

That's great news! Looking forward to seeing the results. 

Just got my controller wired up this morning and hope to turn it on later today for testing. [image]

marktilles, 2021-11-14:
Wow Looks pretty fancy. Is it based upon the 31856 or 31855? And are those two displays the sensor and target temps?

tanaes, 2021-11-14:
It’s the 31856, and theres a Raspberry Pi under there running the web
server. I haven’t decided what the displays will show but probably
current temp and time remaining. 

—
<https://github.com/jbruce12000/kiln-controller/issues/17#issuecomment-
auth/AB7ISAHY53MOI77MBC7QVI3UL74AVANCNFSM42YM72FQ>.

iog-creator, 2021-11-14:
Wow. That looks complicated. do you have a diagram?

Bryon McCoy
203-989-0875

tanaes, 2021-11-14:
The PCB is on GitHub here: <https://github.com/tanaes/KilnHat>

marktilles, 2021-11-14:
Time remaining is a difficult one, since that’s completely dependent on the
heating capacity of the oven and the curve you have programmed. It will
also be affected by the installation quality of the oven, how much energy
it leaks etc. Personally I’d rather see the same as shown in the GUI, set
temperature and actual temperature. Or maybe you could have a push button
that you could scroll through the various configurations values values like
PID, etc. But you can easily change that in your software :-)

iog-creator, 2021-11-14:
Nice! I want! Have the revisions been made? Where did you get it?

Bryon McCoy
203-989-0875

bjornicus, 2021-12-18:
Something I've been thinking about while developing a very similar setup to this project (https://github.com/bjornicus/kilntrol, I only just found out about this project today from @marktilles) is having it it notify me (probably by text message using twillio) when something unexpected happens (or something expected that you want to be notified about).  That could be useful as another way to help not burn the house down or ruin your firing.  I'll be sure to share if I do find time to figure that out, but thought I'd share the idea in case it inspires someone else.

marktilles, 2022-01-13:
Jon, I like the idea of the emergency shutoff relay mod, but it might be worth noting in your release notes or instructions that in case of a power glitch or any CPU hang, you're still relying on the Pi system ti reset the GPIO. The "ideal" backup safety will always be a completely independent safety system. 
Mark

rondoc, 2022-01-22:
I am new to this but I am an EE and have been a potter for over 70 years! My kiln is a rotary one with a single silicon carbide element. works well and very reliable. The current control is manual with 16 levels of heating by a hard logic chip set up doing a PWM synced to the line frequency. I know this is off topic but I am trying to use JB's controller as is. Is there a way to use a test run to set up the PID numbers? I think I saw something but running loads of tests seems a bit much for me. AS far as safety goes the only safe method would IMNSHO be a completely separate cutoff with its own thermocoouple. It could be an external one if you knew the relationship between kiln temp/ time/ external skin temp...after all you want to prevent damage in rare event of s a shorted SSW. BTW my switch has been running for 35 years , made by Westinghouse. Use a voltage rating double the peak of the line input and a current also double your max and it will be very reliable. The isolator is a MOC 3060 and I use a back to back SCR module. sorry to ramble...
ronald.

rondoc, 2022-01-22:
PS I built the kiln myself, extremely efficient 8 cu ft.

jbruce12000, 2022-03-21:
@rondoc see https://github.com/jbruce12000/kiln-controller/blob/master/docs/pid_tuning.md#the-tuning-process for the autotuner. I tried it and it was close, but not perfect. It involves one test to 400 degrees which produces a csv file. That file is used to calculate the PID parameters.