jbruce12000/kiln-controller #18: Log file

Giterd, 2021-04-12:
My understanding is that logs are saved in /var/log/daemon.log.  This file exists on my Pi.  However, it contains data unrelated to the KilnController, seemingly related to the Pi OS.  So I don't get a log of my kiln runs.  Can the kiln log be renamed or saved elsewhere?  Or is there different problem/solution?  TIA

FYI, I've implemented KilnController exactly as described here, and it works a treat!  But my coding knowledge ended with Fortran many years ago and I rely on the generosity of strangers.  Thank you.

Giterd, 2021-04-14:
> See https://github.com/jbruce12000/kiln-controller/blob/master/docs/logs.md
> There is a command at the bottom of the page that will "grep" or extract the correct log lines. It uses sed, a stream editor, to grab data from each line and convert it to csv for import into a spreadsheet. I have a sample spreadsheet publicly available where I've done this. You can copy that google sheet and get decent stats with little work.

Jason, it seems I have not made myself clear.  The /var/log/daemon.log file does not contain any KilnController data, so there is none to extract.  The file seems to contain logs of system events.   It may be due to how my system is configured, but I have a fresh standard RasPi OS installed, with no changes besides your KilnController.

 if I run your grep command I get "Permission denied".  If I add a path to my home folder for the out.csv file, it works, but the csv is empty.  I hope I am not being stupid here, but something is not working, or conflicting.

murraycs, 2021-04-25:
grep and sed are not the difficult part to learn... focus your energy on learning Regex. Writing the actual search parameters is more important than what piece of software you use to do the search.

Giterd, 2021-04-27:
> grep and sed are not the difficult part to learn... focus your energy on learning Regex. Writing the actual search parameters is more important than what piece of software you use to do the search.

Thanks for the suggestion.  I'll start educating myself, although this one is lower in the priority list.  Thanks.