jbruce12000/kiln-controller #19: Are you interested in my kiln display code?

adq, 2021-04-19:
Hi, I'm not submitting this as a merge request yet. I was just wondering if you would be interested in me doing so?

I bought one of these for my pi: https://www.adafruit.com/product/4484

I wrote software talking to the exsting websocket API: https://github.com/adq/kiln-controller/blob/kilndisplay/kiln-display.py
(The buttons do not do anything right now)

Attached to my kiln, it looks like this (running the 200/250 test profile):  [image]

Main problem is I had to shift the GPIOs along to make room for the LCD.

jbruce12000, 2021-04-29:
cute. so small. so it shows schedule, elapsed time, time left and current temp (not set point, but actual)?

jbruce12000, 2021-04-29:
I see some libs in that code that are not used anywhere else. When you submit a PR, make sure to modify requirements.txt and docs.

adq, 2021-04-29:
Awesome. willdo!

Yeah, its just the current temp. Good call about the setpoint; will add it before MR!

adq, 2021-04-29:
Created an MR, so closing this issue.

closed 2021-04-29 by adq