jbruce12000/kiln-controller #21: Short to ground and VCC, stuck temperature

Giterd, 2021-04-22:
I'm having the problem that as the kiln temperature gets above 1000 deg F, I start to see intermittent thermocouple errors, initially with a short to ground, and then a short to VCC.  The run continues until eventually there is a divide by zero and thereafter the temperature is "stuck" and does not change.  The kiln keeps on firing as it thinks the temperature is not increasing per the profile.  Last night I rebooted the Pi when this happened, and things started working properly again (with short to ground but not VCC errors) and I was able to complete the run.

I'm guessing this is a problem somewhere between the MAX31855, the thermocouple wiring and the thermocouple itself.  In any event, the software runs out of ideas.  Maybe it can handle the errors somehow?  I did some googling, and it seems the short to ground problem is fairly common.  Suggested remedies include grounding the MAX ground terminal, inserting capacitors across the thermocouple and power supply connections.  Please let me have your ideas.  I include a relevant part of the log below.

2021-04-21 20:03:21,072 ERROR oven: problem reading temp
Apr 21 20:03:21 KilnController python[428]: Traceback (most recent call last):
Apr 21 20:03:21 KilnController python[428]:   File "/home/pi/kiln-controller/lib/oven.py", line 125, in run
Apr 21 20:03:21 KilnController python[428]:     temp = self.thermocouple.get()
Apr 21 20:03:21 KilnController python[428]:   File "/home/pi/kiln-controller/lib/max31855.py", line 42, in get
Apr 21 20:03:21 KilnController python[428]:     self.checkErrors()
Apr 21 20:03:21 KilnController python[428]:   File "/home/pi/kiln-controller/lib/max31855.py", line 82, in checkErrors
Apr 21 20:03:21 KilnController python[428]:     raise MAX31855Error("Thermocouple short to VCC")
Apr 21 20:03:21 KilnController python[428]: max31855.MAX31855Error: 'Thermocouple short to VCC'
Apr 21 20:03:21 KilnController python[428]: 2021-04-21 20:03:21,474 ERROR oven: problem reading temp
Apr 21 20:03:21 KilnController python[428]: Traceback (most recent call last):
Apr 21 20:03:21 KilnController python[428]:   File "/home/pi/kiln-controller/lib/oven.py", line 125, in run
Apr 21 20:03:21 KilnController python[428]:     temp = self.thermocouple.get()
Apr 21 20:03:21 KilnController python[428]:   File "/home/pi/kiln-controller/lib/max31855.py", line 42, in get
Apr 21 20:03:21 KilnController python[428]:     self.checkErrors()
Apr 21 20:03:21 KilnController python[428]:   File "/home/pi/kiln-controller/lib/max31855.py", line 80, in checkErrors
Apr 21 20:03:21 KilnController python[428]:     raise MAX31855Error("Thermocouple short to ground")
Apr 21 20:03:21 KilnController python[428]: max31855.MAX31855Error: 'Thermocouple short to ground'
Apr 21 20:03:22 KilnController python[428]: Exception in thread Thread-1:
Apr 21 20:03:22 KilnController python[428]: Traceback (most recent call last):
Apr 21 20:03:22 KilnController python[428]:   File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
Apr 21 20:03:22 KilnController python[428]:     self.run()
Apr 21 20:03:22 KilnController python[428]:   File "/home/pi/kiln-controller/lib/oven.py", line 130, in run
Apr 21 20:03:22 KilnController python[428]:     self.temperature = sum(temps)/len(temps)
Apr 21 20:03:22 KilnController python[428]: ZeroDivisionError: division by zero
Apr 21 20:03:22 KilnController python[428]: 2021-04-21 20:03:22,189 INFO oven: temp=1520.5, target=1521.7, pid=1.000, heat_on=2.00, heat_off=0.00, run_time=23086, total_time=34500, time_left=11413
Apr 21 20:03:22 KilnController python[428]: 2021-04-21 20:03:22,190 INFO oven: pid percents pid=104.32 p=31.25 i=63.97 d=4.79
Apr 21 20:03:24 KilnController python[428]: 2021-04-21 20:03:24,192 INFO oven: temp=1520.5, target=1521.8, pid=1.000, heat_on=2.00, heat_off=0.00, run_time=23088, total_time=34500, time_left=11411
Apr 21 20:03:24 KilnController python[428]: 2021-04-21 20:03:24,193 INFO oven: pid percents pid=106.83 p=32.85 i=62.47 d=4.67
Apr 21 20:03:26 KilnController python[428]: 2021-04-21 20:03:26,196 INFO oven: temp=1520.5, target=1521.9, pid=1.000, heat_on=2.00, heat_off=0.00, run_time=23090, total_time=34500, time_left=11409
Apr 21 20:03:26 KilnController python[428]: 2021-04-21 20:03:26,196 INFO oven: pid percents pid=109.34 p=34.39 i=61.05 d=4.57
Apr 21 20:03:28 KilnController python[428]: 2021-04-21 20:03:28,199 INFO oven: temp=1520.5, target=1522.0, pid=1.000, heat_on=2.00, heat_off=0.00, run_time=23092, total_time=34500, time_left=11407
Apr 21 20:03:28 KilnController python[428]: 2021-04-21 20:03:28,200 INFO oven: pid percents pid=111.85 p=35.85 i=59.69 d=4.46
Apr 21 20:03:30 KilnController python[428]: 2021-04-21 20:03:30,202 INFO oven: temp=1520.5, target=1522.1, pid=1.000, heat_on=2.00, heat_off=0.00, run_time=23094, total_time=34500, time_left=11405

At this point the temperature does not change and the kiln keeps heating.

murraycs, 2021-04-25:
The hardware fixes are going to be the best solution to have things function properly. Good grounding is important in general, and especially if it causes errors.

My proposed change to the software would be to have this create a rather prominent warning notice on the display, and then automatically trigger a shutdown if not manually overridden within a certain period of time. The option to manually override an auto-shutdown for something that is not necessarily going to be an emergency would allow one to finish a firing cycle manually and not lose all their ware to an improper firing if it shut down prematurely for non-safety reasons. Having that option to manually override be time-limited provides a backup so as not to allow the kiln to run indefinitely, and thus create a hazard similar to a failed SSR.

adq, 2021-04-26:
Not sure if its the same issue you're having, but I think I may have this one: https://forums.adafruit.com/viewtopic.php?f=31&t=169135#p827564

It starts happening around 900C exactly as that issue mentions. I'm using the same adafruit part

Giterd, 2021-04-27:
> Not sure if its the same issue you're having, but I think I may have this one: https://forums.adafruit.com/viewtopic.php?f=31&t=169135#p827564
> 
> It starts happening around 900C exactly as that issue mentions. I'm using the same adafruit part

Thank you for the great link, which certainly seems to capture my problem.  I had speculated that the plasma forming in the hot kiln might have something to do with it.

To add some background:  I am using the KilnController on a small, 2 KW kiln, which uses single phase, 110 V.  My Pi 4 is powered by a standard 18 W USB-C charger, plugged into the same outlet as the kiln.  In the beginning I was communicating with the Pi by wifi, but I found that the wifi caused noise in the temperature readings.  So I hard-wired an ethernet connection to a network switch nearby, which eliminated the noise, having turned off the wifi and bluetooth.  But the high temperature short to ground problems remain.

Python coding is above my pay grade, so my request to Jason and other contributors is to look at the suggestions 1 and 2 in https://forums.adafruit.com/viewtopic.php?f=31&t=169135#p827564 and consider implementing them.  I am willing to run the code as a guinea pig.

Thanks again to everyone for the support.

adq, 2021-04-27:
Interesting. What I'm trying to do is retrofit one of these with a more user friendly controller: https://www.prometheushobby.com/Kilns/Prometheus-Kilns-Pro1-PRG-p121c59.html

I'm using everything the came with the kiln untouched in place - the K-type themocouple, the SSR, the heating element - all I have done is replace the controller board.   The original thermocouple is not shielded as far as I can see - its just two pieces of wire in a heat protective sleeve.

Reading the link I sent closer, it says:
> temperature measurement in furnaces or kilns with a heating coil exposed to air, once the temperature reaches around 900-1000C

As you can see that kiln does NOT have an exposed heating coil (unless it is exposed somewhere close to the thermocouple hole). Hum.

I've been experimenting with disabling the VCC/GND error checking, and as the link says I do get temperature measurements that look about right, but above 900 I think they do tend to be a bit noisier (perhaps 'cos of the current leakage issue?). 

So in order to combat that, I have also replaced the simple 5-sample-per-period-temperature-loop with a moving average of 40 samples per period in order to try and smooth things out better. 

However, I'm still attempting to tune the PID settings so I can't say if the above are actually helping or not!

adq, 2021-04-28:
@Giterd I put an experimental "temp" branch together here: https://github.com/adq/kiln-controller/tree/temp
It seems to be working much better for me, but I'd keep a **very** careful eye on it if you do try it.

The changes are as follows:
* Now refuses to start a profile at all if noconnection/vccshort/groundshort errors reported.
* During profile running, it ignores vccshort/groundshort errors completely.
* Will automatically abort and cleanly shutdown if it loses connection to the thermocouple during profile run.
* Added moving average temperature code to smooth out the temperature readings - I found they got more noisy at higher temps. 
* It now outputs the raw k/p/i values from the pid loop so I can watch them while tuning the Kp/Ki/Kd values.

I think this should remove the divide by zero errors completely.

Giterd, 2021-04-29:
> @Giterd I put an experimental "temp" branch together here: https://github.com/adq/kiln-controller/tree/temp
> It seems to be working much better for me, but I'd keep a **very** careful eye on it if you do try it.

Thank you very much, it looks very promising.  I'll probably run it Friday and will be sure to report my experience.

adq, 2021-04-29:
Cool! be interested to know how it works for you.

I've spotted another problem (and I've been reading faaaar too much about PID controllers!).

I'm getting really bad "integral wind up". The integral part of the PID controller's function is to accumulate errors over time as a sort of memory. The problem is during "start up" or "wind up", that error tends to build up and eventually dominates the signal causing erroneous behaviour.

As an example, I set my kiln to ramp to 920C (from 20C room temp). This takes a while, so the integral memory becomes huge! So the integral dominates everything even when I'm at the desired temperature. There are various common solutions to this problem which I'm experimenting with.

I've not put them on that branch yet 'cos I'm still playing, but will update here when I do.

Giterd, 2021-04-29:
I'm doing a short, low-temp run of your branch as I write.  There were a few hiccups to get it running, with missing RPi.GPIO and max31856 stuff, but managed to sort it out.  So far, so good, but a high temp run is needed to properly test.

I have also noticed some weird things with the PID calculation.  As expected, the integral winds up when the temperature is too low, but does not seem to wind down when there is an overshoot.  So the set point offset does not get corrected, even when the profile calls for a steady temperature.  I think there may be some problem with the PID calculation algorithm.  Good that you are looking into it, thanks.

adq, 2021-04-29:
Ah yeah, AFAICS the PID code is correct, but the windup is causing exactly the problem we're seeing. I tried setting the Ki term really really small so it didn't accumulate so much, but then the integral didn't have any effect at all.

I'm by no means an expert on PIDs though, I've just been doing a lot of reading this week!

jbruce12000, 2021-04-29:
The divide by zero is certainly a bug. I'll put in a fix for that tonight. Note that you are failing 5 reads of the thermocouple in a row to run into this bug. That's a lot. I have had these noise errors in the past, but never so frequent.

A good technique to NOT build up that integral error is to have a schedule that mimics your kilns ability to heat. Start the schedule at ambient (60F or whatever) and heat the kiln at an achievable pace.

Another possibility is to ignore the integral altogether.

closed 2021-04-29 by jbruce12000

jbruce12000, 2021-04-29:
@Giterd I'm not sure what you're profile looks like so it's tough to comment about the PID behavior. If the actual is way below the setpt for a long period, the integral will accumulate a large value which effectively shifts the temperature above the setpt. It may take a very long time to settle depending on the Integral value you are using. I can see in my testing that the integral accumulates below the setpt and sheds above the setpt.

Please note that about a month ago I inverted the I value. The smaller the value, the bigger the impact. Most modern PID controllers you purchase work this way.

adq, 2021-04-29:
@jbruce12000 Even though the max31855 reports VCC/GND errors, it still seems to read the temperature ok. Check the link about the adafruit max31855 part I'm using: https://forums.adafruit.com/viewtopic.php?f=31&t=169135#p827564

I also see the same as @Giterd  - the integral is either HUGE by the time I get to 920C, or else the Ki has to be so insignificant (or large since its inverted in the config file) as to have pretty much zero effect on the PID controller's operation.

However, I do totally appreciate you've been developing this a lot longer than I have! I'll submit a merge request when I'm done: up to you if you want it or not!

jbruce12000, 2021-04-29:
This is an example of a run where the setpt is 150F, but the starting temp is 60F. You can see integral gets collected until it hits the setpt and then it slowly begins to decrease. So I think the PID loop is operating as it should.

https://gist.github.com/jbruce12000/e8ba58942d7c4172141d48564f5d0042

I'd urge you to change your profile so that it ramps to 920C at a rate that your kiln can achieve so that you don't accumulate so much error. Does that make sense?

jbruce12000, 2021-04-29:
@adq Noise is the cause of the failed reads in my experience. I also get some at high temps using the max-31855. The failed reads are intermittent. This is why I implemented the five reads and then take an average.

adq, 2021-04-29:
Thanks for the advice, I'll keep messing with it!

adq, 2021-04-29:
The slow ramp would make sense, except my original kiln controller lets me specify a ramp of "999" - ie "ramp as fast as you possibly can".  It can happily do that to 920C and then stabilise. I've not managed to do the same with the code as-is.

However, I've actually just managed to do it with my experimental anti-windup code. For more info on anti-windup, see (for example) https://info.erdosmiller.com/blog/pid-anti-windup-techniques -- I'm trying out the clamping technique.

I guess my use case (Jewelry kiln with PMC clay @920C for 30 mins) is different to yours though, perhaps this is why we're expecting different performance characteristics.

jbruce12000, 2021-04-30:
An interesting idea might be to have a setting that only allows Integral accummulation to happen if the temperature is relatively close to the setpt (say within X%). This would allow for heating as fast as possible but would prevent massive accumulation.

BTW, my suggestion is not to create a slow ramp, it's to create a ramp that matches the max heating curve of your kiln. I understand this is a PITA because it requires extra work.

jbruce12000, 2021-04-30:
I'm testing the **stop_integral_windup** setting. One thing that's interesting is that you can make the integral more aggressive without having to worry about drifting way above the setpt.

jbruce12000, 2021-04-30:
I also think these two new settings should be **on** by default. It's what people expect to happen....

heat as fast as possible to my setpt, and stay as close to my setpt as possible

Giterd, 2021-05-01:
@jbruce12000 and @adq I have just completed a cone 04 (1945 deg F) run using Andrew's 'temp' code, and it was perfect!  None of the errors I had previously, no temperature noise, just following the profile nicely.  I did not make any hardware modifications, like inserting capacitors or grounding.  I've not opened the kiln yet, but have no doubt it was ok.  Very happy and my thanks to both of you.

Regarding windup, I'm sure you understand that the integral is the area between the actual and the set temperature, positive or negative.  If there is a large difference between the two temperatures, a large integral can quickly accumulate, which can take a very long time to cancel out when the difference is small, so the reset might never happen.  The trick is to start the profile set temperature at the same value as the actual temperature.  I do like the idea of limiting windup and would like to see it implemented.

Sorry to resurrect a closed discussion, but I wanted to report my findings.