jbruce12000/kiln-controller #28: Results of testing stop_integral_windup code

adq, 2021-05-04:
I've been testing the stop_integral_windup code. Summary: it works really well!

First of all -for comparison - I tested with it _disabled_ at 500C:
[image]

Then I enabled it and tested at three temperature bands. No other changes:
200/250C
[image]

500/550C
[image]

920C
[image]

So, its **very** effective, many many thanks!

Giterd, 2021-05-07:
Nice.  Thanks for the feedback and making this happen.

jbruce12000, 2021-05-11:
Nice. glad it's worked well. closing!