jbruce12000/kiln-controller #30: Run never starts

EasyEJL, 2021-05-18:
I've just installed the kiln controller, and everything seems ok. the logs show it as starting

May 18 07:23:32 raspberrypi python[298]: 2021-05-18 07:23:32,125 INFO kiln-controller: Received (control): {"cmd":"RUN","profile":{"data":[[0,200],[1740,200],[1800,250],[3600,250]],"type":"profile","name":"test-200-250"}}
May 18 07:23:32 raspberrypi python[298]: 2021-05-18 07:23:32,131 INFO kiln-controller: RUN command received
May 18 07:23:32 raspberrypi python[298]: 2021-05-18 07:23:32,134 INFO oven: Running schedule test-200-250
May 18 07:23:32 raspberrypi python[298]: 2021-05-18 07:23:32,136 INFO oven: Starting

However the "timer" never advances, so the target temp stays at 0 and the time next to the clock icon never changes.

EasyEJL, 2021-05-18:
This is running on a pi zero w.

EasyEJL, 2021-05-18:
Also it does appear that the main loop is running as the template does update varying by the 1 degree back and forth around the actual temperature

EasyEJL, 2021-05-18:
Nevermind, it was config related. I think the config.example isn't in sync with vars added later.

closed 2021-05-18 by EasyEJL

rondoc, 2022-04-01:
> Hi, I'm having the same problem now and the time doesn't change can you
> help me?

Try opening and executing kiln-conttoller in the terminal. Does it run? If
not what do you see.

rondoc, 2022-04-01:
It takes less than hour to do a complete pi install and kiln-controller.
Might be you best solution. Use a new microsd formated . Add the pi os .
Then update that. Then install kiln- controller following instructions.
Make changes to the config.sys to meet your set up. Run the load on boot py.
Exec. kiln-controller. Then reboot. Now open the browser and enter
0.0.0.0:8081. Now it should run fine.

jbruce12000, 2022-04-05:
I made sure config.py.EXAMPLE is the same as config.py [and checked the change into the main branch] so hopefully new users wont have trouble when following the instructions.

rondoc, 2022-04-05:
You might need to enter
127.0.0.1:8081 in your browser. I used 0.0.0.0:8081 and it worked for me.
Can anyone explain the difference and what it all means to a non coder.
Thank you.
Ronald.

jbruce12000, 2022-04-06:
https://www.howtogeek.com/225487/what-is-the-difference-between-127.0.0.1-and-0.0.0.0/

I always use 127.0.0.1:8081 or localhost:8081.