jbruce12000/kiln-controller #31: Raspberry Pi OS 2021-05 missing libffi-dev

michthom, 2021-05-29:
The installation instructions / requirements.txt may need an update to include installation of libffi-dev as a prerequisite for recent versions of Raspberry Pi OS?

jbruce12000, 2021-06-01:
I downloaded and installed the latest version of RPI OS on an RPI and did not run into the dependency issue you had above. I did modify the instructions since NOOBs is deprecated. Those changes are pushed. Try the instructions again and open another issue if you have one.

closed 2021-06-01 by jbruce12000

michthom, 2021-06-01:
Perhaps this was because I was using a Raspberry Pi Zero W? I'll give it another go and see if I can replicate the problem but suspect this issue can remain closed. Thanks for your swift action and response.