jbruce12000/kiln-controller #32: MAX31856 not starting

twuyts, 2021-06-10:
Hi,

when starting the application, I get the following error:
```
(venv) pi@raspberrypi:~/kiln-controller $ ./kiln-controller.py 
2021-06-10 22:08:38,200 INFO kiln-controller: Starting kiln controller
2021-06-10 22:08:38,202 INFO kiln-controller: this is a real kiln
2021-06-10 22:08:38,203 INFO oven: import MAX31856 
2021-06-10 22:08:38,203 INFO oven: init MAX31856
2021-06-10 22:08:38,204 INFO oven: {'cs': 27, 'clk': 22, 'do': 17, 'di': 10}
2021-06-10 22:08:38,219 INFO Adafruit_MAX31856.MAX31856: {'cs': 27, 'clk': 22, 'do': 17, 'di': 10}
Traceback (most recent call last):
  File "./kiln-controller.py", line 44, in <module>
    oven = RealOven()
  File "/home/pi/kiln-controller/lib/oven.py", line 383, in __init__
    self.board = Board()
  File "/home/pi/kiln-controller/lib/oven.py", line 49, in __init__
    self.create_temp_sensor()
  File "/home/pi/kiln-controller/lib/oven.py", line 77, in create_temp_sensor
    self.temp_sensor = TempSensorReal()
  File "/home/pi/kiln-controller/lib/oven.py", line 126, in __init__
    ac_freq_50hz = config.ac_freq_50hz,
  File "/home/pi/kiln-controller/lib/max31856.py", line 130, in __init__
    software_spi['do'], software_spi['cs'])
  File "/home/pi/kiln-controller/venv/lib/python3.7/site-packages/Adafruit_GPIO/SPI.py", line 167, in __init__
    gpio.setup(sclk, GPIO.OUT)
  File "/home/pi/kiln-controller/venv/lib/python3.7/site-packages/Adafruit_GPIO/GPIO.py", line 278, in setup
    pull_up_down=self._pud_mapping[pull_up_down])
TypeError: argument 1 must be str, not int
```

(I added a few logging statements to check what was passed as parameter in the `software_spi` struct.)

I've tried to interpret the error, but the only thing I could come up with is configuring the pins as 'GPIO0_22' instead of 22, which just gives me another error:
```
2021-06-10 22:07:48,578 INFO kiln-controller: Starting kiln controller
2021-06-10 22:07:48,580 INFO kiln-controller: this is a real kiln
2021-06-10 22:07:48,580 INFO oven: import MAX31856 
2021-06-10 22:07:48,580 INFO oven: init MAX31856
2021-06-10 22:07:48,581 INFO oven: {'cs': 'GPIO0_27', 'clk': 'GPIO0_22', 'do': 'GPIO0_17', 'di': 'GPIO0_10'}
2021-06-10 22:07:48,597 INFO Adafruit_MAX31856.MAX31856: {'cs': 'GPIO0_27', 'clk': 'GPIO0_22', 'do': 'GPIO0_17', 'di': 'GPIO0_10'}
Traceback (most recent call last):
  File "./kiln-controller.py", line 44, in <module>
    oven = RealOven()
  File "/home/pi/kiln-controller/lib/oven.py", line 383, in __init__
    self.board = Board()
  File "/home/pi/kiln-controller/lib/oven.py", line 49, in __init__
    self.create_temp_sensor()
  File "/home/pi/kiln-controller/lib/oven.py", line 77, in create_temp_sensor
    self.temp_sensor = TempSensorReal()
  File "/home/pi/kiln-controller/lib/oven.py", line 126, in __init__
    ac_freq_50hz = config.ac_freq_50hz,
  File "/home/pi/kiln-controller/lib/max31856.py", line 130, in __init__
    software_spi['do'], software_spi['cs'])
  File "/home/pi/kiln-controller/venv/lib/python3.7/site-packages/Adafruit_GPIO/SPI.py", line 167, in __init__
    gpio.setup(sclk, GPIO.OUT)
  File "/home/pi/kiln-controller/venv/lib/python3.7/site-packages/Adafruit_GPIO/GPIO.py", line 278, in setup
    pull_up_down=self._pud_mapping[pull_up_down])
SystemError: <built-in function setup> returned NULL without setting an error
```

Any idea what is wrong?

(I attached my config.py as well.)
[config.py.txt](https://github.com/jbruce12000/kiln-controller/files/6634214/config.py.txt)

jbruce12000, 2021-06-11:
I'd like to see versions of libraries you have installed. Please do the following:

source venv/bin/activate
pip freeze

and post the output here. What version of python are you running?

python -V

and post that output too.
@adq I don't have a max31856. can you help troubleshoot?

jbruce12000, 2021-06-11:
The section of code that is running appears to be for the Beaglebone Black. I am not sure why a raspberry pi would be running this section of code.

twuyts, 2021-06-12:
```
pi@raspberrypi:~/kiln-controller $ source venv/bin/activate
(venv) pi@raspberrypi:~/kiln-controller $ python -V
Python 3.7.3
(venv) pi@raspberrypi:~/kiln-controller $ pip freeze
Adafruit-BBIO==1.2.0
Adafruit-Blinka==6.10.1
adafruit-circuitpython-busdevice==5.0.6
adafruit-circuitpython-max31856==0.9.7
Adafruit-GPIO==1.0.3
Adafruit-MAX31855==1.6.1
Adafruit-PlatformDetect==3.13.4
Adafruit-PureIO==1.1.8
bottle==0.12.19
gevent==21.1.2
gevent-websocket==0.10.1
greenlet==1.1.0
pkg-resources==0.0.0
pyftdi==0.53.1
pyserial==3.5
pyusb==1.1.1
rpi-ws281x==4.3.0
RPi.GPIO==0.7.0
spidev==3.5
sysv-ipc==1.1.0
zope.event==4.5.0
zope.interface==5.4.0
```

twuyts, 2021-06-12:
@jbruce12000 your comment got me thinking, so I started from scratch in a different virtualenv:
Here's the complete output:
```
pi@raspberrypi:~/kiln/kiln-controller $ virtualenv -p python3 env2
Running virtualenv with interpreter /home/pi/kiln-controller/venv/bin/python3
Using real prefix '/usr'
Path not in prefix '/home/pi/kiln-controller/venv/include/python3.7m' '/usr'
New python executable in /home/pi/kiln/kiln-controller/env2/bin/python3
Also creating executable in /home/pi/kiln/kiln-controller/env2/bin/python
Installing setuptools, pkg_resources, pip, wheel...done.  
pi@raspberrypi:~/kiln/kiln-controller $ source env2/bin/activate
(env2) pi@raspberrypi:~/kiln/kiln-controller $ python -V
Python 3.7.3
(env2) pi@raspberrypi:~/kiln/kiln-controller $ pip freeze
pkg-resources==0.0.0
(env2) pi@raspberrypi:~/kiln/kiln-controller $ pip install --upgrade setuptools
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: setuptools in ./env2/lib/python3.7/site-packages (57.0.0)
(env2) pi@raspberrypi:~/kiln/kiln-controller $ pip install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: setuptools in ./env2/lib/python3.7/site-packages (from -r requirements.txt (line 1)) (57.0.0)
Collecting greenlet
  Using cached https://www.piwheels.org/simple/greenlet/greenlet-1.1.0-cp37-cp37m-linux_armv7l.whl (155 kB)
Collecting bottle
  Using cached https://www.piwheels.org/simple/bottle/bottle-0.12.19-py3-none-any.whl (89 kB)
Collecting gevent
  Using cached https://www.piwheels.org/simple/gevent/gevent-21.1.2-cp37-cp37m-linux_armv7l.whl (5.2 MB)
Collecting gevent-websocket
  Using cached https://www.piwheels.org/simple/gevent-websocket/gevent_websocket-0.10.1-py3-none-any.whl (22 kB)
Collecting Adafruit-MAX31855
  Using cached https://www.piwheels.org/simple/adafruit-max31855/Adafruit_MAX31855-1.6.1-py3-none-any.whl (5.5 kB)
Collecting Adafruit-GPIO
  Using cached https://www.piwheels.org/simple/adafruit-gpio/Adafruit_GPIO-1.0.3-py3-none-any.whl (38 kB)
Collecting zope.event
  Using cached https://www.piwheels.org/simple/zope-event/zope.event-4.5.0-py2.py3-none-any.whl (6.8 kB)
Collecting zope.interface
  Using cached https://www.piwheels.org/simple/zope-interface/zope.interface-5.4.0-cp37-cp37m-linux_armv7l.whl (240 kB)
Collecting spidev
  Using cached https://www.piwheels.org/simple/spidev/spidev-3.5-cp37-cp37m-linux_armv7l.whl (40 kB)
Collecting adafruit-pureio
  Using cached https://www.piwheels.org/simple/adafruit-pureio/Adafruit_PureIO-1.1.8-py3-none-any.whl (12 kB)
Installing collected packages: zope.interface, zope.event, spidev, greenlet, adafruit-pureio, gevent, Adafruit-GPIO, gevent-websocket, bottle, Adafruit-MAX31855
Successfully installed Adafruit-GPIO-1.0.3 Adafruit-MAX31855-1.6.1 adafruit-pureio-1.1.8 bottle-0.12.19 gevent-21.1.2 gevent-websocket-0.10.1 greenlet-1.1.0 spidev-3.5 zope.event-4.5.0 zope.interface-5.4.0
(env2) pi@raspberrypi:~/kiln/kiln-controller $ pip freeze
Adafruit-GPIO==1.0.3
Adafruit-MAX31855==1.6.1
Adafruit-PureIO==1.1.8
bottle==0.12.19
gevent==21.1.2
gevent-websocket==0.10.1
greenlet==1.1.0
pkg-resources==0.0.0
spidev==3.5
zope.event==4.5.0
zope.interface==5.4.0
(env2) pi@raspberrypi:~/kiln/kiln-controller $ ./kiln-controller.py 
2021-06-12 16:52:08,283 INFO kiln-controller: Starting kiln controller
2021-06-12 16:52:08,296 INFO kiln-controller: this is a real kiln
2021-06-12 16:52:08,296 INFO oven: import MAX31856 
2021-06-12 16:52:08,296 INFO oven: init MAX31856
Traceback (most recent call last):
  File "./kiln-controller.py", line 44, in <module>
    oven = RealOven()
  File "/home/pi/kiln/kiln-controller/lib/oven.py", line 382, in __init__
    self.board = Board()
  File "/home/pi/kiln/kiln-controller/lib/oven.py", line 49, in __init__
    self.create_temp_sensor()
  File "/home/pi/kiln/kiln-controller/lib/oven.py", line 77, in create_temp_sensor
    self.temp_sensor = TempSensorReal()
  File "/home/pi/kiln/kiln-controller/lib/oven.py", line 125, in __init__
    ac_freq_50hz = config.ac_freq_50hz,
  File "/home/pi/kiln/kiln-controller/lib/max31856.py", line 127, in __init__
    gpio = Adafruit_GPIO.get_platform_gpio()
  File "/home/pi/kiln/kiln-controller/env2/lib/python3.7/site-packages/Adafruit_GPIO/GPIO.py", line 420, in get_platform_gpio
    import Adafruit_BBIO.GPIO
ModuleNotFoundError: No module named 'Adafruit_BBIO'
```

Maybe something goes wrong in the platform detection, and the library incorrectly decides it needs to use Adafruit_BBIO instead of something else?

twuyts, 2021-06-12:
FYI:
```
ypi:~/kiln/kiln-controller $ gpio -v readall
gpio version: 2.52
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 4B, Revision: 01, Memory: 4096MB, Maker: Sony 
  * Device tree is enabled.
  *--> Raspberry Pi 4 Model B Rev 1.1
  * This Raspberry Pi supports user-level GPIO access.
```

twuyts, 2021-06-12:
found it:
you're using the Adafruit_GPIO library, which is deprecated. It does not correctly detect the underlying hardware. My Pi returns as hardware in `proc/cpuinfo` 'BCM2711'. But the library does not recognize this as a valid Pi, and as such continues as if it were a Beaglebone.
see https://github.com/adafruit/Adafruit_Python_GPIO/blob/a12fee39839665966bd124fd22588b2c87ced9d2/Adafruit_GPIO/Platform.py#L84

I'll patch my local copy of the adafruit gpio library and take it from there.

jbruce12000, 2021-06-13:
last time I checked and tried out the blinka lib (the replacement for adafruit gpio), it did not support vanilla linux. This means that simulations would have to be removed. That is a show-stopper for me. Someone will eventually add linux to blinka. Until then, I don't want to switch.

closed 2021-06-13 by jbruce12000