jbruce12000/kiln-controller #33: config not imported

artcank, 2021-06-24:
I am having an issue where it says it can't import the config file and that I need to copy it from the example. Even though I have done that and changed what I needed to in it.

I am using the latest rasbian BTW.

jbruce12000, 2021-06-25:
Are you running on a raspberry pi? Are you running a simulation? Are you using max31855 or max31856? did you pip install the requirements you need for your hardware?

artcank, 2021-06-25:
Running on raspberry pi following all the installation
requirements/procedures. I am using a max31856.

jbruce12000, 2021-06-25:
ok. Can you post the log? should be sent to /var/log/syslog

artcank, 2021-06-25:
[syslog.txt](https://github.com/jbruce12000/kiln-controller/files/6717693/syslog.txt)

artcank, 2021-06-26:
[image]

I switched to another raspberry pi and I was able to read my GPIO's but when I tried to run the config.py file I got this error:
ModuleNotFoundError: No module named 'Adafuit_GPIO'

jbruce12000, 2021-06-28:
open a shell and...

```
source venv/bin/activate
pip install Adafruit-GPIO
```

artcank, 2021-06-28:
That didn't work but I was able to follow the instructions on adafruit's github_GPIO page and get that installed. Then it said I didn't have RPi.GPIO installed. I got that installed and it WORKED!

It is reading my S type sensor!!! 

Now to get the SSD wired up and make a box to install on the kiln.

Thanks

jbruce12000, 2021-06-29:
ok. I'm interested to see what you did so I can change instructions if needed. Did you keep a log of the commands you ran? Were you using a virtualenv?

jbruce12000, 2021-06-29:
glad you got it working though. I think you're the third one to get an s-type thermocouple working with a max-31856. congrats!

jbruce12000, 2021-07-01:
assuming you're not tracking this. closing. lmk if you have some details I should change in the docs.

closed 2021-07-01 by jbruce12000

artcank, 2021-07-02:
Okay.
I had it working on a temp breadboard but when I made a soldered board it's not working again... Trying to troubleshoot and get it figured out.

artcank, 2021-07-15:
Got it running on the kiln today! Trying my first bisque fire. I had trouble like someone else with the GPIO numbering but that is now sorted. I'll look at the data to try and tune the PID later.

I had a thought while getting it up and running headless on the kiln, how hard would it be to put a shutdown/reboot RaspberryPi button on the web page? Also maybe a Restart Kiln-Controller