jbruce12000/kiln-controller #37: 2 or 3 phase kiln

felixn-unity, 2021-08-03:
Hi

If I want to use a 2 or 3 phase kiln, should I just double/trible the diagram to the solid relays or can I just connect solid relays in parallel?

Thanks!
Felix

marktilles, 2021-08-07:
> Hi
> 
> If I want to use a 2 or 3 phase kiln, should I just double/trible the diagram to the solid relays or can I just connect solid relays in parallel?
> 
> Thanks!
> Felix

I put my the three Chinese 40A SSRs (but after disecting one, I see mine are actually 30A-ers) in parallel. Works fine like this.

jbruce12000, 2021-08-09:
Well, using separate relays is not great. They will likely not energize at the same time. This may or may not be a problem for you. You can get zero crossing ssrs, but even these will not be in sync. It's best to get a ganged relay like this if you need to control multiple legs. https://www.auberins.com/index.php?main_page=product_info&cPath=2_30&products_id=331

closed 2021-08-09 by jbruce12000