jbruce12000/kiln-controller #38: ModuleNotFoundError: No module named 'RPi'

och29, 2021-08-07:
`~/kiln-controller $ source venv/bin/activate; ./kiln-controller.py
2021-08-07 12:31:28,131 INFO kiln-controller: Starting kiln controller
2021-08-07 12:31:28,149 INFO kiln-controller: this is a real kiln
2021-08-07 12:31:28,152 INFO oven: import MAX31855
2021-08-07 12:31:28,155 INFO oven: init MAX31855
Traceback (most recent call last):
  File "./kiln-controller.py", line 44, in <module>
    oven = RealOven()
  File "/home/pi/kiln-controller/lib/oven.py", line 382, in __init__
    self.board = Board()
  File "/home/pi/kiln-controller/lib/oven.py", line 49, in __init__
    self.create_temp_sensor()
  File "/home/pi/kiln-controller/lib/oven.py", line 77, in create_temp_sensor
    self.temp_sensor = TempSensorReal()
  File "/home/pi/kiln-controller/lib/oven.py", line 109, in __init__
    from max31855 import MAX31855, MAX31855Error
  File "/home/pi/kiln-controller/lib/max31855.py", line 2, in <module>
    import RPi.GPIO as GPIO
ModuleNotFoundError: No module named 'RPi'
`

I'm getting the following error when trying to get the service to run on a Pi Zero W using the documentation

Any suggestions?

murraycs, 2021-08-08:
MAX31855.py imports RPi.GPIO, but the requirements.txt has that library commented out. I am not at home to try this, but you could manually add RPi.GPIO or uncomment it from your requirements.txt and follow the docs again.

och29, 2021-08-08:
Thanks, uncommenting it worked a treat.

closed 2021-08-08 by och29

jbruce12000, 2021-08-09:
uncommented in master

captaincaden, 2022-01-31:
I'm getting the same error, and can't figure out how to get passed it. 

2022-01-31 14:30:45,271 INFO kiln-controller: Starting kiln controller
2022-01-31 14:30:45,277 INFO kiln-controller: this is a real kiln
2022-01-31 14:30:45,277 INFO oven: import MAX31856 
2022-01-31 14:30:45,278 INFO oven: init MAX31856
Traceback (most recent call last):
  File "/home/pi/kiln-controller/./kiln-controller.py", line 44, in <module>
    oven = RealOven()
  File "/home/pi/kiln-controller/lib/oven.py", line 381, in __init__
    self.board = Board()
  File "/home/pi/kiln-controller/lib/oven.py", line 47, in __init__
    self.create_temp_sensor()
  File "/home/pi/kiln-controller/lib/oven.py", line 75, in create_temp_sensor
    self.temp_sensor = TempSensorReal()
  File "/home/pi/kiln-controller/lib/oven.py", line 120, in __init__
    self.thermocouple = MAX31856(tc_type=config.thermocouple_type,
  File "/home/pi/kiln-controller/lib/max31856.py", line 127, in __init__
    gpio = Adafruit_GPIO.get_platform_gpio()
  File "/home/pi/kiln-controller/venv/lib/python3.9/site-packages/Adafruit_GPIO/GPIO.py", line 417, in get_platform_gpio
    import RPi.GPIO
ModuleNotFoundError: No module named 'RPi'

I've never really done anything like this before, so it's amazing that I got this far. I'm using MAX31856 if that makes a difference, and RPi.GPIO is uncommented in requirements.txt