jbruce12000/kiln-controller #40: max31856 pinout

felixn-unity, 2021-08-31:
Hi

I have a max31856

so pinput is like this:

D0 = SD0 (max31856)
CS = CS (max31856)
CLK = SCK (max31856)

Or?

Thanks

[image]

JpTemps, 2021-08-31:
I was able to get the Adafruit Universal Thermocouple Amplifier MAX31856 working by connecting as described in the Adafruit guide:
[image]

then modify config.py as shown below:
### Outputs-  changed from 23 to 16  due to conflict with SPI_CLK
gpio_heat = 16  # Switches zero-cross solid-state-relay GPIO 16 Pin 36

### I am using an S-Type thermocouple, but see lib/max31856.py for other thermocouple_type
thermocouple_type = MAX31856.MAX31856_S_TYPE

### max31856 Thermocouple Connection (using bitbang interfaces)
gpio_sensor_cs = 5      # pin 29  (GPIO 05)
gpio_sensor_clock = 11  # pin 23 (GPIO 11, SPI_CLK)
gpio_sensor_data = 9    # pin 21 (GPIO 09 SPI_MISO)
gpio_sensor_di = 10     # pin 19 (GPIO 10 SPI_MOSI) 

Hopefully this helps.

felixn-unity, 2021-08-31:
Hi @JpTemps 

This is super helpfull and thanks a lot!

Looking forward to get the MAX31856 and test it out - I gather that "kiln-tuner.py" will function the same way?

Are you using this project and did you do any approvement on it?

Felix

JpTemps, 2021-09-01:
Hi Felix:

I haven't got to the point of using the "kiln-tuner.py" as I am stilling gathering the components to assemble the circuit, but I am guessing it should work.
I am working on adding a main 2 pole - 40 amp contactor to control the power to the kiln. but am having issues controlling a second gpio output.  I've gotten to a point where I can turn on my "gpio_power" on the start of the run , but haven't been able to determine where the code to turn it off should go. 
I'm not much of a programmer so its a lot of trial and error on my part.
Ideally I would like to add a  "Power Management" button on the main screen to provide a screen of options to turn on the kiln's main power, turn off the kiln's main power  and to shut down the Raspberry Pi . I have modified index.html to display these options but haven't  been able to figure out how to get them to work.
One step at a time....

John

felixn-unity, 2021-09-01:
Ok but having the SSD relay is that not enough - when not on there is no power going to the oven?

And there is a "STOP" button that will stop the current run

Felix

felixn-unity, 2021-09-02:
btw.

I can't see this is an issue:

Outputs- changed from 23 to 16 due to conflict with SPI_CLK
gpio_heat = 16 # Switches zero-cross solid-state-relay GPIO 16 Pin 36

pin 16 GPIO 23 is free?

I will get my MAX31856 today I hope and will test it out and let you know :)

Felix

JpTemps, 2021-09-03:
Hi Felix:
I have a 2-pole contactor in the circuit as I don’t want to rely on the SSR in the event of an SSR failure.  I have modified the code so that when the current profile is stopped, the contactor is opened, disconnecting the power to the kiln. In the event of a  shorted SSR  and the temperature reaches the emergency_shutoff_temp, this configuration will actually kill the power to the kiln.

Also on a 240 volt kiln, the SSR would only disconnect one hot leg, so the second hot leg would be energized, providing 120 volts to the kiln.     

I have all of the features that I added working now, and am in the process of verifying all my logic is correct and cleaning up the code.

Let me know how the MAX31856  is working.

John

felixn-unity, 2021-09-03:
Can the SSR fail?

I don't undertand the stuff with 120v? the SSR will cut the fase right? so should be no power going to the kiln or?

I have 3 phase SSR so all my fases are on/off.

Anyway I got the MAX today, so need to wire it up and test it :)

Felix

felixn-unity, 2021-09-03:
It seem to work, but I get some of these errors in the logs

2021-09-03 20:51:41,773 ERROR oven: Problem reading temp N/C:False GND:False VCC:False ???:True

I think the connection is not optimal.

I turn on my kicthen oven to 250 degrees and is it normal that the thermocoupler is very slow getting up to the tempature?

It took about 15 min for the kiln-controller to show the temperatur, and oven was at 250 maybe after 8-10 min.

Felix

JpTemps, 2021-09-04:
Did you set the correct type of thermocouple in the configuration file? I think it defaults to S-type.

FlexyZ-xx, 2021-09-04:
yes I did select Type K

Is it normal for detection of temperature is that "slow"?

Felix

JpTemps, 2021-09-04:
Mine appears fairly slow as well. I tested the MAX31856 on a different Raspberry Pi using the Adafuit example programs to ensure it was working correctly,  before moving it to the Kiln Controller Pi. 
You may want to check the polarity on the thermocouple as well.
As for the service supply, my kiln is connected to a 120/240 volt, 3 wire, single phase circuit. The three wires consist of two hot legs from the output of the utility transformer with a neutral connection on the center point of the transformer winding. The neutral point is also grounded. 
There is 120 volts from each hot leg to neutral and 240 volts from hot leg 1 to hot leg 2.
The kiln only uses the two hot legs and a ground. By switching only one hot leg you can control the heating, but with only the SSR, one hot leg is still energized to 120 volts to ground. That is the reason I need the two pole contractor.

FlexyZ-xx, 2021-09-04:
ok so you have two 120 fases? maybe you could use a 3 phase ssr?

anyway I will solder it all up and try it on my sisters kiln on Sunday 

It looks like in the code that is will turn off the ssr if over max temperature 

Thanks
Felix

JpTemps, 2021-09-04:
Hi Felix:
Have fun soldering and good luck with the kiln on Sunday!
I could use a 3 phase SSR, or a second single phase SSR, but as I see it there are two issues with this solution, 1- you still have the problem that you cannot programmatically  disconnect the power in the event of an SSR failure, and 2- cost. 
A high quality SSR of adequate capacity, along with the required heat sink would cost over $400 CAD for a three phase unit  and over $110 CAD for a single phase unit. A 2 pole contactor costs just over $40 CAD .  To drive the contactor coil, I  have  a $10 CAD 5v DC relay between the Raspberry Pi and the contactor.    So for around $160.00 ( single phase SSR & heat sink, 2-pole contactor, and a 5v DC relay), I can control the kiln and be confident the program can shut off the kiln.  

John

jbruce12000, 2021-09-19:
@JpTemps strange that you would need to control two gpio outputs. You can have one output drive both SSRs, or use a ganged SSR like https://www.auberins.com/index.php?main_page=product_info&cPath=2_30&products_id=331

jbruce12000, 2021-09-19:
@JpTemps gotcha. you're trying to handle an SSR failure condition with another SSR. I have had a few SSRs fail in my time. They always fail closed - so full power to the kiln. It's unlikely, but you could have a cascading failure of SSRs. That tact I've taken is to monitor the kiln while it's firing. All kilns should be tended regardless of controller type or likelihood of failure.

JpTemps, 2021-09-19:
Hi Jason, 
The second gpio output controls a two pole electro-mechanical contactor that is energized at the start of the run and de-energized at the end of the run. The  contactor interrupts both hot legs of supply to the kiln. I know its overkill, but I want to ensure the program can turn off the kiln in the event of a failure , and yes the kiln will be monitored when firing.

iog-creator, 2021-10-06:
> I also use 3-phase 230VAC SSRs, but have a $20 PID controller from
> Banggood or Wish with its own dedicated S-type thermocouple together with
> the original contactor as a safety override kill system. Two independent
> systems: Raspberry pi to operate, cheap PID as safety kill override.

Dang. lol

closed 2021-10-21 by jbruce12000