jbruce12000/kiln-controller #41: Running in parallel with an existing controller

grantland, 2021-09-07:
Hi! I have a Skutt 818 which has an electronic controller and was wondering if it was possible to run this in parallel with the existing electronic controller. My use case would be to fire the kiln with the existing controller and basically use this project to monitor temps remotely and was thinking of being able to wire the thermocouple to both the boards in parallel without any relays, but I'm not confident if that would work. Thanks!

amahpour, 2021-09-13:
@grantland I also have a Skutt with an electronic controller ( + a retrofitted manual kiln with this setup) and I just use this without the relay to monitor the kiln progress. Set everything up like you would normally without the relay and create a bogus profile. In my case, for example, I created a profile to reach a temperature of 2000 F after 20 hours. The white line won't follow the green line but it gets the job done without any extra configuration.

jbruce12000, 2021-09-19:
certainly possible. Wire up the MAX-31855 or 31856, connect your thermocouple to that. Don't wire any output. The software will send signals to control whatever output you configure, but the output would just be ignored. @amahpour is correct that you just create a really long schedule with a high temp that you'll never reach.

closed 2021-09-19 by jbruce12000

grantland, 2022-09-30:
@amahpour @jbruce12000 Sorry to confirm, I can wire up the existing relay connected to the existing mainboard in parallel to the rpi or would I have to get an additional thermocouple that's isolated from the existing set up? Thanks!

jbruce12000, 2022-09-30:
@amahpour I wouldn't mess with existing thermocouples or relays. Since you just want to monitor temperatures remotely, you just need to wire up a max-31855 or max-31856 to a new thermocouple and connect that to your rpi. The new thermocouple should be connected through one of the peep holes with a custom made clay mount. This mount is super-easy to make, just form enough clay around your thermocouple and slightly taper it to fill the peep hole. This would replace one of your other peep plugs.

rondoc, 2022-09-30:
I  respectfully disagree . It should perfectly ok to use same thermocouple,
unless one side of tc is grounded. The adapter boards need isolated tc
wires. Also you need to use thermocouple wires to keep errors to a minimum.

jbruce12000, 2022-10-01:
@rondoc Well, you gotta know the type of the existing thermocouple to do this. You must match your controller to it.

I have never tried this, so you'll have to experiment.

rondoc, 2022-10-01:
Yes, you are right. I did assume same tc in both cases. I have similar
situation using one tc to feed both my Fluke thermometer and your pi
controller. The input impedance of both the Fluke and the adapter board (
xxxx55 type) are very high so dont load the tc. Using separate TC's would
certainly remove a possible problem.
My biggest problem not being  that comfortable with file handling is to
extract just a part of a file that has a fail in it. The log files can be
large.
Ron.