jbruce12000/kiln-controller #44: kiln-tuner.py dosent stop when target temp is reached

felixn-unity, 2021-10-10:
Hi

I just tried to run the kiln-tuner, but is does not stop when max temp is reached.

the compare don't think to work when temp from kiln is is float and taget temp is in int.

I tried to change code like this:

`recordprofile(args.csvfile, float(args.targettemp))`

and seems to fix it, but kiln is not turned off even though code is called:

`oven.output.heat(0)`

felixn-unity, 2021-10-10:
config.py -> simulate = false

and my kiln is being turned on and temparture is getting back correctly

I am running it on a Raspberry PI

Thanks

iog-creator, 2021-10-10:
Interesting. I'd run the simulation for like 5 min and set temperature to 5
degrees or something like that and see if the problem happens in the
simulator. I'm currently running a simulation but it won't be done for like
another several hours. I'll let you know the results if you'd like?

iog-creator, 2021-10-12:
> I don't get it. When I start kiln-tuner SSD is turned on and it starts
> heating. when tempature is reached I can see it calls :
>
> oven.output.heat(0)
> stage = 'cooling'
>
> But SSD is still on and heating just continues
>
> If I start kiln-controller the SSD stops
>
> Felix

Humm... I've got my hardware almost finished. I'll see if it happens to me.