jbruce12000/kiln-controller #53: zero degrees

lurkio55, 2021-10-30:
Hi I would really like to get this working but I'm not getting any temperature reading
I can get the control panel up , have changed values in the config.py but the temperature always reads 0degC

pi3
max31856 with ktype sensor
wiring as per this post - https://github.com/jbruce12000/kiln-controller/issues/40
I would post a log but I dont know how to do that

hope you can help
Tx Lx

iog-creator, 2021-10-30:
I'm having the same issue. I can get it to work by messing with the gpio
states 17 22 and 27. However it never worked twice and the last time it
worked it gave me values all over the place. Since it was working just fine
with no messing with those gpio locations before having this same issue I
can only infer I blew my max31855. I'm going to try ordering a new one. But
I have played with the gpio states in every possible configuration for
those gpio locations and have not been able to get it to work a third time.

iog-creator, 2021-10-31:
Ok fixed it. Download raspberry controller from app store on Android.
Connect to your pi via wireless, you'll have to put your credentials in and
save the profile. Run the kiln program using the terminal icon. Then go to
GPIO which should be the first icon on top. Go to wrench at bottom and
click. Find GPIO 17. Put check in box. Hit back button. It likely says in
and 0 for the value. Make sure you change to out and 1 for value. Changes
are in real time. I didn't realize I was playing with 2 variables not 1.
It's out and 1 on GPIO 17 that worked for me.

Hope it works for you too. 😊

iog-creator, 2021-10-31:
One more thing. I used the example config.py file and plugged into 5v
instead of 3.3. That may have done it as well but you'll need to fix GPIO
17 first. The state will change automatically as it should after you get it
working.

lurkio55, 2021-10-31:
Thanks for your interest iog
unfortunately no joy with your suggestions
I did manage to get some info from terminal after running the start server command
is this any help ? ......
2021-10-31 08:29:57,882 INFO kiln-controller: Starting kiln controller
2021-10-31 08:29:57,886 INFO kiln-controller: this is a real kiln
2021-10-31 08:29:57,887 INFO oven: import MAX31856 
2021-10-31 08:29:57,887 INFO oven: init MAX31856
2021-10-31 08:29:57,897 ERROR oven: Problem reading temp N/C:True GND:False VCC:False ???:True
2021-10-31 08:29:57,902 INFO kiln-controller: listening on 0.0.0.0:8081
Traceback (most recent call last):
  File "./kiln-controller.py", line 277, in <module>
    main()
  File "./kiln-controller.py", line 273, in main
    server.serve_forever()
  File "/home/pi/kiln-controller/venv/lib/python3.7/site-packages/gevent/baseserver.py", line 398, in serve_forever
    self.start()
  File "/home/pi/kiln-controller/venv/lib/python3.7/site-packages/gevent/baseserver.py", line 336, in start
    self.init_socket()
  File "/home/pi/kiln-controller/venv/lib/python3.7/site-packages/gevent/pywsgi.py", line 1545, in init_socket
    StreamServer.init_socket(self)
  File "/home/pi/kiln-controller/venv/lib/python3.7/site-packages/gevent/server.py", line 180, in init_socket
    self.socket = self.get_listener(self.address, self.backlog, self.family)
  File "/home/pi/kiln-controller/venv/lib/python3.7/site-packages/gevent/server.py", line 192, in get_listener
    return _tcp_listener(address, backlog=backlog, reuse_addr=cls.reuse_addr, family=family)
  File "/home/pi/kiln-controller/venv/lib/python3.7/site-packages/gevent/server.py", line 288, in _tcp_listener
    sock.bind(address)
  File "/home/pi/kiln-controller/venv/lib/python3.7/site-packages/gevent/_socketcommon.py", line 563, in bind
    return self._sock.bind(address)
OSError: [Errno 98] Address already in use: ('0.0.0.0', 8081)

iog-creator, 2021-10-31:
Check the ground and power.

2021-10-31 08:29:57,897 ERROR oven: Problem reading temp N/C:True GND:False
VCC:False ???:True

Are you connected to GPIO 17, 23, and 27?

What pins are they plugged into on the max31855? 3VO should have nothing
attached. That's between ground and power. CLK to GPIO 22, CS to GPIO 27
and DO to GPIO 17. Confirm your ground and power.

You also have no ip address configured as the server. It's listening on
0.0.0.0

In the config.py file enter your pi IP address. If it has wireless or
Ethernet connect to your network and change 0.0.0.0 to the pi address.

Then type that into any browser with:8081 next to it.

lurkio55, 2021-10-31:
Hi iog you mentioned the MAX 31855 but I have a MAX31856 the pinouts are very different
I have been trying to follow this guide
https://github.com/jbruce12000/kiln-controller/issues/40

There is 3v supply to the sensor and I can see the web interface 
The temp just reads 0degC 
the IP change doesn't seem to make a difference and I still get the OS error

latest log...

2021-10-31 20:10:03,095 INFO kiln-controller: Starting kiln controller
2021-10-31 20:10:03,099 INFO kiln-controller: this is a real kiln
2021-10-31 20:10:03,099 INFO oven: import MAX31856 
2021-10-31 20:10:03,100 INFO oven: init MAX31856
2021-10-31 20:10:03,113 INFO kiln-controller: listening on 192.168.1.28:8081
Traceback (most recent call last):
  File "./kiln-controller.py", line 277, in <module>
    main()
  File "./kiln-controller.py", line 273, in main
    server.serve_forever()
  File "/home/pi/kiln-controller/venv/lib/python3.7/site-packages/gevent/baseserver.py", line 398, in serve_forever
    self.start()
  File "/home/pi/kiln-controller/venv/lib/python3.7/site-packages/gevent/baseserver.py", line 336, in start
    self.init_socket()
  File "/home/pi/kiln-controller/venv/lib/python3.7/site-packages/gevent/pywsgi.py", line 1545, in init_socket
    StreamServer.init_socket(self)
  File "/home/pi/kiln-controller/venv/lib/python3.7/site-packages/gevent/server.py", line 180, in init_socket
    self.socket = self.get_listener(self.address, self.backlog, self.family)
  File "/home/pi/kiln-controller/venv/lib/python3.7/site-packages/gevent/server.py", line 192, in get_listener
    return _tcp_listener(address, backlog=backlog, reuse_addr=cls.reuse_addr, family=family)
  File "/home/pi/kiln-controller/venv/lib/python3.7/site-packages/gevent/server.py", line 288, in _tcp_listener
    sock.bind(address)
  File "/home/pi/kiln-controller/venv/lib/python3.7/site-packages/gevent/_socketcommon.py", line 563, in bind
    return self._sock.bind(address)
OSError: [Errno 98] Address already in use: ('192.168.1.28', 8081)

jbruce12000, 2021-11-01:
The **Address already in use** error that you are getting happens when you already have an instance of the kiln-controller.py process running... or something else running on port 8081. You can change that port to something else in config.py if you want.

jbruce12000, 2021-11-14:
I'm gonna close this soon due to lack of response. I'm assuming you fixed your issue.

lurkio55, 2021-11-14:
Ah sorry no the issue isn't fixed (real life intervened for a moment)

I'm thinking that maybe I have killed my MAX 13856 board 
Can you confirm that the wiring on post
https://github.com/jbruce12000/kiln-controller/issues/40
is accurate and works
The software seems to work and I can get the webpage up ok but I have a 0dgC temperature reading which doesn't change
Is there anyway of testing the board ?

jbruce12000, 2021-11-15:
I don't own a max-31856, so I can't verify the wiring... but there are others that have gotten it to work.

marktilles, 2021-11-16:
Here is my config for 31856 with an S thermocouple attached. Might this help?

### Outputs
gpio_heat = 23  # Switches zero-cross solid-state-relay

### Thermocouple Adapter selection:
#   max31855 - bitbang SPI interface
#   max31856 - bitbang SPI interface. must specify thermocouple_type.
max31855 = 0
max31856 = 1
# see lib/max31856.py for other thermocouple_type, only applies to max31856
thermocouple_type = MAX31856.MAX31856_S_TYPE

### Thermocouple Connection
gpio_sensor_cs = 5
gpio_sensor_clock = 11
gpio_sensor_data = 9
gpio_sensor_di = 10 # only used with max31856

lurkio55, 2021-11-19:
Thanks Mark for your advice, I tried your pinout etc but no joy - still stuck at 0deg C

I have now got my self a MAX 31855 and wired it as the main instruction page

I get as far as starting the server and i get this error - sigh -

pi@raspberrypi:~/kiln-controller $ source venv/bin/activate; ./kiln-controller.py
2021-11-19 19:04:52,260 INFO kiln-controller: Starting kiln controller
2021-11-19 19:04:52,267 INFO kiln-controller: this is a real kiln
2021-11-19 19:04:52,268 INFO oven: import MAX31855 
2021-11-19 19:04:52,268 INFO oven: init MAX31855
Traceback (most recent call last):
  File "/home/pi/kiln-controller/./kiln-controller.py", line 44, in <module>
    oven = RealOven()
  File "/home/pi/kiln-controller/lib/oven.py", line 381, in __init__
    self.board = Board()
  File "/home/pi/kiln-controller/lib/oven.py", line 47, in __init__
    self.create_temp_sensor()
  File "/home/pi/kiln-controller/lib/oven.py", line 75, in create_temp_sensor
    self.temp_sensor = TempSensorReal()
  File "/home/pi/kiln-controller/lib/oven.py", line 107, in __init__
    from max31855 import MAX31855, MAX31855Error
  File "/home/pi/kiln-controller/lib/max31855.py", line 2, in <module>
    import RPi.GPIO as GPIO
ModuleNotFoundError: No module named 'RPi'

lurkio55, 2021-11-19:
here is the output from the terminal - looks to me like it isn't installing something it needs

`(venv) pi@raspberrypi:~/kiln-controller $ pip install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: setuptools in ./venv/lib/python3.9/site-packages (from -r requirements.txt (line 1)) (59.2.0)
Requirement already satisfied: greenlet in ./venv/lib/python3.9/site-packages (from -r requirements.txt (line 2)) (1.1.2)
Requirement already satisfied: bottle in ./venv/lib/python3.9/site-packages (from -r requirements.txt (line 3)) (0.12.19)
Requirement already satisfied: gevent in ./venv/lib/python3.9/site-packages (from -r requirements.txt (line 4)) (21.8.0)
Requirement already satisfied: gevent-websocket in ./venv/lib/python3.9/site-packages (from -r requirements.txt (line 5)) (0.10.1)
Collecting RPi.GPIO
  Downloading RPi.GPIO-0.7.0.tar.gz (30 kB)
Collecting Adafruit-MAX31855
  Downloading https://www.piwheels.org/simple/adafruit-max31855/Adafruit_MAX31855-1.6.1-py3-none-any.whl (5.5 kB)
Collecting Adafruit-GPIO
  Downloading https://www.piwheels.org/simple/adafruit-gpio/Adafruit_GPIO-1.0.3-py3-none-any.whl (38 kB)
Collecting websocket-client
  Downloading https://www.piwheels.org/simple/websocket-client/websocket_client-1.2.1-py2.py3-none-any.whl (52 kB)
     |████████████████████████████████| 52 kB 289 kB/s 
Collecting adafruit-pureio
  Downloading https://www.piwheels.org/simple/adafruit-pureio/Adafruit_PureIO-1.1.9-py3-none-any.whl (12 kB)
Collecting spidev
  Downloading https://www.piwheels.org/simple/spidev/spidev-3.5-cp39-cp39-linux_armv7l.whl (39 kB)
Requirement already satisfied: zope.interface in ./venv/lib/python3.9/site-packages (from gevent->-r requirements.txt (line 4)) (5.4.0)
Requirement already satisfied: zope.event in ./venv/lib/python3.9/site-packages (from gevent->-r requirements.txt (line 4)) (4.5.0)
Building wheels for collected packages: RPi.GPIO
  Building wheel for RPi.GPIO (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/kiln-controller/venv/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-pw3f01d1/rpi-gpio_318402cf2fd14897b2f1f368c3291035/setup.py'"'"'; __file__='"'"'/tmp/pip-install-pw3f01d1/rpi-gpio_318402cf2fd14897b2f1f368c3291035/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-79u4ebvf
       cwd: /tmp/pip-install-pw3f01d1/rpi-gpio_318402cf2fd14897b2f1f368c3291035/
  Complete output (87 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-3.9
  creating build/lib.linux-armv7l-3.9/RPi
  copying RPi/__init__.py -> build/lib.linux-armv7l-3.9/RPi
  creating build/lib.linux-armv7l-3.9/RPi/GPIO
  copying RPi/GPIO/__init__.py -> build/lib.linux-armv7l-3.9/RPi/GPIO
  running build_ext
  building 'RPi._GPIO' extension
  creating build/temp.linux-armv7l-3.9
  creating build/temp.linux-armv7l-3.9/source
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/pi/kiln-controller/venv/include -I/usr/include/python3.9 -c source/c_gpio.c -o build/temp.linux-armv7l-3.9/source/c_gpio.o
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/pi/kiln-controller/venv/include -I/usr/include/python3.9 -c source/common.c -o build/temp.linux-armv7l-3.9/source/common.o
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/pi/kiln-controller/venv/include -I/usr/include/python3.9 -c source/constants.c -o build/temp.linux-armv7l-3.9/source/constants.o
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/pi/kiln-controller/venv/include -I/usr/include/python3.9 -c source/cpuinfo.c -o build/temp.linux-armv7l-3.9/source/cpuinfo.o
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/pi/kiln-controller/venv/include -I/usr/include/python3.9 -c source/event_ [... cut ...] tall-pw3f01d1/rpi-gpio_318402cf2fd14897b2f1f368c3291035/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-p98hzc0l/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/kiln-controller/venv/include/site/python3.9/RPi.GPIO Check the logs for full command output.
(venv) pi@raspberrypi:~/kiln-controller $`

lurkio55, 2021-11-22:
Hi Jason - thanks for trying to help
Had to start again with a fresh image
working through your instructions the first part seems to go well
I then do the 
"If you want to deploy the code on a PI for production:" 
I can't cd because venv is running so I "deactivate"
it goes well until
pip install -r requirements.txt
after which i get loads of errors which I have copied for you below (click me)

it seems to start with "Building wheel for RPi.GPIO (setup.py) ... error"

I ran the code you pasted above, it seems to confirm python3-dev installed
"pi@raspberrypi:~/kiln-controller $ sudo apt-get install python3-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-dev is already the newest version (3.9.2-3).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded."

but when i reach "pip install -r requirements.txt" I get the same errors again

<details><summary>CLICK ME</summary>
<p>
pi@raspberrypi:~ $ cd kiln-controller
pi@raspberrypi:~/kiln-controller $ virtualenv -p python3 venv
created virtual environment CPython3.9.2.final.0-32 in 974ms
  creator CPython3Posix(dest=/home/pi/kiln-controller/venv, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/pi/.local/share/virtualenv)
    added seed packages: pip==20.3.4, pkg_resources==0.0.0, setuptools==44.1.1, wheel==0.34.2
  activators BashActivator,CShellActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator
pi@raspberrypi:~/kiln-controller $ source venv/bin/activate
(venv) pi@raspberrypi:~/kiln-controller $ pip install --upgrade setuptools
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: setuptools in ./venv/lib/python3.9/site-packages (44.1.1)
Collecting setuptools
  Downloading https://www.piwheels.org/simple/setuptools/setuptools-59.2.0-py3-none-any.whl (952 kB)
     |████████████████████████████████| 952 kB 122 kB/s 
Installing collected packages: setuptools
  Attempting uninstall: setuptools
    Found existing installation: setuptools 44.1.1
    Uninstalling setuptools-44.1.1:
      Successfully uninstalled setuptools-44.1.1
Successfully installed setuptools-59.2.0
(venv) pi@raspberrypi:~/kiln-controller $ pip install greenlet bottle gevent gevent-websocket
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting greenlet
  Downloading https://www.piwheels.org/simple/greenlet/greenlet-1.1.2-cp39-cp39-linux_armv7l.whl (149 kB)
     |████████████████████████████████| 149 kB 862 kB/s 
Collecting bottle
  Downloading https://www.piwheels.org/simple/bottle/bottle-0.12.19-py3-none-any.whl (89 kB)
     |████████████████████████████████| 89 kB 677 kB/s 
Collecting gevent
  Downloading https://www.piwheels.org/simple/gevent/gevent-21.8.0-cp39-cp39-linux_armv7l.whl (5.5 MB)
     |████████████████████████████████| 5.5 MB 881 bytes/s 
Collecting gevent-websocket
  Downloading https://www.piwheels.org/simple/gevent-websocket/gevent_websocket-0.10.1-py3-none-any.whl (22 kB)
Collecting zope.interface
  Downloading https://www.piwheels.org/simple/zope-interface/zope.interface-5.4.0-cp39-cp39-linux_armv7l.whl (242 kB)
     |████████████████████████████████| 242 kB 577 kB/s 
Collecting zope.event
  Downloading https://www.piwheels.org/simple/zope-event/zope.event-4.5.0-py2.py3-none-any.whl (6.8 kB)
Requirement already satisfied: setuptools in ./venv/lib/python3.9/site-packages (from gevent) (59.2.0)
Installing collected packages: zope.interface, zope.event, greenlet, gevent, gevent-websocket, bottle
Successfully installed bottle-0.12.19 gevent-21.8.0 gevent-websocket-0.10.1 greenlet-1.1.2 zope.event-4.5.0 zope.interface-5.4.0
(venv) pi@raspberrypi:~/kiln-controller $ cd kiln-controller
-bash: cd: kiln-controller: No such file or directory
[... 209 lines cut ...]
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/py_gpio.o:/tmp/pip-install-gl_m1hm2/rpi-gpio_90a911dc7b2043adac890683f18d21aa/source/constants.h:28: multiple definition of `input'; build/temp.linux-armv7l-3.9/source/constants.o:/tmp/pip-install-gl_m1hm2/rpi-gpio_90a911dc7b2043adac890683f18d21aa/source/constants.h:28: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/py_gpio.o:/tmp/pip-install-gl_m1hm2/rpi-gpio_90a911dc7b2043adac890683f18d21aa/source/constants.h:27: multiple definition of `low'; build/temp.linux-armv7l-3.9/source/constants.o:/tmp/pip-install-gl_m1hm2/rpi-gpio_90a911dc7b2043adac890683f18d21aa/source/constants.h:27: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/py_gpio.o:/tmp/pip-install-gl_m1hm2/rpi-gpio_90a911dc7b2043adac890683f18d21aa/source/constants.h:26: multiple definition of `high'; build/temp.linux-armv7l-3.9/source/constants.o:/tmp/pip-install-gl_m1hm2/rpi-gpio_90a911dc7b2043adac890683f18d21aa/source/constants.h:26: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/py_pwm.o:/tmp/pip-install-gl_m1hm2/rpi-gpio_90a911dc7b2043adac890683f18d21aa/source/common.h:38: multiple definition of `gpio_direction'; build/temp.linux-armv7l-3.9/source/common.o:/tmp/pip-install-gl_m1hm2/rpi-gpio_90a911dc7b2043adac890683f18d21aa/source/common.h:38: first defined here
[... cut ...] p-record-ibu8iww5/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/kiln-controller/venv/include/site/python3.9/RPi.GPIO Check the logs for full command output.
(venv) pi@raspberrypi:~/kiln-controller $ ls
config.py          kiln-controller.py  lib        requirements.txt  venv
config.py.EXAMPLE  kiln-logger.py      public     start-on-boot
docs               kiln-tuner.py       README.md  storage
(venv) pi@raspberrypi:~/kiln-controller $ 
<p>
</details>

jbruce12000, 2021-11-22:
ok. it sounds like you're going through the instructions all in one go... which is fine, but you can leave some of the stuff out, but you MUST have the virtualenv activated before you do the pip install or all kinds of things will likely go wrong.

```
$ sudo apt-get update
$ sudo apt-get dist-upgrade
$ sudo apt-get install python3-dev python3-virtualenv libevent-dev virtualenv
$ git clone https://github.com/jbruce12000/kiln-controller
$ cd kiln-controller
$ virtualenv -p python3 venv
$ source venv/bin/activate
$ pip install --upgrade setuptools
$ pip install greenlet bottle gevent gevent-websocket
$ pip install -r requirements.txt
```

That should solve the problem.

There is one thing about your post that is not clear to me at all. You said you could not cd because you had the virtualenv activated, so you had to decactivate. I don't really understand why a sourced virtualenv would stop you from running a cd command. but you [for sure] do not want to deactivate the virtualenv. when you do that and then run pip install, python tries to install libraries system-wide [which you likely did not give it rights to do], so all kinds of errors will result.

lurkio55, 2021-11-22:
Hi Jason
I'm following your instructions on your github page line by line in sequence
ie I copy a line paste it into terminal hit enter and wait till it finishes
is that not ok ?
I'm not really sure what is meant by "deploying code for production" but I took that as a necessary step ?

this is my experience of trying to "cd" without deactivating  
I used ls just to check the directory existed
"cd" just would not work unless I deactivated - maybe i was supposed to do something else?

(venv) pi@raspberrypi:~/kiln-controller $ cd kiln-controller
-bash: cd: kiln-controller: No such file or directory
(venv) pi@raspberrypi:~/kiln-controller $ ls
config.py          kiln-controller.py  lib        requirements.txt  venv
config.py.EXAMPLE  kiln-logger.py      public     start-on-boot
docs               kiln-tuner.py       README.md  storage
(venv) pi@raspberrypi:~/kiln-controller $ deactivate
pi@raspberrypi:~/kiln-controller $ 

Ill wipe the SD card again and try your code above

lurkio55, 2021-11-22:
Ok clean install and followed your instructions above
fails at the same place

`Building wheels for collected packages: RPi.GPIO
  Building wheel for RPi.GPIO (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/kiln-controller/venv/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-3btszhku/rpi-gpio_c4f9473cc69d48f9b508f etc etc etc`

jbruce12000, 2021-11-23:
Sorry you're having so much trouble. Keep your head down, keep working on it and we'll fix it together...

The reason you're having trouble with...

`cd kiln-controller`

is because you're already in that directory.

Running all the commands in sequence is not what is intended. There are two use cases these commands try to solve...

1. I just downloaded the software and I want to test it using a simulation
2. I set up my hardware and I'm ready to control a kiln

ok, you're clearly having trouble doing a pip install of RPi.GPIO. See if you can do this instead. You do not need to source the virtual environment, and it does not matter what directory you are in...

```
sudo apt-get update
sudo apt-get install python3-rpi.gpio

```
If the package cannot be found, try this...

```
sudo apt-get update
sudo apt-get install rpi.gpio

```
This will install an older version of rpi.gpio, but it should work just fine for our purposes.

lurkio55, 2021-11-23:
Tx Jason I admire your patience !
It says Rpi.GPIO is already installed but it still fails at module not found when i try to start the server

`(venv) pi@raspberrypi:~/kiln-controller $ sudo apt-get update
Get:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB]    
Get:2 http://archive.raspberrypi.org/debian bullseye InRelease [23.5 kB]       
Get:3 http://archive.raspberrypi.org/debian bullseye/main armhf Packages [242 kB]
Fetched 281 kB in 2s (136 kB/s)   
Reading package lists... Done
(venv) pi@raspberrypi:~/kiln-controller $ sudo apt-get install python3-rpi.gpio
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-rpi.gpio is already the newest version (0.7.0-0.2+b1).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
(venv) pi@raspberrypi:~/kiln-controller $ cp config.py.EXAMPLE config.py
(venv) pi@raspberrypi:~/kiln-controller $ ls
config.py          kiln-controller.py  lib        requirements.txt  venv
config.py.EXAMPLE  kiln-logger.py      public     start-on-boot
docs               kiln-tuner.py       README.md  storage
(venv) pi@raspberrypi:~/kiln-controller $ source venv/bin/activate; ./kiln-controller.py
2021-11-23 16:56:11,370 INFO kiln-controller: Starting kiln controller
2021-11-23 16:56:11,394 INFO kiln-controller: this is a real kiln
2021-11-23 16:56:11,394 INFO oven: import MAX31855 
2021-11-23 16:56:11,395 INFO oven: init MAX31855
Traceback (most recent call last):
  File "/home/pi/kiln-controller/./kiln-controller.py", line 44, in <module>
    oven = RealOven()
  File "/home/pi/kiln-controller/lib/oven.py", line 381, in __init__
    self.board = Board()
  File "/home/pi/kiln-controller/lib/oven.py", line 47, in __init__
    self.create_temp_sensor()
  File "/home/pi/kiln-controller/lib/oven.py", line 75, in create_temp_sensor
    self.temp_sensor = TempSensorReal()
  File "/home/pi/kiln-controller/lib/oven.py", line 107, in __init__
    from max31855 import MAX31855, MAX31855Error
  File "/home/pi/kiln-controller/lib/max31855.py", line 2, in <module>
    import RPi.GPIO as GPIO
ModuleNotFoundError: No module named 'RPi'
(venv) pi@raspberrypi:~/kiln-controller $ sudo apt-get update
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
(venv) pi@raspberrypi:~/kiln-controller $ sudo apt-get install rpi.gpio
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'python3-rpi.gpio' for regex 'rpi.gpio'
Note, selecting 'python-rpi.gpio' for regex 'rpi.gpio'
Note, selecting 'rpi.gpio-common' for regex 'rpi.gpio'
python3-rpi.gpio is already the newest version (0.7.0-0.2+b1).
rpi.gpio-common is already the newest version (0.7.0-0.2+b1).
rpi.gpio-common set to manually installed.
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
(venv) pi@raspberrypi:~/kiln-controller $ source venv/bin/activate; ./kiln-controller.py
2021-11-23 16:57:20,203 INFO kiln-controller: Starting kiln controller
2021-11-23 16:57:20,208 INFO kiln-controller: this is a real kiln
2021-11-23 16:57:20,208 INFO oven: import MAX31855 
2021-11-23 16:57:20,209 INFO oven: init MAX31855
Traceback (most recent call last):
  File "/home/pi/kiln-controller/./kiln-controller.py", line 44, in <module>
    oven = RealOven()
  File "/home/pi/kiln-controller/lib/oven.py", line 381, in __init__
    self.board = Board()
  File "/home/pi/kiln-controller/lib/oven.py", line 47, in __init__
    self.create_temp_sensor()
  File "/home/pi/kiln-controller/lib/oven.py", line 75, in create_temp_sensor
    self.temp_sensor = TempSensorReal()
  File "/home/pi/kiln-controller/lib/oven.py", line 107, in __init__
    from max31855 import MAX31855, MAX31855Error
  File "/home/pi/kiln-controller/lib/max31855.py", line 2, in <module>
    import RPi.GPIO as GPIO
ModuleNotFoundError: No module named 'RPi'
(venv) pi@raspberrypi:~/kiln-controller $ `

lurkio55, 2021-11-23:
It is strange that I didnt have this problem when I was trying to get the MAX 31856 to work - I managed to get the web interface up and start a simulation it was just that the temperature reading was stuck at 0 deg C

jbruce12000, 2021-11-23:
I'm running into nothing other than dead-ends on my side. The last suggestion I found was...

```
$ sudo apt-get update
$ sudo apt-get dist-upgrade
$ sudo apt-get install python3-dev python3-virtualenv libevent-dev virtualenv
$ git clone https://github.com/jbruce12000/kiln-controller
$ cd kiln-controller
$ virtualenv -p python3 venv
$ source venv/bin/activate
$ sudo pip install --upgrade setuptools
$ pip install greenlet bottle gevent gevent-websocket
$ pip install -r requirements.txt
```
The only thing that changed is putting **sudo** in front of the **pip install --upgrade setuptools** on the 3rd line from the end. I am skeptical that this will do anything.

If this does not work, I need a link to the image you downloaded so I can try it out on one of my Raspberry PIs.

lurkio55, 2021-11-23:
Nope no luck with the sudo thing - failed at the same place
Not sure how to provide you with a link to the image ?
I just used the Raspberry pi imager app to download the latest Raspberry OS to a SDcard

jbruce12000, 2021-11-23:
ok. never used rpi-imager. what did you pick?

raspberry pi 32 bit with desktop (full)
raspberry pi 32 bit without desktop (lite)

lurkio55, 2021-11-23:
the full one

jbruce12000, 2021-11-24:
ok, I have a fix. Not sure why it works.

```
export CFLAGS=-fcommon
pip3 install RPi.GPIO

```

lurkio55, 2021-11-24:
Genius - it works !
what was the problem in the end? had something changed in newer versions of Raspberry OS

A couple more things if I may
Once the server is running I can no longer issue commands via the terminal - so how do i pause or stop it if i need to edit config or something

Also I have set it to auto start on boot  - how do I undo this if i need to ? 

many thanks
Lx

iog-creator, 2021-11-24:
Likely it resets the gpio states after something else messed them up. I had
to use a different program to manually adjust the gpio states so I would
get a reading other than zero.

Bryon McCoy
203-989-0875

murraycs, 2021-11-25:
@lurkio55 You should be able to open multiple shell sessions by pressing alt-F1-F12. Here's a good thread about it on Stackexchange: https://raspberrypi.stackexchange.com/questions/7023/open-multiple-terminals-without-gui-startx

jbruce12000, 2021-11-29:
How do I start and stop the process manually and not tie up my shell? see https://www.tecmint.com/linux-process-management/

`source venv/bin/activate; ./kiln-controller.py &`

The & puts the process in the background. To kill it, get the process id:

`ps -ef |grep kiln-controller`

use that process id to kill the controller:

`kill -9 number-from-ps-above`

How do I disable autostart after I have enabled it?

`sudo systemctl disable kiln-controller
`

I'm going to close this. Open another issue if needed.

closed 2021-11-29 by jbruce12000