jbruce12000/kiln-controller #54: Config : "Could not import config file."

steeleh, 2021-11-25:
I have been running an old version using a MAX31855: works OK but I have some 'Short to Vcc' 'Short to ground' blips at high temps so I have built a new version using a MAX31856.

I am having a really strange problem: when I run kiln-controller.py I get the 'Could not import config file.' fault. I have tried deleting the config.py file and reinstating by copying from config.py.EXAMPLE - still no joy. 

OS is a clean full 32 bit Rasp Pi OS using Rasp Pi Imager v1.6

Any ideas?

steeleh, 2021-11-25:
Managed to coax things past the initial error by importing config first but I now get module error as below. Any ideas?

(venv) pi@raspberrypi:~/kiln-controller $ source venv/bin/activate; ./kiln-contr                                                                                       oller.py
2021-11-25 17:49:34,504 INFO kiln-controller: Starting kiln controller
2021-11-25 17:49:34,524 INFO kiln-controller: this is a real kiln
2021-11-25 17:49:34,528 INFO oven: import MAX31856
2021-11-25 17:49:34,531 INFO oven: init MAX31856
Traceback (most recent call last):
  File "/home/pi/kiln-controller/./kiln-controller.py", line 45, in <module>
    oven = RealOven()
  File "/home/pi/kiln-controller/lib/oven.py", line 381, in __init__
    self.board = Board()
  File "/home/pi/kiln-controller/lib/oven.py", line 47, in __init__
    self.create_temp_sensor()
  File "/home/pi/kiln-controller/lib/oven.py", line 75, in create_temp_sensor
    self.temp_sensor = TempSensorReal()
  File "/home/pi/kiln-controller/lib/oven.py", line 120, in __init__
    self.thermocouple = MAX31856(tc_type=config.thermocouple_type,
  File "/home/pi/kiln-controller/lib/max31856.py", line 127, in __init__
    gpio = Adafruit_GPIO.get_platform_gpio()
  File "/home/pi/kiln-controller/venv/lib/python3.9/site-packages/Adafruit_GPIO/                                                                                       GPIO.py", line 417, in get_platform_gpio
    import RPi.GPIO
ModuleNotFoundError: No module named 'RPi'

giorgiodidon, 2021-11-25:
See issue closed #45!

steeleh, 2021-11-26:
The problem was not covered by issue 45. However, the problem was resolved via another site using 'export PYTHONPATH' as in $export PYTHONPATH=/usr/lib/python3/dist-packages

closed 2021-11-26 by steeleh