jbruce12000/kiln-controller #55: No module named 'RPi'

Stonyfield, 2021-12-05:
[image]

I have had this issue since switching from stretch to bullseye. Using rpi 4 fresh install of bullseye. Has issues building gpio wheel when I get to $ pip3 install -r requirements.txt

BlackNet, 2021-12-05:
Gpio has been depreciated. There is a new library to use.

Stonyfield, 2021-12-06:
Thanks for your response.  How do I resolve this?  I'm super new to this and can really only get things working assuming everything works flawlessly.

BlackNet, 2021-12-06:
The code will have to be updated for the new version.

Stonyfield, 2021-12-06:
> 

Got ya.

BlackNet, 2021-12-06:
The changes that will be needed will involve many code changes and not a simple change 2-3 lines of code. So until that happens you should use something like a raspberry 3 with older code, I.e. buster.

Stonyfield, 2021-12-06:
Thanks again. I am running it with the legacy buster os problem free now.

jbruce12000, 2021-12-11:
@BlackNet https://sourceforge.net/projects/raspberry-gpio-python/ does not show that it is deprecated. Where did you see that? What is the replacement library? blinka?

jbruce12000, 2021-12-11:
@Stonyfield I don't have a RPi4, but I'd like to see the output from:

pip3 install -r requirements.txt

I recently changed from using pip to pip3 for another build problem. you might be able to use pip instead of pip3 and it might work fine.

BlackNet, 2021-12-11:
> @BlackNet https://sourceforge.net/projects/raspberry-gpio-python/ does not show that it is deprecated. Where did you see that? What is the replacement library? blinka?

https://github.com/adafruit/Adafruit_Python_GPIO shows deprecated.

jbruce12000, 2021-12-11:
@BlackNet ok, but @Stonyfield has a MAX-31855 according to the log he posted. That doesn't even use the Adafruit GPIO lib. The error he is having is in building RPi.GPIO which has not been deprecated.

I have no interest in upgrading to blinka to replace the Adafruit GPIO lib because it does not yet support linux. It only supports embedded devices and RPi.

BlackNet, 2021-12-14:
This one does support the Raspberry Pi Bullseye.

[https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/raspi-blinka.py](https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/raspi-blinka.py)

BlackNet, 2021-12-14:
RPI.gpio is still in alpha stage to.

**Note that the current release does not support SPI, I2C, hardware PWM or serial functionality on the RPi yet. This is planned for the near future - watch this space! One-wire functionality is also planned.**

[https://pypi.org/project/RPi.GPIO/0.7.1a4/](url)

jbruce12000, 2021-12-14:
@BlackNet I appreciate your interest, but none of that is relevant to this issue. The original issue was that RPi.GPIO could not be installed. You suggested the library was deprecated, which it is not. You suggested to circumvent the issue by going back to buster. This was a successful work-around for installing RPi.GPIO. The same thing can likely be achieved in bullseye by using pip instead of pip3. Is RPi.GPIO alpha? yep. lots of software is alpha. alpha means something different to every developer & team. Does RPi.GPIO support hw pwm or spi or i2c? nope. Do I need it to? nope.

As for the problem you mentioned. The MAX-31856 uses Adafruit_GPIO. That library is deprecated. It has been replaced by blinka. I don't use a MAX-31856, so if you want to rewrite the code to use blinka, you're welcome to fork the project. I have chosen not to use blinka because it would mean losing the ability to do simulations on linux... so no one would be able to test drive the software to see if they even want to consider using it.

steeleh, 2021-12-14:
I had problems installing RPi-GPIO. All attempts produced lots of red text and no install in the venv. The solution was

$ export CFLAGS=-fcommon
$ pip3 install RPi.GPIO

I have no idea why - but it worked for me.

SamSkjord, 2021-12-15:
@steeleh thanks, that solved it for me

jbruce12000, 2021-12-16:
ok, I added...

export CFLAGS=-fcommon

to the installation instructions.

closed 2021-12-16 by jbruce12000