jbruce12000/kiln-controller #56: Contactor adaptation

steeleh, 2021-12-06:
Please bear with me, I am an amateur potter and a Python illiterate.
I am using your Rasp Pi set up to control an old electric kiln with a contactor. Thank you! 

 I originally used a MAX31855 and a kiln-controller branch pre oven.py rewrite. Because I am controlling a contactor (built into the kiln and as a fail-safe) I added a couple of lines within oven.py def set_heat(self, value): (see below) to test the state of pin 23 and to only change the state when needed (so only turn on if now off, only turn off if now on). This stopped unnecessary on/off wear on the contactor and was a lot quieter. 

I am having trouble with MAX31855 Short to Vcc, Short to ground problems at higher temps so I have bought MAX31856 and run it with the latest version of kiln-controller. It works but the poor old contactor now sounds like an Uzi. I think I can get control more binary by reducing window_size in line 477 of oven.py and can increase service cycle but I am struggling to work out where the ON and OFF bit (equivalent to def set_heat(self, value):)  is in the rewritten oven.py so I can add pin 23 state test to reduce the contactor switching. 

Any help gratefully received.

def set_heat(self, value):
        if value > 0:
            self.heat = 1.0
            if gpio_available:
               if config.heater_invert:
                 GPIO.input(config.gpio_heat)==0 # test state of pin 23
#                 time.sleep(self.time_step * value)
                 GPIO.output(config.gpio_heat, GPIO.HIGH)
                 time.sleep(self.time_step * value)
               else:
                 GPIO.input(config.gpio_heat)==1
                 GPIO.output(config.gpio_heat, GPIO.LOW)
                 time.sleep(self.time_step * value)__

jbruce12000, 2021-12-11:
In lib/oven.py see Class RealOven def heat_then_cool. This is what controls how long to keep each of heat and cool active. see the variables heat_on and heat_off.

heat_on = float(self.time_step * pid)
heat_off = float(self.time_step * (1 - pid))

shortly after that, the output is called to either heat or cool.

self.output.heat(heat_on)
self.output.cool(heat_off) 

note that cool does nothing but sleep because there is no active cooling [like a fan]. So everything happens in output.heat:

    def heat(self,sleepfor):
        self.GPIO.output(config.gpio_heat, self.GPIO.HIGH)
        time.sleep(sleepfor)
        self.GPIO.output(config.gpio_heat, self.GPIO.LOW)

wubbington, 2021-12-11:
Perfect: thanks for your help. Very nice project: I owe you a pint

closed 2021-12-14 by steeleh