jbruce12000/kiln-controller #62: Kiln Venting Option

koboldgn, 2022-01-12:
First of all thank you for the code I am very happy to use it.

I need to ventilate the ceramic kiln during firing. 
How can I do this? It would need to be on for several hours.(gpio on) Then turn it off. (gpio off)
So I think an option where you could control a relay based on time or temperature would be useful.
Optional: Editing this in the web interface.

If someone could help me with this, I would be happy to pay. So that others can benefit from it.

Thank you!

marktilles, 2022-01-13:
Hi! I myself intend to add hooks just like this for events at certain temperatures, first in the form of triggers to system events like sendmail or SMS text messages. If the system could just remind me to put my vent plug in at 500C that's good enough for me :) ,,, and then adding GPIO triggers like you suggest would be trivial once the former is implemented. My vision is to have a table one can enter such event triggers in the GUI. But time-based? ,,,  well that seems to me not to be so useful because all kilns have individual heating capacities.

At the current day my friend Marko and I have integrated a scheduler and are considering spinning off our own fork, most of our proposals and mods are information-based and others here might not really see the need to test and integrate those sorts of mods.  But we'll see! Maybe we need more test-volunteers in this project?

Here's a video of my GUI and view of the scheduler being turned on:
  https://youtube.com/watch?v=SfgqH2mQqVg&feature=share

koboldgn, 2022-01-13:
@marktilles 
That sounds good thanks for the reply.
The time would make sense because it would allow you to define the exact gpio connection.
In the case of temperature, it may switch more than once.
In the attached picture you can see that for temperature it would switch twice.

[image]
?usp=sharing

marktilles, 2022-01-13:
True but there is surely a way around that regarding by simply watching whether we are currently on a rise or a decline in temperature. I am just not a fan of time components especially if you might want to do something special at high end because the heating rates can be very different between ovens.

koboldgn, 2022-01-13:
In the meantime, I just found this. On this page, the blue box is probably exactly what we think it is. Does it work? You might know the code better.

https://apollo.open-resource.org/mission:resources:picoreflow (image)

https://github.com/apollo-ng/governess (code)

marktilles, 2022-01-13:
Myself I’m just an amateur hacker/coder, my buddy Marko is the guy that helped me with the actual detailed programming. I come up with the ideas. I’m not sure what you mean by blue box either.

koboldgn, 2022-01-13:
[image]

marktilles, 2022-01-13:
Picoreflow is the very origin of this project, as well as many other forks, the current project is just a different way the table was is shown. The current table has the same data points, what I can see.
I don’t think any features have been removed from the original project… But I could be wrong! I’m no expert on this, just a user and hacker :-)

koboldgn, 2022-01-13:
I could not find the circulator table in the in kiln-controller.  What you see in the previous picture.

marktilles, 2022-01-13:
Oh that must be the cooling option. I have seen some leftover stuff in the code for cooling. Might not be functioning in the current code. But I think I understand what you mean now.

koboldgn, 2022-01-13:
Yes, it would be great if you could help me.

jbruce12000, 2022-02-10:
Hi @koboldgn I use a kiln vent on my kiln. It's on all the time and is not controlled by gpio. If you are worried about the impact the kiln vent has on temp control, don't worry, it'll work just fine and the PID controller will compensate. There is no reason not to have it on continuously that I can think of. The main reasons to use a kiln vent are 1. to evacuate noxious gas from the kiln to somewhere outside 2. to more evenly fire