jbruce12000/kiln-controller #63: pip install gevent

cretzuuu, 2022-01-17:
pip install gevent latest stable release 21.12.0 crashes the deployment in production on the RasPi (at least ZeroW). 
Error given is Illegal Instruction -> seems to not be built for the armv6 processor. 

Solution: replace 21.12.0 with 21.8.0 and everything is up and running

ChipHildreth, 2022-01-26:
If you happen to get this in the next day or so, would mind pointing me to where I can learn how to replace modules with older versions like that?
Thanks for your time.

rondoc, 2022-01-27:
I have followed instructions and I get correct temp reading for a second after opening on browser. Then it reverts to 32F and stay there...HELP!!

rondoc, 2022-02-11:
> @rondoc <https://github.com/rondoc> can you post a log so I can see
> what's happening?

I can do that , how do I get a log file? I am computer savvy but no coder.
Thanks

jbruce12000, 2022-02-12:
by default logs will go to /var/log/daemon.log. you can transfer that file to the computer your using to post issues in github. You can attach files by dragging and dropping them right below your post.

rondoc, 2022-02-12:
Will do that with the 56 board when I remove power and the system doesn't
recover upon power restore. Thank you.

jbruce12000, 2022-02-28:
I don't see any attachments. how did you send the log?

jbruce12000, 2022-03-21:
Is this problem solved? still active?

rondoc, 2022-03-21:
Yes, it is all working. Changed from a 56 breakout board to a 55.  I am now
working on the PID settings. It would be nice if the initial heating from
cold was not so aggressive . For a kiln that has a continually increasing
temp rise I am not sure that a PID control loop is ideal. My manual method
is to start at low power and increase the power to keep up with the
required slope. With this there is no overshoot at start up. I have 16
power levels. I know an enormous amount of work went into kiln-controller
so I don't expect any result from these comments. Unless a coder is
interested... mmmmm.... I am an EE but no coding experience.

Ronald.

jbruce12000, 2022-03-23:
@rondoc you have control over that initial heating. If it is too aggressive, you can change the slope of the profile so it heats more slowly. The overshoot can be controlled by changing the P and D values too: decrease P or increase D. Make changes to one at a time by cutting in half or doubling. test. repeat.

I must admit that configuring PID parameters for the entire temperature range of a kiln is tough [but possible]. If you want quick reactions at high temperatures, you'll have some overshoot at low temps.

If you have more problems, open a new issue.

closed 2022-03-23 by jbruce12000