jbruce12000/kiln-controller #66: gevent-websocket not geventwebsocket?

rondoc, 2022-01-27:
In kiln-controller.py the line import geventwebsocket does not work and it appears that geventwebsocket should be hyphenated. Is this an error or what..please explain? 
Ronald.

jbruce12000, 2022-02-11:
https://github.com/jgelens/gevent-websocket shows the import line should not be hyphenated...

from geventwebsocket import WebSocketServer, WebSocketApplication, Resource

there's got to be something else happening. can you post the log so I can check it out?

rondoc, 2022-02-11:
Yes I will post log when I find out how to do that. I need some help on
that.
Thanks   , Ronald

jbruce12000, 2022-02-12:
by default logs will go to /var/log/daemon.log. you can transfer that file to the computer your using to post issues in github. You can attach files by dragging and dropping them right below your post.

rondoc, 2022-02-12:
daemon.log as of tday with 56 board and lost power symptom, no recovery. 
[daemon.log](https://github.com/jbruce12000/kiln-controller/files/8054348/daemon.log)

closed 2022-02-12 by rondoc