jbruce12000/kiln-controller #67: ERROR oven : problem reading temp.

rondoc, 2022-01-28:
Here is the terminal output after inital ok reading it goes to this
ERROR oven : problem reading temp N/C: False GND: False VCC: False ???: True.
It reads tp until reaches this . Help!

steeleh, 2022-02-04:
What sort of thermocouple amp breakout board are you using? I had lots of same sort of problems with cheap MAX31855/56 boards from China. Part solution was to connect RaspPi GND to the same earth as the kiln earth: I still got intermittent problems at temperatures over 700C. I then bought a 'checked' MAX31856 (UK) from a local supplier. I haven't had any further, similar problems.

rondoc, 2022-02-04:
It is same with both my boards are adafruit. I will hardwire and continue
checking.

SamSkjord, 2022-02-04:
Having similar issues, everything is fine until I get to ~1000c at which point the temperature will start dropping and then fluctuate by 50-100c every few seconds, tried a new thermocouple today with the same issues.
Using an Adafruit 31856 and have another spare I'm trying tomorrow, also on the list to try is twisting the thermocouple wires to reduce noise and grounding the Pi DC gnd to the kiln case

@steeleh am also UK, where did you source yours from?

[image]

rondoc, 2022-02-05:
Get a tc meter , I use a Fluke, connect in parallel to the breakout board.
This will check on tc . I also.put a 1mfd capacitor across the tc wires at
the breakout board. Did not make any difference. I would check all the
connections from the board to the pi. I am going to hardwire and check. The
pin and socket work well if matched but the pi seems to have thinner pins
than my lead wire sockets.
Ronald.

SamSkjord, 2022-02-07:
Tried:
- Twisted thermocouple cable,
- connecting pi gnd to earthed kiln chassis
- 1µF cap between the TC terminals on the 31856 (data sheet says it has one already but worth a try)

Crapped out again at just over 1000c, measured TC voltage at ~9.6mV which is the expected range for that temperature with a type s TC

Remaining ideas:
- try spare 31856 (both are from adafruit)
- shorten signal cables between 31856 and pi
- hardwire rather than jumper leads for signal cables
- cry
- throw kiln in bin
- see doctor, kiln too heavy to lift into bin

rondoc, 2022-02-08:
Ok the 31855 board seems work much better for k thermocouples AND
intermittent opens in the breaker board circuits. I think that the 56 chip
has fault modes that the kiln-controller doesn't address. The 55 chip
doesn't have the same system and if you have a 55 board poweror tc open it
recovers immediately. Unless you use S type TC and I can't think why as K
type are fine for potters temps ,the 55 board would be my choice. I have
wasted many hours trying with the 56 board but could never get a simple
recovery from an open TC. The 55 board does ok under this fail mode. If you
lose the TC connection either board will report low temps. so keep an eye
on your kiln especially when it is near to max setting.

rondoc, 2022-02-08:
Get a K thermocouple they are cheap and reliable.

rondoc, 2022-02-08:
> That's jumped to the top of the list of things to try, will also leave the
> S type and 31856 in place and work on trying to get a decent reading out of
> it in parallel

I have repeatedly disconnected TC with 55 board and it recovers
immediately. The 56 board does not until kiln-controller is run. There
needs to a signal to the 56 board to reset . 55 board doesn't need it.
Needs some code added . Sorry not my forte.

SamSkjord, 2022-02-11:
Well that was an easy fix, K type and the 55 board:
[image]

jbruce12000, 2022-02-11:
That looks mighty fine. Right on the dot. Well tuned it looks like!

closed 2022-02-11 by jbruce12000

SamSkjord, 2022-02-13:
> That looks mighty fine. Right on the dot. Well tuned it looks like!

Cheers, that's with your defaults for PID, gets a bit messy when the ramp is lower and the temps are higher, but still only a few degrees either way:
[image]