jbruce12000/kiln-controller #68: Shutting down prematurely.

secretbadger, 2022-01-31:
Hello,
I wonder if somebody could offer me a bit of guidance. I have this amazing kiln controller fully operational now, it's great. However, I have noticed on a few occasions, long before the cycle has due to complete, suddenly the system stop's and a message comes up suggesting the cycle is completed or done ( I forget the exact working? ) I am wondering if perhaps it might be the fact that the kiln can keep up with the ramp perhaps and just gives up? Would this make the programme prematurely end if it cants catch up?   or might it be some excessive external electrical noise or suchlike?  Any suggestions are warmly received.

Many thanks in advance.  :-)

jbruce12000, 2022-02-10:
even if your kiln cannot keep up with the schedule you define [too aggressive etc], the schedule will continue to run.

can you post the log so I can take a look and see what the problem is?

secretbadger, 2022-02-11:
Thank you.  I will track down the log file and post it.

maabaum, 2022-02-21:
I've had a similar experience with my setup. What seems to happen is the thermocouple read function in oven returns large numbers of errors like:
2022-02-21 19:09:08,852 ERROR oven: Problem reading temp N/C:False GND:True VCC:False ???:False
2022-02-21 19:09:08,904 ERROR oven: Problem reading temp N/C:False GND:True VCC:False ???:False
2022-02-21 19:09:44,003 ERROR oven: Problem reading temp N/C:False GND:True VCC:False ???:False
2022-02-21 19:09:44,054 ERROR oven: Problem reading temp N/C:False GND:True VCC:False ???:False
2022-02-21 19:09:52,161 ERROR oven: Problem reading temp N/C:False GND:True VCC:False ???:False
2022-02-21 19:09:52,212 ERROR oven: Problem reading temp N/C:False GND:True VCC:False ???:False
2022-02-21 19:10:10,041 ERROR oven: Problem reading temp N/C:False GND:True VCC:False ???:False
2022-02-21 19:10:16,627 ERROR oven: Problem reading temp N/C:False GND:True VCC:False ???:False
 and eventually panics with:
emergency!!! too many errors in a short period, shutting down

Oddly, just looking at the trace on the controller interface, it seems to working just fine up to that point. A couple of years back I used an Arduino-based controller (also designed by Mr. Bruce, thanks!) with much the same hardware and I found that it threw many many NaN errors in between valid results, so I just added a little code to essentially ignore them. But probably there's a better answer?

I love this project, by the way. Though I'd like a graceful way to kill things and shut down from the web interface.

rondoc, 2022-02-22:
Please retry with a 55 board. I am convinced that the current code does not
address the fault codes that the 56 chip has. If you goto maxwell and look
up the max31856 chip you will see the code details. May be you can add to
oven.py some code that will reset the 56 chip after a fault.
The 55 chip has no such code problems and with a K tc works very smoothly.
I have deliberately shorted the tc, opened the tc ,  and also lost power to
see if the system recovers. With the 56 it never recovers and always with
the 55..
Ronald.

secretbadger, 2022-02-22:
I have been using the 55 board. (Gave up on the 56 board a while back)
Unfortunately I am struggling to identify the log file for the occasion I had my issue. (Have done a re install since then)
Will post a log file if I have the same experience on the next firing.
Credit to all involved in this great project.
Thanks to all 👍

maabaum, 2022-02-22:
The system I referenced above also uses the 55 board.
maab

maabaum, 2022-02-23:
I'd like to amplify my earlier message about my experience with this project, as it _may_ be relevant to secretbadger's issue. I have my version of the pi controller hooked up to a k-type thermocouple through the Adafruit-style MX31855 board.

When I start a run of the kiln-controller.py program, I get something like this
`2022-02-23 14:44:08,478 INFO kiln-controller: Starting kiln controller
2022-02-23 14:44:08,486 INFO kiln-controller: this is a real kiln
2022-02-23 14:44:08,486 INFO oven: import MAX31855 
2022-02-23 14:44:08,487 INFO oven: init MAX31855
2022-02-23 14:44:08,498 INFO kiln-controller: listening on 192.168.1.12:8081
2022-02-23 14:44:09,497 ERROR oven: Problem reading temp N/C:False GND:True VCC:False ???:False
2022-02-23 14:44:10,881 ERROR oven: Problem reading temp N/C:False GND:True VCC:False ???:False
2022-02-23 14:44:17,043 ERROR oven: Problem reading temp N/C:False GND:True VCC:False ???:False
2022-02-23 14:44:17,673 ERROR oven: Problem reading temp N/C:False GND:True VCC:False ???:False
2022-02-23 14:44:19,686 ERROR oven: Problem reading temp N/C:False GND:True VCC:False ???:False
2022-02-23 14:44:20,317 ERROR oven: Problem reading temp N/C:False GND:True VCC:False ???:False
2022-02-23 14:44:22,330 ERROR oven: Problem reading temp N/C:False GND:True VCC:False ???:False
... usw.
`

You'll note that it seems to be complaining about a short-to-ground error. I've inspected the circuitry closely, and haven't yet found anything likely to cause that. But more to the point, I wrote a short python script that basically calls the same MAX31855 library used by kiln-controller.py and simply asks what the current temperature is. The call to get() is wrapped in a try...except structure to check for error conditions. It never seems to have a problem reading the thermocouple, and the results jibe with an external thermometer reading. All this leads me to wonder if there's some subtle flaw somewhere in the oven object code.

maab

maabaum, 2022-02-23:
Okay, imma modify the above comment. It occurred to me that the error condition might happen only now and then, so I changed my test script to more explicitly check error conditions, especially the short-to-ground. Then I ran it multiple times as fast as a bash script could loop it. It threw a short-to-ground error 4 times out of 20. While apparently correctly reporting the temperature at the same time. 

So not a fault with Mr. Bruce's code. Which didn't seem all that likely anyway. But what is going on here? The rpi and circuit boards and such are just setting there on a table in a sheltered environment. The chip and/or thermocouple is not just casually developing a ground short and then losing it in the space of a fraction of a second? Would appreciate help from someone more clever than I.

maab

rondoc, 2022-02-23:
How are you connecting the breakout board with 55 chip to the pi?  This
does seem more like a hardware problem than a code issue.
The pi pins are slightly smaller than most so you might just be getting a
poor connection. If you have a scope it might be worth monitoring the
connections. Don't you just hate intermittent problems!

maabaum, 2022-02-23:
I take your point. At the moment, this is just breadboarded together using those arduino-style jumper cables, so not exactly NASA-grade hardware. 

But what I keep coming back to is how randomly consistent it is. Bear in mind, this is switching between no-error and one-specific-error pretty damn quickly. Most recent test run was 50 trials over probably less than a second. 8/50 positive error cases. IN EVERY CASE, the error code value is 65536, that is, 2^16. And the temperature read was accurate to the extent that I can check it. 

That just doesn't say "bad connection" to me, but as you will have divined, I'm much more a software guy than a hardware guy. Sigh.

maab

maabaum, 2022-02-23:
I felt I wasn't being sufficiently appreciative of rondoc's question about hardware, particularly since I KNEW that the jumper cable I had on the MAX31855 board's ground pin was suspiciously wobbly. :<

So I replaced it with a nice, new, positive-fit cable that was nice and tight. The next 50-iteration test returned error code 65536 7 out of 50 times.

Still on a breadboard, of course. :(

maabaum, 2022-02-28:
Thanks for the comment, Jason. On my limited sampling of two chips it seems to be an endemic problem with the MAX31855 board with no obvious explanation. In my experience, anyway, temperature doesn't seem to have a lot to do with it. I get lots of these ground-short errors apparently randomly at room temperature. When I used one on the Arduino version of your kiln controller it manifested in my logs as a 'NaN' error, and I ultimately modified the script to ignore them, since the temperature readings seemed correct.

I use a reasonably decent thermocouple sold by the Shenzhen Zhongjin conglomerate. Stainless housing, steel shielding for the length of the cable. And as you say, once I set honour_theromocouple_short_errors to False (memo to self: RTFM) everything seems to work. I'm using this on a glass fusing kiln, so it doesn't get to really really high temperatures. I really appreciate your work on this!

Don't know if ANY of this applies to secretbadger's original problem. Sigh.

secretbadger, 2022-03-08:
Good afternoon,
after much fiddling and remaking, I got around to doing a firing last night but sadly the same failure happened.

I have a totally new set-up. New thermocouple, quality twisted pair thermocouple connecting wire, ferrite beads on the end of the thermocouple wire, decoupling capacitors, new Pi, latest software install and all industry best practices regarding the physical installation etc.
11 hours into the program at 910C (just before 19:05) it stopped with the 'Run Completed' sign coming up.

I quickly created a fresh program to take the firing from the 910c to the final 1220C and restart the kiln which it did OK. However this wasn't before I noticed on the section where you enter how many minutes each stage should run for, it has changed to second, not minutes as per the configuration. The ‘Ramp’ box also failed to populate.

I left it running again, it reached 1010C and failed again with ‘Run Completed’ displayed at approx 20:00 7 March 2021.
This time I rebooted the, created a fresh program to take me to the final desired temperature. This time the time section had reverted to Minutes and the Ramp boxes were populated. All was normal.

It failed again sometime later, this time in frustration I bypassed
[Glaze Fire P3.txt](https://github.com/jbruce12000/kiln-controller/files/8207737/Glaze.Fire.P3.txt)
[config.txt](https://github.com/jbruce12000/kiln-controller/files/8207738/config.txt)
 the controller and connected 5V directly to the SSR and ran it for the last hour at 100% to complete the glazing. This was probably around 23:00 & March 2021 so the log won't have any relevance from this point.

At around midnight 00:00 8th March I decided to cut the power to the kiln but leave the Pi running to it could remotely observe the cooling temperature and it is still cooling. This evening we get to see the end product.  

I have attached the log file, the congig.py and the program json file and hopefully, somebody might be able to interpret what is going on.

Gratefully yours.
Andrew

[daemon.txt](https://github.com/jbruce12000/kiln-controller/files/8207742/daemon.txt)
[Glaze Fire P3.txt](https://github.com/jbruce12000/kiln-controller/files/8207745/Glaze.Fire.P3.txt)
[config.txt](https://github.com/jbruce12000/kiln-controller/files/8207747/config.txt)

secretbadger, 2022-03-08:
sorry, omitted to mention that this firing started at  Mar  7 05:44:35

Mar  7 05:44:35 raspberrypi python[389]: 2022-03-07 05:44:35,026 INFO kiln-controller: RUN command received

secretbadger, 2022-03-11:
Very interesting suggestion. We are using the official Pi power pack and previously had used a large 12v lead acid battery with a quality buck converter, had the same issue. 

I will convert an old ATX power supply and make a 5v power supply. This I would imagine is the most economic way to get a gold plated quality power supply.  Will keep you all posted.

I have uploaded my log file but I don't really know what I am looking for, perhaps somebody more experienced might be able to look at the log file.

jbruce12000, 2022-03-21:
I see, you bypassed the software and hard-wired the SSR. That explains that. Seems like you just need a setting that prevents stopping of a schedule.

jbruce12000, 2022-03-21:
check out my notes for the run:

https://docs.google.com/spreadsheets/d/1v6weY4lO3XpZcDY8i9Eeti4qzRUmo_Y5_2wBJFWUmig/edit#gid=904527049

The two big problems I see, I will restate:

1. your kiln has trouble maintaining temp after 980c. could be lots of causes. old elements, overactive kiln vent, knobs not set to high etc
2. failure to read thermocouple is causing run to stop [and things would probably be fine if this were fixed]

One other thing, you do not need to write a new profile when a profile stops. Using the API you can restart a schedule at any point. You can look in the logs, see where it died and restart it there. see https://github.com/jbruce12000/kiln-controller/blob/master/docs/api.md

jbruce12000, 2022-03-21:
ok, I added the ignore_emergencies setting to config.py. pull the latest master and set that value to True and try another run. Needless to say, you should NOT leave your kiln unattended... even after you're sure this setting works as intended.

jbruce12000, 2022-03-23:
btw, the errors that you are seeing are common for the max-31855... see https://forums.adafruit.com/viewtopic.php?f=31&t=169135#p827564

jbruce12000, 2022-05-27:
I suggest upgrading to the latest version. I've made many changes over the last month. The way the controller works has changed as follows:

1. if the kiln is outside a specific window, by default 5 degrees below or above the set point, the controller acts as a binary switch - either 100% heating or cooling until back inside the window.
2. inside the window, PID control takes over.

- no integral accumulates outside the window.
- if kiln_must_catch_up = True, that means that the schedule is paused any time the temperature is outside the window. It stays paused until it gets back inside the window.
- you define the size of the window, so you have control

My personal firings and those of other users has been good over the past few weeks. Lastly, the logs have changed, so I can more easily troubleshoot problems. If you have a problem, I'll ask you to run a script to zip up the logs... then send them to me so I can check them out.

closed 2022-05-27 by jbruce12000