jbruce12000/kiln-controller #69: No module named RPi.GPIO

captaincaden, 2022-01-31:
After running source venv/bin/activate; ./kiln-controller.py I get the following responses, and can't get any further. 

I'm using 31856, and RPi.GPIO is uncommented in the requirements.txt file. 

2022-01-31 14:30:45,271 INFO kiln-controller: Starting kiln controller
2022-01-31 14:30:45,277 INFO kiln-controller: this is a real kiln
2022-01-31 14:30:45,277 INFO oven: import MAX31856 
2022-01-31 14:30:45,278 INFO oven: init MAX31856
Traceback (most recent call last):
  File "/home/pi/kiln-controller/./kiln-controller.py", line 44, in <module>
    oven = RealOven()
  File "/home/pi/kiln-controller/lib/oven.py", line 381, in __init__
    self.board = Board()
  File "/home/pi/kiln-controller/lib/oven.py", line 47, in __init__
    self.create_temp_sensor()
  File "/home/pi/kiln-controller/lib/oven.py", line 75, in create_temp_sensor
    self.temp_sensor = TempSensorReal()
  File "/home/pi/kiln-controller/lib/oven.py", line 120, in __init__
    self.thermocouple = MAX31856(tc_type=config.thermocouple_type,
  File "/home/pi/kiln-controller/lib/max31856.py", line 127, in __init__
    gpio = Adafruit_GPIO.get_platform_gpio()
  File "/home/pi/kiln-controller/venv/lib/python3.9/site-packages/Adafruit_GPIO/GPIO.py", line 417, in get_platform_gpio
    import RPi.GPIO
ModuleNotFoundError: No module named 'RPi'

**This may be another issue,** or it may just not be something I understand, but how do you stop it from automatically starting on boot once you've already told it to do so? If anyone could help, that'd be great. Thanks!

captaincaden, 2022-01-31:
Corrected it by using pip install rpi.gpio after export cflags=-fcommon

closed 2022-01-31 by captaincaden

steeleh, 2022-01-31:
First I am no expert at all so caveat emptor. I had the same type of 'No module' problems that resulted from me installing modules in places that kiln-controller.py was not finding them. The solution I found was to find where the 'not found' module had been installed  and then either re-install using pip3 or use PYTHONPATH to point to the 'missing' module.

captaincaden, 2022-01-31:
I think that may have been the issue on my end too, but it worked after using 

$ export cflags=-fcommon
$ pip install -r requirements.txt

after doing the installation of bottle, gevent, and websocket in the original string of commands when installing it.