jbruce12000/kiln-controller #71: Problem reading temp.

captaincaden, 2022-02-01:
I'm noticing when the terminal is open and the program is running, but there isn't a websocket open, It's showing the following intermittently when there isn't a client opened. 

2022-01-31 19:48:01,063 ERROR oven: Problem reading temp N/C:False GND:False VCC:False ???:True

Is this normal? The temperature reads normally when I do have a client open, and no errors are given other than when I close out of a websocket. Other than that error being written, it seems like everything is running fine. 

Mind you, I'm just using this to monitor temperature at the moment, there is no control happening for right now.

rondoc, 2022-02-01:
I bet you're loosing a GND to the breakout board.
If so you might have to reboot. Then run the kiln-controller.py then open
browser 0.0.0.0:8081 .
If you want to check the TC put timeout in the config file to False. Then
do a run without turning kiln power on. The tc output should read the kiln
temp without any significant changes.

captaincaden, 2022-02-01:
@rondoc the thermocouple is behaving as expected, every time I've looked at it during this firing. I just opened the remote desktop to view if anything was happening in the terminal on the pi now that it was cooling (sensor temp was reading normally during the firing, and is still behaving as expected on cooldown) but had noticed the intermittent fault in the terminal there, when I didn't have any :8081 windows open anywhere. When I do have one of the :8081 windows open in a browser, everything behaves normally, and the temperature is as it is expected to be behaving, and there aren't any visible errors written in the teminal.

rondoc, 2022-02-01:
I don't know enough about the server function to know what's occurring, but
I have had a sudden unexplained server inactive warning. sorry not much
help but maybe someone else can help here.

captaincaden, 2022-02-03:
Closing because it doesn't seem to be an consistent issue, don't see this error coming up much now, and continues to read normally, even when this error is given.

closed 2022-02-03 by captaincaden