jbruce12000/kiln-controller #72: (not an issue) Text message updates.

captaincaden, 2022-02-04:
Alright, so I can get kiln-controller to run, and I can also create a script using smtplib to send an email or text message. Anyone have any idea how I could combine the two, to make the kiln controller send an email alert at specific temperature values? I'm thinking it would have to be in the oven.py file, but I'm not sure how to word it or figure it out.

jbruce12000, 2022-02-10:
I would change the api to return a json blob of the current state of the kiln. This state is sent to web pages every N seconds, so it would be a pretty easy thing to add.

See kiln-controller.py starting at line 51.

Once you make this change, you could use curl to grab the temperature and send a text when you want.

closed 2022-02-10 by jbruce12000

captaincaden, 2022-02-10:
While that makes sense in my head, I'm not familiar enough to know how to write it/word into the code, and my other attempts have failed, so I'll have to do more research first.