jbruce12000/kiln-controller #75: Making changes into kiln-controller.py or oven.py stops start_on_boot

captaincaden, 2022-02-18:
So I've made some changes, adding some personalization for some other functions in oven.py, as well as in kiln-controller.py, but now I can't get the start on boot function to work. The code I've added and changed works great when I run it manually from source venv/bin/activate; ./kiln-controller.py but wont start on boot, and I'm not sure where to start to correct that. 

Just to note, it did start on boot prior to adding the additional functions, but it won't do so now with them added. For the time being, I'm just running it manually, but would like to have the start on boot function back. 

Any ideas?

rondoc, 2022-02-18:
Remove your changes and test. If ok add back change one at a time. What are
yr changes?

captaincaden, 2022-02-18:
I did remove the changes, and it worked normally without them, but even just adding back one section (all of what I added relies on other functions, so I can't just put it back one at a time, otherwise it will error) of it took away the start on boot. 

I've added code to trigger an smtp server to send an email when certain parameters are met within the loop. In this case, temperature, and the emergency shut down function. Again, all of these changes work and tested great when I manually start it, but it won't start on boot with the existing start on boot procedure for some reason.

captaincaden, 2022-02-18:
@rondoc I'm using the 31856, and don't have any issues relative to the thermocouple. I also use K type, but when I was first testing it away from the kiln I used a J type and it worked as well. 

I just restarted and looked in the daemon log, confirming that it still didn't boot the program, but the dameon log shows that it started kiln-controller without any errors around it, but it definitely didn't start, or at least I can't access the websocket until I manually go in cd kiln-controller source venv/bin/activate; ./kiln-controller.py

rondoc, 2022-02-18:
If you have a 55 board I would try it. It is simpler than the 56 that has
some fault settings on it. Sorry that's about it from my skill knowledge
level.

jbruce12000, 2022-02-28:
To verify, you can manually start the process and everything works fine... but once configured under systemctl, it says it starts in the daemon.log, but does not?

post the output of...

`sudo systemctl -l status kiln-controller
`
restart the service by...

`sudo systemctl restart kiln-controller
`

captaincaden, 2022-02-28:
@jbruce12000 correct. Runs perfectly, including the changes I made, but just won't start unter systemctl, and daemon.log shows that it started, and there's no errors anywhere around it in the log, but the server is still dead when I try to open it from any other device, until I manually start the process. 

Seeing the status, I'm thinking the problem is with the smtplib module for some reason not allowing it to start on boot. This module is what I'm using to send emails at certain temperature variables. 

I don't exactly mind the manual starting of the process, but I'm also thinking of the many reasons why it also would be desirable to start on boot. Now knowing the issue, I at least know what's causing it. 

```
kiln-controller.service - kiln-controller
     Loaded: loaded (/etc/systemd/system/kiln-controller.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2022-02-26 12:18:31 EST; 2 days ago
    Process: 336 ExecStart=/home/pi/kiln-controller/venv/bin/python /home/pi/kiln-controller/kiln-controller.py (code=exited, status=1/FAILURE)
   Main PID: 336 (code=exited, status=1/FAILURE)
        CPU: 1.214s

Feb 26 12:18:31 raspberrypi python[336]:   File "/usr/lib/python3.9/smtplib.py", line 310, in _get_socket
Feb 26 12:18:31 raspberrypi python[336]:     return socket.create_connection((host, port), timeout,
Feb 26 12:18:31 raspberrypi python[336]:   File "/usr/lib/python3.9/socket.py", line 822, in create_connection
Feb 26 12:18:31 raspberrypi python[336]:     for res in getaddrinfo(host, port, 0, SOCK_STREAM):
Feb 26 12:18:31 raspberrypi python[336]:   File "/usr/lib/python3.9/socket.py", line 953, in getaddrinfo
Feb 26 12:18:31 raspberrypi python[336]:     for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
Feb 26 12:18:31 raspberrypi python[336]: socket.gaierror: [Errno -3] Temporary failure in name resolution
Feb 26 12:18:31 raspberrypi systemd[1]: kiln-controller.service: Main process exited, code=exited, status=1/FAILURE
Feb 26 12:18:31 raspberrypi systemd[1]: kiln-controller.service: Failed with result 'exit-code'.
Feb 26 12:18:31 raspberrypi systemd[1]: kiln-controller.service: Consumed 1.214s CPU time.

```

Upon using `sudo systemctl restart kiln-controller` the service seems to start without incidence, but I'm not exactly certain if my additions to the code are running. The smtp server logs in at the beginning of oven.py and displays an additional log of "smtp server log in successful" (if it's successful) in the initial string of "starting kiln-controller, listening on 0.0.0.0:8081" etc.etc. etc. which when I used the systemctl restart, and the got the systemctl -l status again, it did not include that additional log. and I don't really have a way to test for certain if the additional codes are functioning like this or not. 

**Upon full reboot of the pi:** 
The same error result as above is printed in status, and the service does not restart. But, again, manually restarting the process with `cd kiln-controller, source venv/bin/activate; ./kiln-controller.py` runs the program without issue.

captaincaden, 2022-02-28:
I'm wondering if it's somehow an issue of it trying to run and connect to the smtp server before the Pi is fully connected to the internet. But then I think, the internet has to be established for the kiln-controller to connect to the websocket, right?

jbruce12000, 2022-03-01:
@captaincaden I'm not sure where your smtp server is running, but the error is consistent with inability to resolve the hostname of that smtp server. is it running on localhost on port 25? are you using 127.0.0.1? is the smtp server starting before the kiln-controller [because it must]?

captaincaden, 2022-03-01:
I call to import the SMTP server at the end of the import lines in oven.py, and then prior to anything else in the oven.py file is where the server retrieves its sign in data from config.py, and then connects, logs in to the server. This all happens in oven.py because I also have it written to send emails when certain failures occur. 

After importing in oven.py, the following lines are run first, along with the code for retrieving the sender_email/password from the config.py file. 
`server = smtplib.SMTP('smtp.gmail.com',  587)
server.starttls()
server.login(sender_email, password)
print ("Login to notification server successful.")`

and then later in the code when it calls to look for the failures, it runs the `server.sendmail` along with the rest of the information needed to send it if the failure is present. 

Realistically, I probably don't need to login to the server until the condition is met, but this was how I wrote it when testing the additions in a virtual machine (without being able to test start-on-boot) and it worked well in testing, so I implemented it like this. 

I'm not sure if it would change to call to start the server at the time of the condition instead of calling it in the beginning of the oven.py file? Or is the issue when it imports it?

captaincaden, 2022-03-01:
It now fails with: 
` Loaded: loaded (/etc/systemd/system/kiln-controller.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Tue 2022-03-01 11:33:12 EST; 1min 28s ago
    Process: 339 ExecStart=/home/pi/kiln-controller/venv/bin/python /home/pi/kiln-controller/kiln-controller.py (code=exited, status=1/FAILURE)
   Main PID: 339 (code=exited, status=1/FAILURE)
        CPU: 1.186s

Mar 01 11:33:12 raspberrypi python[339]:   File "/usr/lib/python3.9/smtplib.py", line 310, in _get_socket
Mar 01 11:33:12 raspberrypi python[339]:     return socket.create_connection((host, port), timeout,
Mar 01 11:33:12 raspberrypi python[339]:   File "/usr/lib/python3.9/socket.py", line 843, in create_connection
Mar 01 11:33:12 raspberrypi python[339]:     raise err
Mar 01 11:33:12 raspberrypi python[339]:   File "/usr/lib/python3.9/socket.py", line 831, in create_connection
Mar 01 11:33:12 raspberrypi python[339]:     sock.connect(sa)
Mar 01 11:33:12 raspberrypi python[339]: OSError: [Errno 101] Network is unreachable
Mar 01 11:33:12 raspberrypi systemd[1]: kiln-controller.service: Main process exited, code=exited, status=1/FAILURE
Mar 01 11:33:12 raspberrypi systemd[1]: kiln-controller.service: Failed with result 'exit-code'.
Mar 01 11:33:12 raspberrypi systemd[1]: kiln-controller.service: Consumed 1.186s CPU time.
`

also files is listed before dns in hosts > /etc/nsswitch.conf 

So this confirms that the issue is that when it's calling for the server, the network isn't started yet? Brainstorming how to resolve that issue. 

Also, do I keep the added IP of smtp.gmail in /etc/hosts or remove it?

captaincaden, 2022-03-01:
Even though I'm sure I'm the only one who had this specific issue, I'll post how I corrected it:

I commented out the smtp.starttls() at the beginning of oven.py, and instead just gave the variables to retrieve the information from the config.py file at the beginning, and then added the smtp.starttls() plus related codes to each function for sending emails instead. This means that it starts and logs in to the notification server each time, and only when the conditions in the other functions are met. 

After testing in a vm, and then transferring all over to my pi with my real kiln, and performing tests, everything seems to be functioning as expected, both the additions and the normal process, as well as it is starting on boot and returning no errors when it runs. 

This was all done with simulated values, so I'll see how it works when my kiln actually runs later this week, but I can't imagine it be different than my original manual starting of the process, except that I'm able to start on boot again, and the server gets logged in at each event, instead of in the beginning. 

Thank you again @jbruce12000 for your help with this issue, and for all you did with this project!

closed 2022-03-01 by captaincaden

jbruce12000, 2022-03-02:
@captaincaden yep, you want to remove the IP from /etc/hosts. Otherwise, if google ever changes it, your software will no longer send email.

yep, the problem was networking was not started at the time the kiln-controller.py process started.