jbruce12000/kiln-controller #76: "Could not import config file"

lumonic, 2022-02-23:
Getting this when I am trying to launch the server. I get this with either the default config.py example content copied over or the version where I have customized for my GPIO pins. 

Any ideas?

`pi@raspberrypi:~/kiln-controller $ source venv/bin/activate; ./kiln-controller.py
Could not import config file.
Copy config.py.EXAMPLE to config.py and adapt it for your setup.`

SamSkjord, 2022-02-23:
I've had this a few times, did you use `cp config.py.EXAMPLE config.py` to create the new config file?

lumonic, 2022-02-23:
Yes, I did

captaincaden, 2022-02-24:
@lumonic run the following lines in terminal.

`$ cd kiln-controller
$ source venv/bin/activate
$ export CFLAGS=-fcommon
$ pip install -r requirements.txt`

then try again. If that doesn't solve it, recheck all your values in the config.py file, and make sure your simulation status is the correct status for your situation.

lumonic, 2022-02-24:
`Successfully installed Adafruit-GPIO-1.0.3 Adafruit-MAX31855-1.6.1 RPi.GPIO-0.7.1 adafruit-pureio-1.1.9 spidev-3.5 websocket-client-1.2.3
`

Yep, that was it, none of the libraries got installed. Working now. Thanks!

jbruce12000, 2022-02-28:
nice job @captaincaden thx for helping.

closed 2022-02-28 by jbruce12000