jbruce12000/kiln-controller #77: Doesnt pich up upon cooling down

Hansuitherlev, 2022-03-03:
Dear Jason, still we are using hour kiln controller a lot. Just installed it on kiln nr 2, however we noticed a major issue. When running a profile where a peak temperature is followed by a cooling down to a certain temperature the program doesn t pick up when reaching that temperature; it just keeps on cooling. 

The old version of the software dis not have this issue. What to do?

An example of a Profile could be

0-5 min       30oC
5-60 min     increase to 715 0C
60-60 min   cool down to 350 oC
60-120 min  350 oC

Normally the kiln would hit the 350 oC after roughly 80-100 min and then continue at 350 0C until 120 min.
Currently the machine doesn't pick up at 350 but keeps on cooling.

On test runs I notice that the software "shifts to a different scheme" when the real temp curve is more than around 10 oC different from the programmed target curve. And then things go wrong

jbruce12000, 2022-03-03:
in order to help figure out the issue, I need you to post a few things:

- config.py so I can see your configuration
- contents of /var/log/daemon.log for a run so I can see all the log messages
- contents of the profile you're using (this one doesn't matter so much since you posted the profile above, but might be useful)

jbruce12000, 2022-03-03:
I'd guess that you have the following settings in config.py:

```
kiln_must_catch_up = True
kiln_must_catch_up_max_error = 10

```
but I can't visualize how these two would cause the problem you're seeing. That's why I need the logs.

Hansuitherlev, 2022-03-03:
Thx Jason, I’ll try to dig them up. But playing with these settings might already solve the issues I imagine. Maybe increase max error to say 50 ? Sendt fra Mail til Windows Fra: Jason BruceSendt: 3. marts 2022 15:30Til: jbruce12000/kiln-controllerCc: Hansuitherlev; AuthorEmne: Re: [jbruce12000/kiln-controller] Doesnt pich up upon cooling down (Issue #77) I'd guess that you have the following settings in config.py:kiln_must_catch_up = Truekiln_must_catch_up_max_error = 10 but I can't visualize how these two would cause the problem you're seeing. That's why I need the logs.

jbruce12000, 2022-03-03:
I think you've run into a unique bug and I'd like you to test something to prove my theory. I want you to change the profile to look like this...

0-5 min 30
5-60 min increase to 715
60-120 min cool down to 350 (this is where the change is)
120-180 min hold at 350

Instead of making the cool down period just one minute, I've made it much larger. It takes your kiln 80 to 100 mins to cool from 715 to 350, so changing the schedule per above will not impact the run. I think this larger cooling window will fix your problem.

Hansuitherlev, 2022-03-03:
Actually you’re right; I measured the cooling time and made a profile where i programmed that cooling time (think it was 60-100 min cool Down. Indeed it did pick up (though with a weird Little dip at he transition point. Problem is that with glass art cooling down needs to go as fast as possible and that cooling Down time differs a lot between an empty kiln and a totally filled kiln. In my number one kiln controller (with the previous version of the software i pressume) this was not an issue at all.    Sendt fra Mail til Windows Fra: Jason BruceSendt: 3. marts 2022 15:39Til: jbruce12000/kiln-controllerCc: Hansuitherlev; AuthorEmne: Re: [jbruce12000/kiln-controller] Doesnt pich up upon cooling down (Issue #77) I think you've run into a unique bug and I'd like you to test something to prove my theory. I want you to change the profile to look like this...0-5 min 305-60 min increase to 71560-120 min cool down to 350 (this is where the change is)120-180 min hold at 350Instead of making the cool down period just one minute, I've made it much larger. It takes your kiln 80 to 100 mins to cool from 715 to 350, so changing the schedule per above will not impact the run. I think this larger cooling window will fix your problem.

jbruce12000, 2022-03-03:
ok. as a work-around, I'd set the cool down to be closer to reality... set it to five minutes under what max cooling could possibly be for an empty kiln... maybe 30 minutes if max cooling rate dictated 35 minutes.

jbruce12000, 2022-03-03:
This is going to be a tough bug to troubleshoot and fix. It is related to the short window and code execution time. Code execution time differs on every rpi/machine on which this runs. If this is indeed the problem, I'll have to think about how to fix it.

maabaum, 2022-03-06:
Unfortunately not a unique bug, as I ran in to it two days ago. Hansuitherlev has a good point in that with art glass, the instruction to cool a couple hundreds of degrees "as rapidly as possible" is very common. If you're working with fairly primitive kiln hardware -- mine, as an example -- there aren't a lot of options beyond turning the heat off and waiting, but the time is quite variable.

I think I could dig out logs and such if that were helpful, but it sounds like you have  a handle on it? I was kind of wondering about some sort of higher-level instruction overriding the normal PID loop that basically meant "turn the heat off -- wait for temp=target -- resume program"?

Hansuitherlev, 2022-03-06:
The interesting thing is that it did work well in the previous version ?? of the software. 

But great to see some combined great brains delving into this. I ll try to get some log files as soon as I m home again. 

Best regards,
Hans van den Brink

jbruce12000, 2022-03-07:
@Hansuitherlev The time-shifting code has not changed since March of 2021... so not recently changed.

Well, I'm stewing about how to fix it. Until then, make the cooling window as large as you can to avoid the bug.

Hansuitherlev, 2022-03-08:
> you can expect to see stair steps like this... the smaller a value for kiln_must_catch_up_max_error, the more steps you'll see.

Hej Jason,

Sorry to say but we re not home yet. I keep on meeting the same issue. 

Yesterday I implemented the changes you suggested. Today I ran a trial profile. Heating up fast to 550 and then cooling rapidly to 300. The temperature drops way below the 300 target and then sticks around 260. With the test “kiln must catch up, too cold, shifting schedule”

Starting at runtime 2761, temp 258.09 oC at 10:09
At 10:15 the temp was 265 with the same message

L

Best regards,
Hans van den Brink

Hansuitherlev, 2022-03-08:
Would you have the old version of the software (say before march 21? ). That one worked flawless. 

Best regards,
Hans van den Brink

jbruce12000, 2022-03-08:
Can you post the log from the last run you did? There is something else happening here that I do not understand.

jbruce12000, 2022-03-08:
The change that was made last March was a complete re-write. If you wish to go back to the old version the day before that change, do this:

1. change to the directory where you have the code checked out
2. git checkout 5bd8c28388a868f06c6dfd916c732c9cfb639008

you'll need to move config.py to another directory temporarily and then move it back after the checkout... or you can look up how to use **git stash**.

Hansuitherlev, 2022-03-08:
Thabks a great deal for your engagement and willingness to help. Highly appreciated!!

Best regards,
Hans van den Brink

Hansuitherlev, 2022-03-08:
Where do I find a log file?

A simple screen dump only shows the last period

Best regards,
Hans van den Brink

jbruce12000, 2022-03-08:
logs go to /var/log/daemon.log...

https://github.com/jbruce12000/kiln-controller/blob/master/docs/logs.md

Hansuitherlev, 2022-03-09:
Hej Jason, Uploaded the log file with we transfer. Not sure if it is useful at all. BrHans https://we.tl/t-cv6pKM4ERG Sendt fra Mail til Windows Fra: Jason BruceSendt: 8. marts 2022 16:13Til: jbruce12000/kiln-controllerCc: Hansuitherlev; MentionEmne: Re: [jbruce12000/kiln-controller] Doesnt pich up upon cooling down (Issue #77) logs go to /var/log/daemon.log...https://github.com/jbruce12000/kiln-controller/blob/master/docs/logs.md

jbruce12000, 2022-03-10:
downloaded and looked at the log. it has no output from the kiln-controller. I have no idea where your output is being sent. In order to troubleshoot this problem, I gotta have the log. Try this...

```
cd /var/log
grep -c "INFO oven" * 2>/dev/null |more

```
That will give you a count of lines that match "INFO oven" for every file in /var/log. If there is one file in that directory that has A LOT more than any of the others, that is where stuff is being logged.

maabaum, 2022-03-12:
Mr. Bruce, I have a (hopefully quick) question about the kiln-controller code when you have a moment. In the module oven.py at line 424.ish, when a Profile object is created, the profile data in the form of a list of lists gets sorted as it is passed to the new object. I'm rather out of my depth here, but I imagine that the sorted function is defaulting to putting the sublists in numerical order by the first element in each one? Basically time goes forward relentlessly?

Since the profiles would be created in order anyway, and seem to be stored that way without change, is this action just a sort of belt-and-suspenders kind of safety measure, or is there same important reason to run the sort?

Tnx,
maab

jbruce12000, 2022-03-13:
@maabaum this is a safety measure to put the data in time order. It happens only at init [so when each profile loads]. Run python interactively at the command line to see it in action:

```
python3
>>> a = {"data": [[0, 200], [2200,325], [1740, 200], [1800, 250], [3600, 250]], "type": "profile", "name": "test-200-250"}
>>> sorted(a["data"])
[[0, 200], [1740, 200], [1800, 250], [2200, 325], [3600, 250]]
>>> 

```

jbruce12000, 2022-04-14:
Closing this due to lack of response. I know you tried really hard to post the logs, but I have yet to get the log output from the kiln controller process. if you need help, lmk.

closed 2022-04-14 by jbruce12000

Hansuitherlev, 2022-04-14:
Hej Jason,
Sorry but indeed I couldn t get the data out. However, we got a workable solution by changing the config.sys file so it no longer uses this setup with more then 10 oC difference between target and real. Works fine now. However, when goung down it will undershoot with 10-15 before reaching a plateau as programmed. Not a big deal 

Best regards,
Hans van den Brink

jbruce12000, 2022-04-15:
I'm glad you have a working solution. It's not clear to me exactly what you're doing though.

Hansuitherlev, 2022-04-15:
I understand that it is unclear. Can not remember the command in the config.sys where i changed the setting from true til false. It s the one involved in this shift of programs when the deviation is more then 10 oC. 

Best regards,
Hans van den Brink

jbruce12000, 2022-04-15:
I see... changed config.py setting **kiln_must_catch_up** from True to False.

Hansuitherlev, 2022-04-15:
Correct. Like I said. It solved the main problem but now it undershoots when coolibg down to a temperature. Didn t do that before. 

Best regards,
Hans van den Brink

maabaum, 2022-04-15:
Jason --
I'm sorry about the lagitude, but things have been busy. I don't know if you'd want to reopen issue 77, but I can now send you some data. I have been trying to do a so-called "contour firing" of Spectrum 96 glass. Spectrum's guidance for this includes one segment where the kiln is supposed to cool about 500 degF "as fast as possible",  after which it holds for about an hour, presumably to anneal. This was where I first encountered this issue of the controller allowing the steep cooling slope, but then just continuing to cool down and never continuing with the balance of the firing schedule.

Based on discussions here, I decided to try and extend the rapid cooling period to get a longer, less abrupt slope. I settled on one based on some previous trials that looks like this: 
{"type": "profile", "data": [[0, 60], [14280, 1050], [16080, 1050], [18960, 1250], [20160, 1250], [22260, 1425], [22560, 1425], [24060, 950], [27600, 950], [30300, 800], [30900, 800], [38400, 100]], "name": "Contour Test A"}

To test it out, I ran an empty kiln with a modified version of the profile, "Contour C", with the temperatures and times halved so that the slopes of the profile were more or less the same, but it ran faster at lower temperatures.

The attached files include the log for the Contour C test run and three charts that show the results. In the charts you will see that after the rapid cooling segment the controller overshoots the cooling by, oh, 35 degF or so, then seems to catch itself and continue without too much trouble (though the same thing happens in a later, less critical, segment.)

Thinking I had a sort-of-kind-of solution, I tried a for-real contour firing at the proper temperatures and times. The second set of log data and charts shows the results. My scaling experiment obviously didn't work as well as desired because the controller never seem to recover from the cooling segment, and the kiln just kept getting colder and colder. I tried using the API feature to halt the firing run and pick it up again past the failure point, but that didn't work, probably because I had let it get far too cold.

I've also attached the config.py file I was using.

Hope this all means something to you.

Michael Baum

[Contour_Firing_Charts.zip](https://github.com/jbruce12000/kiln-controller/files/8497550/Contour_Firing_Charts.zip)
[config.py.txt](https://github.com/jbruce12000/kiln-controller/files/8497885/config.py.txt)
[Contour_Firing_Real_kiln.zip](https://github.com/jbruce12000/kiln-controller/files/8497887/Contour_Firing_Real_kiln.zip)
[Contour_Firing_Test_kiln.zip](https://github.com/jbruce12000/kiln-controller/files/8497888/Contour_Firing_Test_kiln.zip)

Hansuitherlev, 2022-04-16:
Hi Michael and Bruce, pls keep me in the loop. 
The kiln project is a great tool for glass artists; my wife is enjoying it almost daily (not good with current energy pricing 🤨). However, there are some improvements possible. The key issue is the “cool as fast as possible and then retain temperature x for a certain time”. 
We solved it partially but now it undershoots the set temp by 10 oC (which can be toommuch) before hitting the set temp. 

Jason; great project and we have been using it so much the last couple of years. Thanks a lot for it. Hope you ll be able to optimize this part as well!!!

Best regards,
Hans van den Brink

maabaum, 2022-04-19:
This is a further example of some possibly-not-desired behavior of the controller in trying to track certain profiles. This test run used a quick, low-temp test profile: {"type": "profile", "data": [[1200, 400], [1800, 400], [2400, 500], [2700, 500], [3000, 300], [4500, 90]], "name": "test"}

One issue here -- to date, when I've set up firing profile I've always started at time 0 and an approximately room-temperature starting point, but it occurred to me that what I really want to tell the kiln is, "I don't care what your current temperature is, I want you to be at 400 degrees in 20 minutes," sort of thing. So I wrote a profile with a first entry of [1200, 400]. 

When I ran this, I was surprised that the first temperature data from the controller was temp=64.40, target=512.73. I have no idea where the 512.73 would come from. If you look at the first detail of the chart you see that the controller shoots straight up to 512 and oscillates madly for a few minutes before settling down and trying to get to 400 degF, which is where it's supposed to be. 

Another odd thing is shown in Detail B, the final cooling ramp. Notice that the controller tracks with the cooling slope fine until it gets to within about 5 degF of the target, and then it seems to stop trying to get to the target temperature and simply tracks it about 5 degrees too hot. Not that it really matters, but odd.

Sincerely,
Michael Baum

[image]
[config.py.txt](https://github.com/jbruce12000/kiln-controller/files/8516093/config.py.txt)
[image]
[image]
[image]

Hansuitherlev, 2022-04-20:
Hej Michael,

Have seen the behaviour of example B once. Somehow it cooled down following the target curve but abput 10 oC too hot. We solved it by defining a slower cool down. This was after we adapted the config.sys. 
We do see some oscillation but that is typically plus/minus 1 oC. More than with the old program from some years back though. 

Best regards,
Hans van den Brink

jbruce12000, 2022-04-25:
There is a lot in this thread to digest. I want to try to handle each problem, one at a time. Manually changing or creating a profile to start somewhere other than 0 seconds is probably going to surface all kinds of bugs & problems. This is not something I ever expected anyone to do:

{"type": "profile", "data": [[1200, 400], [1800, 400], [2400, 500], [2700, 500], [3000, 300], [4500, 90]], "name": "test"}

change it to something like:

{"type": "profile", "data": [[0,65], [1200, 400], [1800, 400], [2400, 500], [2700, 500], [3000, 300], [4500, 90]], "name": "test"}

That way it starts at near ambient temperature and has a defined, specific slope to get to 400 [even if that slope is steep and fast that is ok]. I could check for this condition in code [no zero time entry found] and just set a temperature, but I'm not sure what other problems that may cause.

jbruce12000, 2022-04-25:
As for the oscillations, this is a sure sign that your PID parameters are not perfect [not critically damped]. In this case, you don't have enough derivative.

When you change PID parameters, change only one. Cut it in half or double it. Keep iteratively testing a single change at a time until your happy with the outcome. Note that no set of PID parameters will be perfect at all temperatures. Make sure the parameters work well in the temperature range that matters to you.

jbruce12000, 2022-04-25:
As for the cool-down with the temperature tracking 5 degrees above set point... My guess is that you have the following settings:

kiln_must_catch_up = True
kiln_must_catch_up_max_error = 5 #degrees

What is happening is the kiln is not cooling as quickly as the cooling slope, so the schedule waits for the kiln to catch up. As soon as the temperature gets within 5 degrees [above the set point in this case], the software tells the coils to heat, they heat [way more than is needed] and BAM you go above 5 degrees. rinse. repeat. What needs to happen here is the system should not heat as much.

So, how do you fix this? I think in this case, the problem is too much Proportional. Start by halving that value. Test. The goal is to get the kiln to heat less aggressively.

https://github.com/jbruce12000/kiln-controller/blob/master/docs/pid_tuning.md

Just for my knowledge, is this kiln small? Seems like your elements cause heating in your kiln much more quickly than mine. The default PID tuning parameters [as set in the master branch] will almost certainly NOT work for you.

jbruce12000, 2022-04-25:
@Hansuitherlev 

So during a cooling phase of your profile, it is constantly 10 degrees C too hot? This is very similar to @maabaum problem. Can you open an issue and post the logs from a run so I can check it out? I expect the same thing is happening to you that is happening to @maabaum whereby you've got this stair step of waiting while cooling [kiln catching up and profile waits] and then BAM it heats up and goes outside of the range defined by:

kiln_must_catch_up_max_error = 10 ???

which is probably set to 10 degrees in your case.

maabaum, 2022-04-25:
Thank you, Jason, I'll try those things and see what happens. The PID
settings I've been using were just the ones produced by that tuning
assistant program, but I haven't yet tried refining them. I take your point
about the 5-degree offset, that looks very much like what is happening.

Yes, this is a really small oven, it's an antique JenKen kiln they call a
"Bead Annealer" <https://jenkenkilns.com/gs-beadannealer.aspx> with a
single heating element in the wall.

Thanks for all your help.

Michael Baum

Hansuitherlev, 2022-04-26:
Michael, just to keep you in the loop. Got this firing profile today. Probably due tp a too fast ramp up we get this weird parallel curve. Anything you recognize?

Best regards,
Hans van den Brink

Hansuitherlev, 2022-04-27:
> Hans --
> I'm sorry, but for some reason I can't see your firing data. I'm not particularly skilled at GitHub, I'm afraid.
>
> Jason understands the PID loop far better than I. I'm trying to adjust my parameters per his suggestions. The thing that most puzzles me at the moment is the data that I originally mentioned a few days ago, a contour firing of some Spectrum glass where the kiln greatly overran the floor of the cooling segment and seemed to just keep cooling. (See graph, maybe.)
>
> I stopped this firing at about the 450 minute mark because it seemed clear that the controller was never going to try to level the temperature out at that annealing plateau. Unlike the previous chart, this one shows the "real time" for the firing as opposed to the runtime values from the kiln log, which keep getting adjusted by the kiln-must-catch-up routine. I think this shows more clearly what is happening.
>
> Michael

Yes, this behaviour I did see a lot. I solved it bumy changing the config.py file line 123. I changed kiln_must_catch_up from “true” to “false”. 
Now we typically will see a short undershoot of some degrees when reaching the plateau level after which it will catch up. 

Attached a screen dump from our wrong going firing yesterday

Best
Hans van den Brink

Hansuitherlev, 2022-04-27:
Correct. The max error is put to 10. But the kiln must catch up is set at false

Best regards,
Hans van den Brink

maabaum, 2022-04-27:
The problem with setting kiln_must_catch_up to false is that the controller then does not compensate for segments of the profile that take much longer than expected. This probably does not affect you professionals, :)  but me with my little toy kiln, I get problems as demonstrated in this chart. The broken green line on the left represents the actual firing profile used by the controller. It finishes in under an hour. The blue and orange lines reflect actual performance. Because the cooling segments take much longer than desired, the actual run takes more than six hours, but without the catch-up adjustment the controller will stop everything when it is only about 15 % complete.

[I _still_ don't see the images you post. I must have some setting incorrect. :( ]

[image]

rondoc, 2022-04-27:
I looked at your profile. If the kiln is well insulated then the slope when
power is off will be very gradual as predicted by thermodynamics. For glass
work or metallurgy you need a cooling system or a very inefficient oven.

Ronald

Hansuitherlev, 2022-04-28:
Hi Ronald,

You re right of course. However, we ve been using this kiln for quite some years and only ran into problems when our pi broke down and we had to replace it with a new one, including new software. Before that it would simply cool down at its own pace (not hitting the set curve if the cooling was set to steep) and pick up when actual hit set temp. Now things live their own life, making firjng curves unpredictable. 

Best regards,
Hans van den Brink

jbruce12000, 2022-04-28:
Hey guys. I committed to fixing the problem with **kiln_must_catch_up** when it is set to **true**. I made some changes a month back that stopped the time slipping that caused this bug @maabaum mentioned... so I thought it was fixed. I'm going to have to run this on my own kiln and see if the problem is fixed or persists.

Hansuitherlev, 2022-04-28:
You’re a hero :-)

Best regards,
Hans van den Brink

rondoc, 2022-04-29:
May be a different issue but during a long glaze firing my kiln stopped. I
saved the terminal output from kiln-controller.py.
What does this mean and what can I do to fix it?
Thanks
  stop glaze.txt
<https://drive.google.com/file/d/1sVR7PQx2AETOCXytlPro1utMZAEwFuhd/view?usp=drivesdk>

jbruce12000, 2022-05-01:
requested access to that file. no clue what caused it without seeing the log.

maybe a power outage. maybe a hardware failure. maybe a bug.

rondoc, 2022-05-01:
Do you need daemon.log ?

jbruce12000, 2022-05-01:
Yep, that's log I need.... but I think I figured out what this problem is and I'm testing a fix. In each case, both of you have asked your kilns/ovens to cool down faster than they physically can. This is causing accumulation of LOTS of integral. What this looks like is temperature goes down WAY below your set point and it will not heat because of this integral accumulation... This behavior will continue, and it will stay WAY below the set point for as long as it takes to zero out that integral accumulation.

The change I'm testing is that, if you are using kiln_must_catch_up=True, any time the limit is hit [either too hot or too cold], the integral is set to zero.

rondoc, 2022-05-01:
Thanks but in my case this was not so. I was in an eas y part of heating.
I'll find the daemon.log,  try and grab just the part before and after
where it stopped,  and send it. After restarting it went to cone10 ok with
nothing else changed except the profile. BTW  I have follow up set to false.
I do so like the app and hope sudden stops are not common. It would be
useful when the stop occurs if an alarm hi could be sent to one of the
spare io pi pins. I could then make lights flash or bells ring etc. I can
then continue my firing on manual control if necessary. Having a firing
suddenly stopping is, as you must know, a major hassle.
Thanks.
Ronald.

jbruce12000, 2022-05-01:
I've never had a firing just stop... unless my kiln sitter flipped. I'll look at the logs if you post them.

rondoc, 2022-05-01:
I looked at var/daemon.log but it doesn't have any kiln-controller stuff.
Is there another log file I should use?

maabaum, 2022-05-01:
rondoc: This is kind of a nasty off-brand hack, but what I did was modify kiln-controller.py somewhere about line 27 thusly:

`# maab 20220303: rearranged and added code here to write log files'
'# to the 'storage/logs' folder of the installation also note the'
'# hard-coded EST time zone on account of lazy'

script_dir = os.path.dirname(os.path.realpath(__file__))
sys.path.insert(0, script_dir + '/lib/')
profile_path = os.path.join(script_dir, "storage", "profiles")
log_path = os.path.join(script_dir, "storage", "logs")

tz = datetime.timezone(-datetime.timedelta(hours=5),name='EST')
control_start_str = datetime.datetime.now(tz).strftime('%Y%m%d-%H%M%S')
log_file = log_path + '/' + control_start_str + '_kiln.log'

logging.basicConfig(filename=log_file, filemode='a', level=config.log_level, format=config.log_format)'
'# logging.basicConfig(level=config.log_level, format=config.log_format)'
'# maab 20220303: end of changes`

You have to create a logs subdirectory under 'storage', and you have to add an `import datetime` line with the other imports. Then the kiln-controller script redirects the log  to a timestamped file that's comparatively easy to find. YMMV

maabaum, 2022-05-01:
jbruce, I love that solution a lot (if it works). Neat and clean.

jbruce12000, 2022-05-01:
for the log, grep for the word kiln in the /var/log folder...

```
cd /var/log
grep -c kiln *
```

that will return a bunch of files with the number of matching lines next to each. you're looking for the one with the highest count.

jbruce12000, 2022-05-01:
[image]

ok, tested the fix and it works. I started at 200F, then had a steep ramp to 150F which was much faster than my kiln can cool. It took 39 mins to reach 150. It went below the setpoint by only a few degrees and continued control smoothly after that. There was no time slippage during the profile and it ran as expected.

Before the fix, it went past 150 and continued downward (because of the accumulated integral) until the integral began to go back to zero, and then heating began again in the 120s. So it looked as if the controller was doing nothing.

jbruce12000, 2022-05-01:
ok, changes committed. test it @maabaum and let me know after your next firing if this is indeed fixed.

make sure to set:

```
kiln_must_catch_up=True
kiln_must_catch_up_max_error = 5 #degrees or whatever you like

```

maabaum, 2022-05-01:
Will do, thanks!!!

Hansuitherlev, 2022-05-02:
This sounds like a very logic explanation. Looking forward to hear if it works

Best regards,
Hans van den Brink

jbruce12000, 2022-05-02:
@Hansuitherlev you should test this too if you use...

`kiln_must_catch_up=True`

and gimme feedback.

jbruce12000, 2022-05-02:
@rondoc I downloaded the shared google doc...

2022-05-02 15:04:01,020 INFO kiln-controller: GET command received
2022-05-02 15:04:28,034 ERROR kiln-controller: Connection is already closed
2022-05-02 15:04:28,035 INFO kiln-controller: websocket (control) closed
2022-05-02 15:04:28,035 INFO kiln-controller: websocket (storage) closed
2022-05-02 15:04:28,036 INFO kiln-controller: websocket (status) closed
2022-05-02 15:04:28,036 INFO kiln-controller: websocket (config) closed
2022-05-02 15:04:28,979 ERROR ovenWatcher: could not write to socket <geventwebsocket.websocket.WebSocket object at 0x7f5a10497220>

are totally normal. This is exactly the set of messages I got when my web browser connected to the server, and then I clicked X to close the browser window. The server was still up-and-running. 

So I need more logs. I need something that shows the server's death [and this is not it].

[29 more replies]