jbruce12000/kiln-controller #78: PID Profile

rondoc, 2022-03-14:
Trying to set up PID for first time. I read process overview but there are some steps I don't get. In ## Step2 it says to run animated. gif..how does this help to get 
the zb zn.profile.csv as it says I need to access the 3 parameters. 
Does anyone have a clearer way of using this process.  Help please.
Thanks 
Ronald.

jbruce12000, 2022-03-14:
Try the default parameters that are set in config.py first. In that file, they are the variables/values that look like this:

pid_kp = 25   # Proportional
pid_ki = 200  # Integral
pid_kd = 200  # Derivative

If these don't work for you, you have two paths. One is to manually tune PID parameters following the instructions https://github.com/jbruce12000/kiln-controller/blob/master/docs/pid_tuning.md which contains an animated gif at the top of the page. This animated gif helps to illustrate the knowledge you'll need to do the tuning. You need to know each parameter's purpose.

The other path is autotuning which is a process where you heat your kiln to 400F and the autotuner will spit out some values that you can substitute for the defaults listed above. When I tried the autotuner, they were kinda close, but not perfect.

jbruce12000, 2022-04-03:
@rondoc open another issue if you're still having trouble tuning

closed 2022-04-03 by jbruce12000