jbruce12000/kiln-controller #79: Cannot connect local

rondoc, 2022-03-23:
I was just updating the PID numbers in config.py. I then saved it and the rebooted. Then when trying to reconnect I entered the 0.0.0.0.:8081 as I have done dozens of times before but it says connection refused.. I thought that when saving the config.py I corrupted something but cannot find what.
Help please!

jbruce12000, 2022-03-24:
@rondoc can you post the log so I can check it out?

Connection refused means that there is nothing listening on that port. It's probably a bug due to the change I made yesterday.

jbruce12000, 2022-03-24:
try adding the following line to your config.py at the end...

`ignore_emergencies = False`

That'll probably fix it.

rondoc, 2022-03-24:
Thanks. Will try today.
I did not know that the app can be changed by internet access. I'd like to
just get a working system and then not having internet access. I just want
a kiln controller to be stable.
Ronald.

jbruce12000, 2022-03-24:
It cannot be changed by internet access. I assumed you did a `git pull` to pull in the latest master branch. If you did not, then post the log so I can check it out and see where things went belly up.

To give you piece of mind, my kiln is internet connected and has not been updated in a while. It works fine on the older version.

rondoc, 2022-03-24:
What log file? I am not sure  where to find it.

rondoc, 2022-03-24:
> Anger is usually regrettable, kindness rarely.

I tried to email the daemon.log file but it was too large to send. Can I
delete conents then try to connect and send an updated file?
If not,  what? Details please.
Thanks.
Ronald

rondoc, 2022-03-24:
Update: I reinstalled pi and kiln controller . It all works now. It seemed
a waste of your time to pursue my problem. Thanks. You can close this issue
unless you want to pursue it. I have the failed system on a Microsoft card
just in case!

jbruce12000, 2022-03-25:
next time, because daemon.log is so big, you can make it a smaller by capturing only the output from the kiln-controller process...

`grep "INFO oven" /var/log/daemon.log > smaller-file.txt`

glad you got it fixed!

closed 2022-03-25 by jbruce12000