jbruce12000/kiln-controller #80: C OR F display ?

rondoc, 2022-03-28:
While running a firing the display on the local pi shows the "C" symbol although reading F. At the same time if accessed via the local network it shows "F" . Although not a problem for me it seems it might be for those who use "C" . Also I worry that this might mean the PID calculations are not based on F or C correctly.
Is it just a display coding problem? Or what?

jbruce12000, 2022-03-28:
in the max31855 lib, which is the one you use I think, everything in the thermocouple code is done in degrees c. if you choose the units as f, a conversion is done each time the temperature is read.

When I change the temp_scale setting and restart the kiln-controller.py process I immediately see an "f" or a "c" in my browser (latest firefox). So, I can't replicate the behavior you're seeing.

The temp_scale doesn't matter in the PID calculations.

ok, so you're using "F" as your setting. what web browser are you using on the PI? because your setting is showing up correctly everywhere but there... right?

jbruce12000, 2022-04-14:
Closing this due to lack of response. Open a new issue if you need.

closed 2022-04-14 by jbruce12000