jbruce12000/kiln-controller #81: Kiln-tuner set up.

rondoc, 2022-04-22:
I am trying to use the kiln-tuner.
I have read the kiln profile pid sheet of info but need some guidance. When I enter python kiln-tuner.py recordprofile zn.csv I get this " could not import config.file etc.. ad nothing happens.
Is there something in the config file that needs to be different from when I do a normal run? 
Help..in simply terms please.
Thanks 
Ronald.
..

jbruce12000, 2022-04-24:
Hey @rondoc. Try the following and post the output if it does not work...

change to the kiln-controller directory, source the virtual environment, run the tuner command...

```
cd kiln-controller
source venv/bin/activate
python kiln-tuner.py recordprofile zn.csv

```

rondoc, 2022-04-25:
All is well. After juggling the zn to a new file called zn.csv . I ran the kiln-tuner and got a PID that kinda works. Thanks.

closed 2022-04-25 by rondoc

rondoc, 2022-04-25:
> thanks, here's what I got after it did a run. what you gave exactly did
> not run but it did come back with
> a remark of what it "should" be as:
>
> python kiln-tuner.py {recordprofile,zn}
>
>  so I tried that and it ran..
>
> I might have slightly misremembered but it did run Ok.
> I now have the run on the terminal screen along with this:
>
>
> venv) ***@***.***:~/kiln-controller $ python kiln-tuner.py zn zn.csv
> Traceback (most recent call last):
>   File "/home/pi/kiln-controller/kiln-tuner.py", line 201, in <module>
>     calculate(args.csvfile, args.tangentdivisor, args.showplot)
>   File "/home/pi/kiln-controller/kiln-tuner.py", line 117, in calculate
>     with open(filename) as f:
> FileNotFoundError: [Errno 2] No such file or directory: 'zn.csv'

I have success and closed this. Thanks.

jbruce12000, 2022-04-26:
great news. glad you got the auto tuner to work. In my experience, it produces a good starting point. It still requires iterative improvement.

rondoc, 2022-04-26:
I did a bisque firing with the PID from the tuner and it tracked within a
degree and no overshoot. Can't want anything better.
It would be nice to have the ability to change the time scale from seconds
to minutes or hours when setting up a profile. Is there a way to do that?
Thanks.
Ronald.

jbruce12000, 2022-04-28:
That is some great news. Congrats on getting the PID tuned for your kiln.

See **time_scale_profile** in config.py. It is set to **m** for minutes by default.

I have not tested this setting much because it was never necessary. You might find bugs.

rondoc, 2022-10-11:
thanks, here's what I got after it did a run. what you gave exactly did not
run but it did come back with
a remark of what it "should" be as:

python kiln-tuner.py {recordprofile,zn}

 so I tried that and it ran..

I might have slightly misremembered but it did run Ok.
I now have the run on the terminal screen along with this:

venv) ***@***.***:~/kiln-controller $ python kiln-tuner.py zn zn.csv
Traceback (most recent call last):
  File "/home/pi/kiln-controller/kiln-tuner.py", line 201, in <module>
    calculate(args.csvfile, args.tangentdivisor, args.showplot)
  File "/home/pi/kiln-controller/kiln-tuner.py", line 117, in calculate
    with open(filename) as f:
FileNotFoundError: [Errno 2] No such file or directory: 'zn.csv'

jbruce12000, 2022-10-11:
instructions here https://github.com/jbruce12000/kiln-controller/blob/master/docs/ziegler_tuning.md show

`python kiln-tuner.py recordprofile zn.csv
`
and then after that completes...

`python kiln-tuner.py zn zn.csv`

The error you posted above tells me that the file zn.csv was NOT created in the first step. check the directory and see what files were created... 

`ls -la`