jbruce12000/kiln-controller #82: Could not import config file

umfan92, 2022-05-04:
I followed all the steps and I keep getting the same message whenever I try to start the server:

"Could not import config file.
Copy config.py.EXAMPLE to config.py and adapt it for your setup."

I'm not sure what I'm missing. I did modify the config.py.EXAMPLE file and copy it to the config.py. I verified that the config.py file is correct. 

Can you help me out?

umfan92, 2022-05-04:
Ok I got to the point where the "source venv/bin/activate: ./kiln-controller.py" command actually works. 

I was able to get the server opened on another computer on the network and got correct values of the thermocouple. 

The only issue I'm having now is getting it to start on boot. It's giving me "No such file of directory" even though I can see it in the folder. Even running the terminal from the folder and it doesn't seem to be able to find the file.

closed 2022-05-04 by umfan92

jbruce12000, 2022-05-04:
Yep. You'll have to modify the path in both of these:

```
start-on-boot
lib/init/kiln-controller.service

```
from the command line, you can test if it is working by doing:

`sudo service kiln-controller restart`

rondoc, 2022-10-11:
Jumping in here.  You need to make sure config.py will run without errors.
You do this by opening it in an editor and then running it. If ok it will
simply do nothing. If not ok you'll get an error which should be easily
fixed. Then you can close config.py then execute kiln-controller in a
terminal. It will show in terminal window waiting for 127.0.0.1:8081.
If so then the server will allow you to access the app in a browser.  127
etc is for local and if you want to access via a pc and network you need
the address of your pi..something like 192.168.39.185:8081 that is mine.
Yours will be different. In the pi window you see your address by clicking
at top right.
Hope I'm giving useful and correct info.
Ronald.