jbruce12000/kiln-controller #83: Cannot start-on-boot

umfan92, 2022-05-04:
Ok I've gotten the server to work, but now I can't get it to start on boot. I've gone through all the systemd documentation and forum posts online and can't figure it out. 

The error I'm getting when I try to see the status of the service using "systemctl status kiln-controller.service" is a bunch of error messages but I think the main problem is:

"[Errno 99] Cannot assign requested address: (10.0.0.103', 8081)"

I've try modifying the service to delay it until after the network is established, or after the user profiles are loaded, etc. I can't get it to work at all. No idea how to get it to start the service on start. I'll keep trying to figure it out, but I'm stumped.

umfan92, 2022-05-04:
I got it to work with a crappy work-around (you can tell I'm not a programmer). 

I added a 20 second delay I mentioned before with ExecStartPre in the systemd service file.

Seems to boot up just fine. It did timeout once when I set it to 60 seconds so I reduced it to 20. I bet I could reduce it further but it works fine like this. I actually added a TimeoutSec = 240 to the file too so that it wouldn't timeout. idk if that will help or not.

If you have a better fix for me with the right way of doing things, I'd love to implement it. Thanks.

jbruce12000, 2022-05-04:
hmm. can you post all the error messages? even if no network is connected to the host running the kiln-controller.py, it should still start. you would only be able to access it locally on that host if there was no network, but it would still start.

did you change the listening ip in config.py to something other than 0.0.0.0? if so, change it back.

jbruce12000, 2022-05-04:
Also very curious, if it starts, but then dies... what is the cause? posting the content of /var/log/daemon.log would help me figure out what's happening.

umfan92, 2022-05-04:
Thanks for much for the response! I'll try to get the content of /var/log/daemon.log later today. 

Also, I think I'm going to just wipe the RaspberryPi sd card and start from scratch. It's very likely that I messed some stuff up during the install. I didn't use "pi" as the username. I had to manually change all the files to my username. Maybe I missed something critical?

I did change the 0.0.0.0 in config.py because I initially kept getting the error message that said I have to adapt it and copy to config.py (from the previous issue I closed: https://github.com/jbruce12000/kiln-controller/issues/82#issue-1224864768)

So I likely screwed some stuff up when trying to fix it. I'll report back with the error message and content of /var/log/daemon.log then I'll wipe it clean and try again.

jbruce12000, 2022-05-04:
ok, let me know how the fresh install goes. if you find problems with the instructions, they should be fixed.

The default user for Raspberry PI os is "pi". You should use that unless you have a good reason not to do so. The default password is "raspberry". I'd change that.

umfan92, 2022-05-06:
Thanks so much for your help. Everything seemed to work perfectly on the fresh install. I used "pi" for the username and I got through the entire process without a problem. I think what may have gone wrong the first time was when I ran the "sudo apt-get dist-upgrade" it didn't work but I wasn't paying attention. 

I'm at the point where I can start up the raspberry pi now and it'll start the server. But when I tried to change the config.py file to remove the simulation, it stopped working. 

By default it's set to "simulation = True". Am I supposed to set it to "False" or just comment it all out? 

Also, I don't quite understand the procedure to change the config.py with the config.py.EXAMPLE. Am I supposed to only modify the EXAMPLE config and copy it to config.py? Or can I just modify the config.py file directly?

umfan92, 2022-05-06:
Yeah as soon as I try to make any changes to the config.py file, it stops working. Not sure about that part of the instructions. I changed to the MAX31856 board and it no longer starts on boot. I'll see if I have to rerun some of the commands to initiate it. But I'm terrible with programming :(

umfan92, 2022-05-06:
Sorry for all the comments. After downloading the Adafruit_Python_GPIO library (which is now deprecated but I don't know enough about coding to change it to the new library) I stopped getting the errors in Thonny. 

But now I keep getting the error: 

"Could not import config file.
Copy config.py.EXAMPLE to config.py and adapt it for your setup."

rondoc, 2022-05-06:
You can just copy using the GUI. For simply folk ( like me) it might be
easiest. The 56 board did not work for me with a K type thermocouple. If
the thermocouple leads were disconnected the kiln controller stopped.. I
found that the 55 board did not do that and resumed ok after a break of the
TC wire. You do not need the 56 unless
You using a TC that's not a K type. You might like to know that the pi has
a built in SSD copier. You just need a USB to SSD adapter.
There is little reason to power off the pi.
After you have modified config you MUST save it. And also do a RUN to check
it's ok.
Ronald.

umfan92, 2022-05-06:
I bought the 56 board because I figured it was the same price and maybe one day I'll get a different thermocouple (for some dumb reason). 

If it's causing this much trouble, I guess I can buy the 55 board. But I was hoping to make it work with this one. 

What does it mean to copy it? Do I just copy the text from the EXAMPLE into the text of the other one? What if I just modify the config.py? I'm sorry I just don't understand what I need to do.

umfan92, 2022-05-07:
At this point, I can only get it to work properly before I change it to "simulate = False", even if I leave the MAX31855 board settings.

rondoc, 2022-05-07:
Simulate false makes sense. I have it as false. I think, and Jason can
verify this,that the simulate is for a dry run without actually firing the
kiln. I don't understand what it's use is. Sorry.
Copy the complete config. Example.py as it's too easy to miss something.
Use t he file manager on pi or you can do it all on a pc. Then save to a
micro SD card. You will need a SD card to USB reader.
Once you have it working it is pretty trouble free. Don't use the browser
on the pi to run the app. It works much better on a pc.
 I recommend following the PID selection process , it runs the kiln upto
400F and then computes the pid values. Might need tweaking but it works as
is for me.
Ronald.

umfan92, 2022-05-08:
No matter how I try to copy the config, I keep getting the same error message. If I leave everything default, it'll run just fine (except it won't actually read my temp since it's the wrong board). 

As soon as I try to change the config.py it give me the error message that I need to "Copy config.EXAMPLE.py and adapt it for your setup."

umfan92, 2022-05-08:
Closed issue since the original issue is no longer the problem.

closed 2022-05-08 by umfan92

jbruce12000, 2022-05-08:
the purpose of simulate=True is so anyone can download and test drive the interface without it being connected to a kiln.

rondoc, 2022-05-08:
Thanks for the reply. I cannot understand what the problem is with the
config.py. Am I correct that running config.py should no errors?

jbruce12000, 2022-05-08:
@rondoc see the other thread. config.py is python code. it gets compiled just like any other python program. if there is a syntax problem, it will error out. if there is a library loading problem... where config.py cannot load something [or load another lib that cannot load something], it will error out.

Since @umfan92 does not understand what a virtual environment is, my guess is that the virtual environment is not created or activated or contain the right libraries. It's tough to tell without more information.

```
source venv/bin/activate
python -m py_compile config.py
```
will show whatever the errors are though.