jbruce12000/kiln-controller #84: Could not import config file

umfan92, 2022-05-08:
Sorry if it's bad etiquette to close my other issue and open a new one. I really don't understand how github works. 

After the fresh install, I was able to get through everything without a problem as long as I don't modify the config.py file. 

Something that I'm doing to the config.py or config.py.EXAMPLE is not working and gives me the error: "Could not import config file. Copy config.py.EXAMPLE to config.py and adapt it for your setup."

I've searched through the code in kiln-controller.py and came across the location of the error:

`try:
    sys.dont_write_bytecode = True
    import config
    sys.dont_write_bytecode = False
except:
    print ("Could not import config file.")
    print ("Copy config.py.EXAMPLE to config.py and adapt it for your setup.")
    exit(1)`

I don't know how to troubleshoot and run this code properly. Not sure what the whole "virtual environment" stuff is but I suspect I can't just run this file to check for errors (?). 

When I change the config.py file to uncomment the relevant lines for the MAX31856 board, and change simulation to FALSE, then everything breaks. Any ideas how I can troubleshoot this? I wish I had more experience with coding so I could contribute but I can't figure it out. I really appreciate your help.

rondoc, 2022-05-08:
1. Please  tell me exactly how you are copying config.py EXAMPLE  to
config.py.
2. The only essential changes you need to make are the pi IO
connections to the  max56 board and the S type to K type.  Also PID and
simulate to false.. Then SAVE..
3. Then try to RUN just this config.py . What do you get?
Ronald.

jbruce12000, 2022-05-08:
A virtual environment allows you to completely isolate a python environment. Inside a virtual environment, you can run any version of python you want. You can download libraries using pip and they will only apply to the virtual environment. It's a form of isolation that guarantees you are not impacted by other installations, python updates, or libraries.

jbruce12000, 2022-05-08:
try the following to see if you made some error in config.py. remember, in python, indentation/spacing matters...

start out in the same directory as config.py

```
source venv/bin/activate
python -m py_compile config.py
```

if there is any output it's an error. fix it. repeat until there are no more errors. If you get stuck, post the output here.

jbruce12000, 2022-05-08:
as far as ettiquette, open a new issue for every new issue you have. that makes it easy to track and to know when a specific problem is solved.

rondoc, 2022-05-08:
Jason,  I think that for non coders using GUI in pi is the easiest. I also
think that spacing etc is so critical in python that non coders can easily
mess up, that which looks simple to you, by making changes to a .py file.
Since there is already a config.py in your kiln.... download what is the
purpose of copying EXAMPLE over?
Thanks.
Ronald.

umfan92, 2022-05-09:
> try the following to see if you made some error in config.py. remember, in python, indentation/spacing matters...
>
> start out in the same directory as config.py
>
> ```
> source venv/bin/activate
> python [...]

Well good news and bad news. Good news is there's no error. Bad news is there's no error. I get no output at all. 

I have no idea what's going on. I still get the same message when trying to start kiln-controller.py.  "Could not import config file."

Sorry to put you through this. I wish I could help with the troubleshooting. Is there anything else I can try or any other logs you need me to post?

umfan92, 2022-05-09:
I definitely think the problem is that I'm using the MAX31856 board. I noticed there's a file called "max31855spi.py" but not one that's for the 56. Should there be one? Maybe that's the issue?

rondoc, 2022-05-09:
I'm sure Jason has the answer but I tried the 56 board and gave up. The 56
setup crashed when the Tc wires broke.. I use the 55 board but still leave
all references to max31856 as-is in the config.py file. I tried changing to
55 but I don't think the library has included in it.
 The IO for the extra one connection that the 56 board uses I ignore.. I
have no problems with that.
BTW after an error free run of config.py. I then open kiln-controller and
execute in the terminal window. You should see it waiting for browser
connection . I use 127.0.0.1:8081 for pi local access but it's much better
on my pc with http://192.168.39.189:8081 but this will be different for you
. This is a local network connection.. The pi has a unique ID that shows
when you click on top right of the GUI window.
Ronald.

umfan92, 2022-05-09:
It's weird because I can get it to work fine with the 56 board when running kiln-controller.py from the IDE. It starts the server and reads the temperature just fine no problem. So I know my wiring is correct and it feels like my config.py code is fine. If I heat up the thermocouple, it reads a higher temperature almost immediately. 

It just won't run from within the terminal and keeps giving me the "Could not import config" error. 

I just wish I had a more specific error message or something that would explain where it's having trouble importing so I can go troubleshoot. I can't figure out _why_  it can't import.

umfan92, 2022-05-09:
Holy shit I just got it to work. I honestly don't even know what I did. It reminds me of this meme. 
[image]

I'm gonna write out what I did right now before I forget.

- I was getting an error when running the config.py code in the IDE where it said it couldn't find "lib.max31856" so I brought the max31856.py file into the kiln-controller folder and removed the "lib." in that line  (doubt this did anything and I'm still getting an error regarding "import max31856" when starting it up but it works fine so.....)

- I re-ran some of the code in the instructions, namely: "pip3 install --upgrade setuptools
$ pip3 install greenlet bottle gevent gevent-websocket" I know I had already run these before. The first line said all requirements were already satisfied but I think the second actually did download something. Not really sure. 

- I ran the code in the instructions under the "Raspberry PI Deployment" section. I didn't think I had to do this before so I didn't do it. But I figured I'd just try it out. I have no idea if this matters at all.

That's really all I remember. I messed around a lot in the max31856.py, oven.py and kiln-controller.py files, but I don't think I did anything substantial. I was just trying to understand the code. Let me know if I can help you in any way to understand what monstrosity I have created.

rondoc, 2022-05-09:
> @rondoc <https://github.com/rondoc> config.py.EXAMPLE is a hold-over from
> the original code. Would it make things clearer if that were removed... and
> removed from the instructions?

Yes I think so. I also think that since most users have K type TCs there is
no advantage to using the 56 board. The 55 board is slightly simpler to use
and works fine ( for me).
The PID setup is tricky as trial and error is a long journey with a large
kiln. The step function followed by analysis using the tuner worked well.
The initial oscillations at low temperatures are not important for a
ceramacist and attempting to reduce them is a waste of time.
I know this is off topic but you could add some of it to the general info
you have.
Thanks again.
Ronald.

umfan92, 2022-05-09:
@jbruce12000 @rondoc 

Let me know if I can help. I'm not a programmer but I am a mechanical engineer and I try to learn pretty quick. This is a really cool project and it would be awesome if I can help contribute.

umfan92, 2022-05-18:
@jbruce12000 

I've finally finished all the electronics assembly and set it up on my kiln. First run worked quite well. I didn't set up remote access to my pi and I hadn't tuned any of the setting for my kiln. So it was overshooting like crazy and delaying a lot, but I'll set up remote so that I can log in and change my PID constants as I go. 

I've been waiting to close this issue because I wanted to update this issue with what changes I made to get my code to work. I've opened the files in Notepad ++ and compared all my .py files with the originals. 

If anyone is struggling with the MAX31856 board, these are the only changes that I made to my files:
- In max31856.py I changed line 92 that starts with "def _init_" to the k type thermocouple
- In oven.py I changed line 63 to remove the comment for the "from max31856 import MAX31856, MAX31856Error"

That's all I really know. I know Jason has been updating the code a lot recently and I like some of the stuff he's been doing. So I'll be trying to update to the new code and see if I break something.

closed 2022-05-18 by umfan92

jbruce12000, 2022-05-18:
@umfan92 yes, please update. lots of changes.

I also need someone to test a new script which zips up all the logs. then send me the output so I can check it out. this will help with future troubleshooting.

see https://github.com/jbruce12000/kiln-controller/blob/master/docs/logs.md near the bottom.

umfan92, 2022-05-18:
@jbruce12000 Awesome! I definitely will. 

I have to say this kiln controller is so much better than the cheap PID controller I had. I had no ramp/soak capabilities, not to mention remote web access. 

I'll work on installing remote access to the RPi today and then updating the new code. Then I'll run a quick schedule and send you the output. How do I send it to you? Do you want me to just post it here and tag you?

jbruce12000, 2022-05-18:
yep, post it here and I'll pick it up and create some graphs from it and post those back.

glad it's useful. thanks for helping.

I think remote access to the pi is easiest using ssh.

umfan92, 2022-05-20:
@jbruce12000 

I was able to finalize the new install with the updated code. It's running so well! I haven't even tried tuning the PID constants. 

Here is the log file you wanted. I'm not really sure if it worked and I have no idea what I'm looking at. But let me know if I can help with anything else. 

[kiln.logs.gz](https://github.com/jbruce12000/kiln-controller/files/8734886/kiln.logs.gz)

rondoc, 2022-10-11:
Make a copy of the micro SD ..NOW.  You will need it.