jbruce12000/kiln-controller #86: API scheduling (not an issue)

captaincaden, 2022-05-11:
I've finally fully completed this project, and am no longer just using it as a smart pyrometer. I've decided that I want it to just control the ramp, but let my kiln-sitter ultimately still decide when the kiln will end. I may have another issue relative to this that I noticed during my testing, but I'll make another issue on that if I find that it's true, or that it does behave in a weird way. (just to summarize: this issue is relative to during one of my tests, when the kiln got above the target temp (this was during the last ramp near top temperature, where my profile slowed down a bit, the pi had logged that the kiln was too hot and must catch up (it was), but the pi continued telling the relay to stay on, and subsequently, held the relay closed until the kiln sitter shut the kiln off, and ignored the target temperature) 

**So the real question of this post**

So, I've read around pull requests and other issues and the docs about using API to set a specific time that the controller would start __ profile, and I'd like to know the exact command that you'd use to achieve that. I've successfully used the written API's to start and stop from the terminal, but I'd like to be able to say: I want the profile to start at 6:00AM on 5/12/2022, how could I do this? What other things are possible to do from the API?

I liked the pull request regarding this, but I have a lot of my own customizations already, and am afraid to go through and implement that specific pull request to add the scheduling to the code and then mess something else I've already done up lol.

jbruce12000, 2022-05-12:
The api was broken until yesterday's commits. Make sure you're running the latest from the master branch.

captaincaden, 2022-05-12:
I haven't updated the master for a few months, mostly because of the other changes I've made, and not exactly understanding how to update, and also how to update and keep the changes and such I've made without copying and pasting them all over. 

I tested the API with the version I'm currently running yesterday, and it worked as expected. I will try out using the at scheduler, thank you! 

Thank you for the response in regards to that question. I thought that was the case, but was just curious and had no way to test it in my current set up. 

I plan to use the same cone that I'm firing to, so in the case of the relay failing closed, the kiln sitter or myself would be there to stop it by the time it was able to reach top temps.

captaincaden, 2022-05-12:
Also, just to understand it better, if I say at 3am, is that just the next time the clock is 3am? Would it repeat that every time it was 3am?

jbruce12000, 2022-05-12:
you should read the "at" docs, but in this case, it would create a single entry in the queue for 3am. it would run once the next time it is 3am. To check the queue, run...

`atq
`
If you want to run a recurring job, you want to use cron... but you probably don't want that.

jbruce12000, 2022-05-12:
to remove an at job from the queue [has not started yet] use:

`atrm N`

where N is the job number you want to remove. If the kiln is already running the profile started by **at**, do this...

`curl -d '{"cmd":"stop"}' -H "Content-Type: application/json" -X POST http://0.0.0.0:8081/api
`

to stop it.

captaincaden, 2022-05-12:
Thank you again for all of your help.

captaincaden, 2022-05-12:
That part of the api also brought up another (unrelated to original topic) question: 

In the event of a power outage, what is the plan of action? I've thought about putting my pi on a UPS, but haven't done so yet. I normally try to plan firings for days that are least likely for a power outage, but things happen beyond control sometimes. 

Would just restarting profile be sufficient? Or what is the best course of action? (presuming the pi was restarted during the power outage)

Thank you for that information. It's something I figured I'd need to read up on before I did, but I'm satisfied with the way it functions now, so I've just been waiting for an update that was a necessary update to learn. 

All of the changes I've made are in the oven.py file, and are relative to sending phone/email notifications for specific temperature events, and for specific emergency events. It's a good maybe 40-50 lines of additional code, as well as some stuff in the config file to configure those events. I'm sure I could work around it in an update, and the added code itself is relatively simple, and doesn't affect any of the original workings of oven.py or the rest of the code, just piggybacks off of functions that already existed, so if I took the time to review it, I'm sure I'd be able to work around any conflicts, but again, I'd need to learn how to merge and update it and all of that first.