jbruce12000/kiln-controller #89: ModuleNotFoundError: No module named 'RPi'

umfan92, 2022-05-18:
I just wanted to bring this to your attention because I encountered this issue when I was installing it the first time. I'm getting this error again.

I guess I have to download the Adafruit GPIO library right? I think that's how I fixed it last time. But I believe that library has been deprecated. Is that something that would cause a problem or is it just an outdated library? 

Anyway, you told me to report issues in case the instructions should be updated. Maybe that's something that needs to be added to the instructions? (Or maybe I'm just doing something wrong)

umfan92, 2022-05-18:
Here's the full error when trying to start up the python script:

pi@raspberrypi:~/kiln-controller $ source venv/bin/activate; ./kiln-controller.py
2022-05-18 19:32:16,682 INFO kiln-controller: Starting kiln controller
2022-05-18 19:32:16,699 INFO kiln-controller: this is a real kiln
2022-05-18 19:32:16,702 INFO oven: import MAX31855 
2022-05-18 19:32:16,704 INFO oven: init MAX31855
Traceback (most recent call last):
  File "/home/pi/kiln-controller/./kiln-controller.py", line 44, in <module>
    oven = RealOven()
  File "/home/pi/kiln-controller/lib/oven.py", line 402, in __init__
    self.board = Board()
  File "/home/pi/kiln-controller/lib/oven.py", line 47, in __init__
    self.create_temp_sensor()
  File "/home/pi/kiln-controller/lib/oven.py", line 75, in create_temp_sensor
    self.temp_sensor = TempSensorReal()
  File "/home/pi/kiln-controller/lib/oven.py", line 107, in __init__
    from max31855 import MAX31855, MAX31855Error
  File "/home/pi/kiln-controller/lib/max31855.py", line 2, in <module>
    import RPi.GPIO as GPIO
ModuleNotFoundError: No module named 'RPi'

captaincaden, 2022-05-18:
I believe I ran into this issue as well when installing and using the following installed what was needed without incident: 

`$ export cflags=-fcommon
$ pip install -r requirements.txt`

jbruce12000, 2022-05-19:
added this to the installation instructions...

`export CFLAGS=-fcommon`

closed 2022-05-19 by jbruce12000

umfan92, 2022-05-19:
......what does it mean? 

lol I still don't really get it. Does that just mean deploying on a RPi?

jbruce12000, 2022-05-19:
Wording changed to this...

`If you're done playing around with simulations and want to deploy the code on a Raspberry PI to control a kiln, you'll need to do this in addition to the stuff listed above:
`
I think that is a little more clear.