jbruce12000/kiln-controller #90: api stats endpoint and watcher script

jbruce12000, 2022-05-21:
I'm lazy. I don't have time to watch a screen all day while my kiln runs. I added a stats endpoint to the api.

`curl -d '{"cmd":"stats"}' -H "Content-Type: application/json" -X POST http://0.0.0.0:8081/api`

I also added a watcher script that I can run on any machine in my local network to monitor the kiln. It sends a message to slack if the kiln is out of whack.

`./watcher.py`

jbruce12000, 2022-05-21:
creating a slack channel and setting up a webhook takes only a few minutes and requires no authentication. good enough for me and I'm already using slack for other things.

captaincaden, 2022-05-21:
Maybe I missed something when transferring the code over for the /api/stats, but I don't believe I did, unless the watcher.py file is needed for that command to work, but using the curl command above I only get a return of `{ }` and nothing else happens. I didn't pull it all over again just to save the work of having to copy all of my additions over, but I copied both of the added sections in kiln-controller.py and that's the result I get. The other API's still work normally and the kiln-controller otherwise works normally, so I must have missed something adding over but not sure what specfically

jbruce12000, 2022-05-21:
you were probably not running a schedule... so you get {}.

try running a schedule. you should get the same as pid.pidstats for that 2s.

the api does not require anything except kiln-controller.py to be running.

captaincaden, 2022-05-21:
Yep, that was it. Didn't realize it needed to be running a schedule, but that makes sense. It works as I expected. Thanks!

captaincaden, 2022-05-23:
I added the watcher script to the kiln-controller directory. I didn't think there was anything additional that needed to be put over, and I copied it raw from the github page. When I try to run ./watcher.py in cd kiln-controller I get the error
 `/usr/bin/env: ‘python\r’: No such file or directory`
 
 I also think it'd be helpful to have more documentation on what Slack is and maybe a link to how people can set that up for themselves if they're interested in it and don't have any ideas what Slack is. It took a fair amount of research to figure it out. Also just out of curiosity, would the watcher still send notifications say if my phone/another computer wasn't on the same network as the pi?

jbruce12000, 2022-05-23:
To troubleshoot that problem you posted, post the output of:

`which env
`

more docs on slack and how to integrate would be useful. lower priority.

The watcher will work on any network. If the process can reach both the PI and Slack, it'll work.

jbruce12000, 2022-05-23:
This watcher script uses the same shebang line as the kiln-controller.py script. It needs a similar environment to run. I'm running mine on a linux machine. If you're running yours on windows, this script may require some modification to work.

captaincaden, 2022-05-23:
Sorry, I’m not home at the moment to run the command but will when I get back to the pi. 

I was under the impression the watcher script should run on the pi. I had tried to test it on the same pi that’s running the kiln-controller. 

I had tried using cd kiln-controller then ./watcher.py as well as tried to use the same cd kiln-controller then source venv/bin/activate ./watcher.py and watcher.py and still got the error.

jbruce12000, 2022-05-23:
Lemme know how the windows tests go. The **no data** response is the expected behavior.

I fixed the missing comma.

jbruce12000, 2022-05-23:
ok. docs added describing the purpose of the watcher, and why/how to hook up slack.

captaincaden, 2022-05-23:
I think those are excellent docs regarding it, and I love the concept of it. I think it pairs nicely along with the text message updates that I added to my kiln-controller. The texts are great for major errors and temperature updates, but I didn't write anything that looked at the process actually running, and any temperature fluctuations, so I think it will pair nicely and complete that whole automation and notification thing for me. 

With just basic tests, just removing the #!/usr/bin/env python from the beginning of the code works well on Windows. I don't know that it was necessary to remove it, but I added some temporary test code to make sure it outputted. I couldn't figure out how to make the simulation actually get behind, but I verified that it did send the `error kiln needs help. {}` every 6 no data responses, and my test code to output to the slack channel in a different location worked.

Something I'm not sure if I'm following though (or maybe It's not working correctly, or something else) , what should be displaying when the kiln is actually running a profile? In my current case, the kiln is physically disconnected at the kiln sitter, but when I start a profile, I still receive the no data response, even though there's a target temp, actual temp, and the command to heat (no heating is actually being done, but still) so I'm curious, when is the script supposed to start seeing data, and what should it be outputting in the console when it is?

captaincaden, 2022-05-23:
Disregard that last section. I figured out the issue. 
My kiln controller uses port 8081, but the watcher script had 8082 in place, so thus it was watching nothing. 

It appears to be working well on windows, and is reporting as I'd expect, as well as the errors I could test seemed to be working well. 

The next thing I'd love to be able to figure out how to do on windows is use your kiln-stats project to get those outputs like you got from the .gz file, but that's another project and topic for another time haha.

jbruce12000, 2022-05-23:
ah, yes, because I was testing on 8082. you'll have to change the port. This brings up a good point though. I should have the code be more verbose when it cannot get any data. check url, check port or whatever.

as far as the kiln-stats project:

1. download and install cygwin (gives windows some linux tools)
2. create a virtual environment
3. source that environment
4. pip install -r requirements.txt

these are the modules in requirements.txt:

```
plotly
kaleido
numpy
pandas
```

If that install works, then the only other dependencies are pdftk and sqlite3 (simple database).

captaincaden, 2022-05-23:
Thank you again. I will try the kiln-stats stuff later!

jbruce12000, 2022-05-23:
Here's output from a sample run. I added instructions if the script cannot connect (like bad port or url).

(venv) jason@upstairs:~/repos/kiln-controller$ ./watcher.jbruce.py 
INFO:__main__:started watching http://0.0.0.0:8082/api/stats
ERROR:__main__:network connection error. check kiln_url and port.
ERROR:__main__:no data
ERROR:__main__:no data
ERROR:__main__:no data
ERROR:__main__:temp out of whack 65.13
ERROR:__main__:temp out of whack 5.25
INFO:__main__:OK temp=64.20 target=68.73 error=4.53
INFO:__main__:OK temp=68.23 target=70.98 error=2.76
INFO:__main__:OK temp=72.17 target=73.24 error=1.07
INFO:__main__:OK temp=75.43 target=75.49 error=0.06
INFO:__main__:OK temp=78.01 target=77.74 error=-0.27

captaincaden, 2022-05-24:
Updated it and tested and  it works well on windows as well.

jbruce12000, 2022-05-26:
closing this. latest updates were tested and everything looks good.