jbruce12000/kiln-controller #91: Heat on icon intermittently broken.

captaincaden, 2022-05-22:
[image]

I've noticed sometimes when I open up the websocket when a profile is running the heat icon appears like this, and only changes by making the red box slightly bigger. It wasn't like this during my firing yesterday, and it wasn't like that when I started this firing today. I had to stop the firing and restart the kiln controller about 25 minutes into the firing (un related to this issue or any issue with the kiln-controller), and when I used the API to restart the profile, it appeared that way. 

I've noticed it do this before, but don't remember it staying that way, and also don't remember what I did prior to it last time. But I've never used the API to restart, so I don't think that the api restart caused it, so I'd imagine it's just an odd thing that happens when it boots sometimes for some reason. I'll play around and see if I can get it to correct itself or otherwise repeat the issue once this firing is completed, but thought I'd bring it up.

rondoc, 2022-05-22:
What browser are you using? Browser on pi doesn't always work but Google on
a pc win10 is always ok.

captaincaden, 2022-05-22:
I'm using it on Google Chrome on windows 10. I also just opened it on Firefox on Windows 10, as well as Safari on iOS, and all have the same issue currently with this firing.

rondoc, 2022-05-22:
At least we know it's not the browser!!
Ok smart guys, what part of the code is responsible!!

captaincaden, 2022-05-22:
Now that my firing has ended, the icon box there is just blank, no heat icon at all.

I believe the icon is created from the picoreflow.css file, and I believe it gets it's functions from picoreflow.js but I'm not sure what would cause it to not appear, and not really sure much more other than that. 

Calling to restart kiln-controller, the icon appears for a second, then disappears. This remains the same on every restart I've tried, including just starting the kiln-controller manually through source venv/bin/activate kiln-controller.py etc.etc.etc.

jbruce12000, 2022-05-23:
I removed the flashing icon and replaced it about a week ago. I replaced it with a progress bar that shows what percentage of the time the kiln coils are being heated. 0% is completely black, 100% is completely red.

That flashing icon was always a pain to maintain (getting it to flash at the correct rate), but even more importantly it didn't tell me anything.

captaincaden, 2022-05-23:
Understood so the image I included is what it’s supposed to be doing? Is it just showing when the heat output is on?

jbruce12000, 2022-05-23:
Yes. exactly that. Judging by how full the progress bar the moment you took this screenshot, I'd say the kiln was heating coils about 70% of the time.

closed 2022-05-23 by captaincaden