jbruce12000/kiln-controller #92: SAVING PROGRAMME ISSUES

MarkKiln, 2022-05-23:
Hi, I have no idea how this works exactly but I have recently purchased a WiFi kiln controller utilising this software and all is working fine other than the SAVE function!

it will let me edit a cycle just fine but will not over ride existing programs and will not save a new one once I create it!

cheers

jbruce12000, 2022-05-23:
I tried saving a profile on the latest master and it worked fine. Can you post the logs so I can see what error you're getting?

MarkKiln, 2022-05-23:
What part of the program should I post? Cheers for the reply!
just now it has let me save a new program to the list (there were 4 originally) so I tried making another and it didn’t save!

jbruce12000, 2022-05-23:
I know nothing about this product you bought. I'm the author of the software. It runs on a raspberry pi. Were you given instructions on how to login to the pi? You'll need this information so you can login and look at the logs. Contact the seller to get login details.

jbruce12000, 2022-05-26:
gimme an update. it's been a few days. did you get the login and password to your pi?

jbruce12000, 2022-05-30:
No worries. glad you found a work-around.

You should contact the seller and ask what's running inside so you know for future reference. If it's a PI, great... get the login and password so you can do updates and fix problems etc.

closed 2022-05-30 by jbruce12000

MarkKiln, 2022-10-11:
I see, this doesn’t run off a raspberry pi I have attached an image of the
unit, it all works as it should and connects all fine!

Just will not let me add or save new programs other than the first program
I set, on that first program it came up with a pop up window asking if I’d
like to over ride

I have also contacted the manufacturer but haven’t heard anything back

Thank you so much for your help

Matt

jbruce12000, 2022-10-11:
I wish I could help more here, but I'm just not sure what to tell you to do. I think you'll have to get support from the seller.