jbruce12000/kiln-controller #93: Defective MAX31855 modules and potential fix.

dsilfvast, 2022-05-23:
I've had several defective MAX31855 modules. If you’re having problems, do a quick check to see if “T-” (pin 2) is shorted to “GND” (pin 1). The datasheet explicitly states “Thermocouple Input. See Table 1. Do not connect to GND”.

It’s not a pretty fix, but I’ve been able to scrape a needle between those IC pins to break the copper connection. It cleared up some real funny problems.

jbruce12000, 2022-05-23:
cool. thx for sharing.

closed 2022-05-23 by jbruce12000

tonymoon61, 2022-10-01:
I've run into this one 4 modules now - two 31855, and two 38156. I wonder if there is some application for them that this is beneficial, because it's awfully common, and always the same spots.

rondoc, 2022-10-02:
I think it's that both modules need the tc to be isolated. So often one
lead will be grounded or has a leaky electrical pathway. Could be worse at
high temps as some glazes and clays become electrically conductive then..
use porcelain or better alumina insulators.
To check disconnect tc from anything. Then using high range ohmeter VOM OK.
Measure either lead to ground..IT MUST BE OPEN TO GROUND . A few hundred
Kohms is not always high enough.
I found the 31855 to be perfectly OK for the pi using K type tc. K is ok
for less than cone 12. Over that temp S type.
I keep a Fluke thermocouple thermometer connected at all times to keep a
check on kiln and also to compare with the pi kiln system. Usually tracking
is within a degree.
Ronald.