jbruce12000/kiln-controller #94: Maybe an API call to write memos to the log?

maabaum, 2022-05-28:
This really isn't an issue so much as a suggestion. I wonder about writing an additional API call along the lines of:

`
    # this new API command allows you to send a memo line to the log file
    # that will be timestamped -- so for example opening the kiln lid
    # to rapidly cool or something

    if bottle.request.json['cmd'] == 'memo':

        msg = bottle.request.json['msg']

        log.info('memo: %s' % msg)
`

There are almost certainly errors in that, but the idea, as stated, would be to allow the user to write time-stamped memoranda lines into the log file to note events that might be import for later diagnostics.

jbruce12000, 2022-05-29:
added. please test. I tested and seems ok.

maabaum, 2022-05-29:
Seems to work fine. Thanks!!

closed 2022-05-29 by maabaum