jbruce12000/kiln-controller #95: Kiln controller falls over mid-run

steeleh, 2022-05-29:
Not really a kiln-controller problem but it was causing me problems using my kiln-controller set-up so I thought it might be of interest. 

My kiln-controller runs on a headless (no screen) Rasp Pi in the garage. The Pi is connected to the noisy house WIFI and controlled by SSH (puTTY) from a tablet in the house.

I had problems with the program falling over at random times, particularly at 2 or 3 hours into a 7 or 8 hour firing (annoying).

Some investigation revealed the problem was likely WIFI hiccups resulting in the SSH connection failing. The solution was to install Linux SCREEN (https://www.gnu.org/software/screen/). This useful little utility allows the Rasp Pi to be detached from SSH (Ctrl A D) and kiln-controller now runs on impervious to any SSH clitches. Kiln-controller now happily runs 9 hour firings with no fails. 

SSH connection can be reinstated at any time from the command line by <screen -r>.

jbruce12000, 2022-05-29:
The raspberry pi, even if not connected to wifi, should still be controlling the kiln.

What are the symptoms when it falls over? can you post the logs? check out ziplogs which will create a gz file for you that you can post here. Are you running the latest version of the software? There have been lots of changes over the past month.

captaincaden, 2022-05-29:
I'm not seeing how the disconnection of the SSH should interrupt kiln-controller unless you're manually starting the kiln each time. Meaning, each time you're going to run the kiln, you're using puTTY to:
 `cd kiln-controller `
 `source venv/bin/activate ./kiln-controller.py`
 
 In that case, if that window got closed or any other disconnection to ssh happened, it would stop running. 
 
 I've use puTTY as well as remote desktop viewer to ssh or work directly on the pi remotely frequently. If you have start-on-boot running, then this issue wouldn't happen (assuming the original statement I assumed was true.). This would also make it so the ssh window wouldn't need to remain open or connected once you started the service. 
 
 Again, just with my experience with puTTY and this software, this is what I'd assume the issue is here. The solution could be to do your workaround, or just to enable start-on-boot. It really does make kiln life much easier.

jbruce12000, 2022-05-30:
good point. I hadn't even thought of someone starting the process manually. That would certainly cause this problem.

closed 2022-06-02 by steeleh