jbruce12000/kiln-controller #96: Can't tune PID.  Kiln-tuner.py isn't even close

JoeyG1973, 2022-06-16:
So I have been trying to get this to work for months.  I eventually figured out that my coils on my kiln were basically useless so I went to   My Kiln is a Norman 3L... it's a 13" cube internal cube door access, door and 3 sides have heating elements.  Top and bottom have no elements.  It's 220 volt.

Here is what kiln-tuner gives me for an initial PID.

pid_kp = 3.536848634417805
pid_ki = 52.76122878534157
pid_kd = 165.00148689669365

This was the result of the test 200-250 run.

[image]

And here is the log.

[test 200-250.log](https://github.com/jbruce12000/kiln-controller/files/8915134/test.200-250.log)

Cutting down the P or increasing the D ( even into the thousands ) has little to no effect on the overshoot.

I am completely lost here.

Please help

JoeyG1973, 2022-06-16:
So a follow up... I think I know what the issue is and it has nothing to do with PID.

The catchup, builds too much momentum and when the pid kicks in at 190 in this case it shuts the heat off completely but it is too late because the buildup blasts straight past 210 well into 230.... just going from 200 to 250 it does the same thing.

I tried to change the pid_control_window to 20 but even then it blasted right past that.  I think what I need here in the code is a way to, based on current temp and error, dynamically change how "hard" the catchup is hitting.  So that at lower temps with a smaller error it doesn't push as "hard" but at higher temps based on error, it pushes "harder".

I know how to code, I write stuff in python all day.  It's the math that I am terrible at.

rondoc, 2022-06-16:
You need to change the startup system. Until the PID makes sense it would
be best if the power level was low enough so the rate of heating matched
your required rate. Once the kiln temp reached about 500 degrees the PID
will keep a very close match to your setting curve. In my kiln I have two
control systems. One is the pi , the other a manual setting of power level.
From prior use I know that 8% duty cycle will give me the rate of rise I
need. When the kiln reaches 500 I switch over and tracking it within a
degree using the PID and pi. If I use just the pi I get a big overshoot at
the start regardless of the PID because the pi program ignores the error
and puts the kiln on full power until the temp difference is under 10
degrees. This is I believe a settable number in the config file.
Ronald.

JoeyG1973, 2022-06-16:
> This is I believe a settable number in the config file.

I am not finding anything in the config.py that says something like "below this temp use this duty cycle" in the comments.

I also thing 500 may be too low... I remember watching kiln-tuner.py rip right up to 400 and then overshoot by almost 40 degrees.

JoeyG1973, 2022-06-16:
Here is my current config.py

```
import logging

# uncomment this if using MAX-31856
#from lib.max31856 import MAX31856

########################################################################
#
#   General options

### Logging
log_level = logging.INFO
log_format = '%(asctime)s %(levelname)s %(name)s: %(message)s'

### Server
listening_ip = "0.0.0.0"
listening_port = 8081

### Cost Estimate
kwh_rate        = 0.1319  # Rate in currency_type to calculate cost to run job
currency_type   = "$"   # Currency Symbol to show when calculating cost to run job

########################################################################
#
#   GPIO Setup (BCM SoC Numbering Schema)
#
#   Check the RasPi docs to see where these GPIOs are
#   connected on the P1 header for your board type/rev.
#   These were tested on a Pi B Rev2 but of course you
#   can use whichever GPIO you prefer/have available.

### Outputs
gpio_heat = 23  # Switches zero-cross solid-state-relay

### Thermocouple Adapter selection:
#   max31855 - bitbang SPI interface
#   max31856 - bitbang SPI interface. must specify thermocouple_type.
max31855 = 1
max31856 = 0
# see lib/max31856.py for other thermocouple_type, only applies to max31856
# uncomment this if using MAX-31856
#thermocouple_type = MAX31856.MAX31856_S_TYPE

### Thermocouple Connection (using bitbang interfaces)
gpio_sensor_cs = 27
gpio_sensor_clock = 22
gpio_sensor_data = 17
gpio_sensor_di = 10 # only used with max31856

########################################################################
#
# duty cycle of the entire system in seconds
#
# Every N seconds a decision is made about switching the relay[s]
# on & off and for how long. The thermocouple is read
# temperature_average_samples times during and the average value is used.
sensor_time_wait = 2

########################################################################
#
#   PID parameters
#
# These parameters control kiln temperature change. These settings work
# well with the simulated oven. You must tune them to work well with
# your specific kiln. Note that the integral pid_ki is
# inverted so that a smaller number means more integral action.
#pid_kp = 25   # Proportional 25,200,200
#pid_ki = 20   # Integral
[... 64 lines cut ...]
thermocouple_offset=0

# some kilns/thermocouples start erroneously reporting "short"
# errors at higher temperatures due to plasma forming in the kiln.
# Set this to False to ignore these errors and assume the temperature
# reading was correct anyway
honour_theromocouple_short_errors = False

# number of samples of temperature to average.
# If you suffer from the high temperature kiln issue and have set
# honour_theromocouple_short_errors to False,
# you will likely need to increase this (eg I use 40)
temperature_average_samples = 40

# Thermocouple AC frequency filtering - set to True if in a 50Hz locale, else leave at False for 60Hz locale
ac_freq_50hz = False

# There are all kinds of emergencies that can happen including:
# - temperature is too high (emergency_shutoff_temp exceeded)
# - lost connection to thermocouple
# - unknown error with thermocouple
# - too many errors in a short period from thermocouple
# and some people just want to ignore all of that and just log the emergencies but do not quit
ignore_emergencies = False
```

jbruce12000, 2022-06-16:
[final.pdf](https://github.com/jbruce12000/kiln-controller/files/8919887/final.pdf)

Sorry you're having so much trouble getting things tuned. The above attachment is a graph of your test. The question I want to ask is...

are you gonna be running this oven/kiln at a temp range of 200 to 250F?

my point is that you need to optimize the PID tuning for what matters. My kiln has a 25F overshoot at low temps like this... and that is fine with me. my kiln PID tuning is optimized for much higher temperatures. my guess is, based on the duty cycle shown in the graphs that this is intended to run at much higher temps. Note that NO tuning will be perfect across a wide temperature range.

You are asking your kiln [via this schedule] to heat as fast as possible to 200F. The pid kicks in at 190F and goes to zero output on the heating elements immediately. Your kiln retains heat well, so it shoots past 200 to 235. If you want to reduce this overshoot, the only option you have is to create a schedule with a more mellow heat-up. I do this on my schedules in some cases... like creating a ramp from 65F to 200F over 5 mins. This allows the PID to kick in much earlier and have an impact.

Please note that at low temps, there will be some overshoot, and that's ok.... again optimize for what matters.

Try a new schedule with the mellow start-up ramp and post the logs here again so I can check it out. Keep the same settings and I'll make suggestions on the tuning.

jbruce12000, 2022-06-16:
Things I can tell from the graph[s] above. Proportional is high, cut it in half. Derivative is low and needs to be higher, double it. Integral seems ok. If you make changes, reduce by half our double one setting, repeat.

rondoc, 2022-06-16:
Are you using a S type TC?
Also the catch up is true but I set mine to false. I think that this is a
problem at the beginning of a run.
In one of issues your issue is discussed . Sorry don't know which one but
it is within last month.
Ronald.

jbruce12000, 2022-06-16:
setting kiln_must_catch_up=False would still allow the kiln to overshoot, it would just NOT hold the schedule until it's back in the window. So, what you'd see at the beginning is the same overshoot of 35F to 235F. The kiln would slowly cool down to 200F (and may not even finish cooling before the quick ramp to 250). Then you'd see the same overshoot at 250 to 275F, followed by a slow cooling until the end of the schedule.

rondoc, 2022-06-16:
Thanks. So what is best solution for this initial large overshoot. Mine is
manual override. Why can't the pi program be modified to handle it by
limiting the initial power input to maintain a given slope?
Something like this..?
1. Turn on power at 5 %
2. Wait for 2 minutes
3. Check error
4. If error negative
5. Inc power by 5  %
6. Loop to 2.

It works well under my manual control just need that in the app. Everything
else in the app is great.
Ronald.

JoeyG1973, 2022-06-16:
Ok I modified the test 200-250 that comes with kiln-controller so that it has a gradual ramp up.  I also doubled my D

I still think I need to take down the P.  Then I will see if I need to adjust the D again, but this looks much better.  I think you may want to clarify the tuning instructions as well as probably modify the test 200-250 so that it has gradual ramp up over 10 minutes.

I made some assumptions based on the tuning instructions.

[image]

[test 200-250 gradual.log](https://github.com/jbruce12000/kiln-controller/files/8921976/test.200-250.gradual.log)

jbruce12000, 2022-06-16:
Yeah, this looks fine to me. The ramps are both a little more mellow, so the overshoot is reduced. You can see the PID kick in right at the beginning so it moderates the power to the elements. You have 20F of overshoot which is acceptable. You can sure try to double derivative again. My guess is that you're pretty close now... just a little more. Proportional looks good, but experiment. There is a balance with proportional though... where you want it to be responsive at higher temps, so make sure to test at higher temps as well. What you want to avoid is a critically damped system that works perfectly at this temperature... because this is not the temp the kiln will usually run at.  It is OK and expected to have overshoot at low temps so you get the correct responsiveness at higher temps.

I candle at low temps too... with ceramics, you want to be careful not to go over 250 which can cause pieces to explode due to steam buildup.

I like the idea of changing the ramp for the test. If the instructions are misleading, pt out where so I can fix... or submit a pull request.

@rondoc I understand your idea. I think the real mental hurdle I have is that it is essentially the same as a mellow ramp controlled by a PID... which you can do now without code modification.  Heat more slowly [with a mellow ramp], let the PID do it's job.

JoeyG1973, 2022-06-16:
Let me get this settled and I will do a pull and update the tuning mark down

Thank you for the help.

jbruce12000, 2022-06-16:
I changed the 200-250 test schedule to be more mellow on both ramps. that change is committed.

jbruce12000, 2022-06-16:
We'll close this once your PR is merged is everything is cool.

JoeyG1973, 2022-06-16:
BTW I found this.  I may go through and add some of these profiles to the pull request too

https://hotkilns.com/support/pottery-kiln-knowledgebase/standard-easy-fire-program-profiles-dynatrol

JoeyG1973, 2022-06-17:
Just a thought is there anyway we can come up with a website for kiln profiles?  Something with an api that let's you pull down community profiles.  Because you can use these kilns for SO much more than pottery.  My wife wants me to put in charts for glass fusing.  I am going to be putting a bunch of stuff on metal melting and metal annealing, hardening, and tempering.  I get some of them are simple like aluminum melt is just fire that thing up but we could get fancy and estimate time to melt by asking what the weight of the aluminum pre melt.

I don't know.

Just thoughts/

JoeyG1973, 2022-06-17:
Here's an example of all the processes you can do to O1 tool steel.  Time depends on thickness.  Hmm now I am wondering if I can write a profile generator.

https://hudsontoolsteel.com/technical-data/steelO1

jbruce12000, 2022-06-17:
The profiles are json and super-simple to generate. I don't use the web interface to create mine, I just create them in vi. There is no reason why you couldn't create profiles automatically from a given data source using a script and populate a whole bunch of 'em in a repo.

I think it would be pretty trivial to store profiles in a github repo, make the directory in the repo configurable at start-up and have it grab all profiles from there instead of [or in addition to] the local file system. This would allow you to use any github repo created or forked from others and any directory in that repo. Having it in a github repo means anyone can contribute. Using folders in the repo means you can organize profiles by kiln type or purpose or whatever. This could also be used to control the profiles that the client loads... for instance, I'm a potter, I only want to see pottery profiles. Your wife might want to see just glass profiles. You metal.

That means your kiln has to be connected to the internet to load profiles. not a problem for me, but might be a problem for others. Hence the reason to make it configurable.

I've never considered this, so I need to look at this code base to see what is impacted.

rondoc, 2022-06-17:
How do you have a slow ramp at startup? I have made a slow ramp up profile
but I still get a large, over 100F surge. After 500F tracking is within a
degree all the way to 1380F.

jbruce12000, 2022-06-17:
That's really a big overshoot. It's a combination of a slow ramp and setting the Proportional low enough that it does not blast way pay the set point. @rondoc I'd be interested to see the logs from a slow ramp up profile with your current PID settings... on a run where it does this.

jbruce12000, 2022-06-17:
@JoeyG1973 After thinking about a website to store profiles... The easiest thing to do would be:

1. create a github repo for storing profiles where lots of folks can contribute
2. on installation of the kiln-controller, pull the contents of that repo ^^^
3. make the local directory configurable

This allows for a website with collaboration, the flexibility to load only those profiles that matter to you [via configurable directory], local storage [so the code doesn't have to deal with connectivity].

JoeyG1973, 2022-06-17:
I think it also might be nice to have an alternate menu for dynamically creating  profiles.  For example select O1 steel, select the process, and if that process for that steel requires additional input, typically the thickness of the material, the user would input that data point and it would dynamically adjust that time and heat.

ex: Austenizing for O1 tool steel needs to have the soak time adjusted based on the thickness of the material.

```
Austenitizing (High Heat): Heat slowly from the preheat to 1475-1500°F (802-816C)

Soak for 30 minutes for the first inch (25.4 mm) of thickness, plus 15 minutes for each additional inch (25.4 mm). 
```

rondoc, 2022-06-17:
I will save run from next firing and send log file.
My kiln is fairly large 8 cuft. My heating is by a single globar that is
near to the TC. This means in a cold kiln when the pi app starts  it puts
the globar on 100%. The TC then heats up very quickly showing a large
overshoot but I am sure the average kiln temp is not that temp. So in real
world I don't have a problem . Doing the manual low power until over about
400F is fine with me. I sometimes do this overnight and I trust my manual
setup more than the pi to do this. (I've been doing it for over 30 years.)
I  still have the pi running so I can monitor the temp which is really
nice.
Thanks Jason for all of this.
Ronald.

JoeyG1973, 2022-06-17:
So a different issue now... and I am almost positive it is because i have a cheap thermo sensor that won't go above this temp because if you look at the graph it had zero issue keeping up until that point.  Folks, am I right in this assessment of the issue?

[image]

```
Jun 17 18:08:36 raspberrypi python[637]: 2022-06-17 18:08:36,911 INFO oven: kiln must catch up, too cold, shifting schedule
Jun 17 18:08:36 raspberrypi python[637]: 2022-06-17 18:08:36,912 INFO oven: kiln outside pid control window, max heating
Jun 17 18:08:38 raspberrypi python[637]: 2022-06-17 18:08:38,915 INFO oven: temp=1733.88, target=2007.76, error=273.88, pid=0.00, p=968.67, i=209.96, d=0.00, heat_on=2.00, heat_off=0.00, run_time=45394, total_time=52500, time_left=7105
Jun 17 18:08:38 raspberrypi python[637]: 2022-06-17 18:08:38,916 INFO oven: kiln must catch up, too cold, shifting schedule
Jun 17 18:08:38 raspberrypi python[637]: 2022-06-17 18:08:38,916 INFO oven: kiln outside pid control window, max heating
Jun 17 18:08:40 raspberrypi python[637]: 2022-06-17 18:08:40,919 INFO oven: temp=1733.81, target=2007.76, error=273.95, pid=0.00, p=968.91, i=209.96, d=0.00, heat_on=2.00, heat_off=0.00, run_time=45394, total_time=52500, time_left=7105
Jun 17 18:08:40 raspberrypi python[637]: 2022-06-17 18:08:40,920 INFO oven: kiln must catch up, too cold, shifting schedule
Jun 17 18:08:40 raspberrypi python[637]: 2022-06-17 18:08:40,921 INFO oven: kiln outside pid control window, max heating
Jun 17 18:08:42 raspberrypi python[637]: 2022-06-17 18:08:42,924 INFO oven: temp=1733.91, target=2007.76, error=273.85, pid=0.00, p=968.58, i=209.96, d=0.00, heat_on=2.00, heat_off=0.00, run_time=45394, total_time=52500, time_left=7105
Jun 17 18:08:42 raspberrypi python[637]: 2022-06-17 18:08:42,924 INFO oven: kiln must catch up, too cold, shifting schedule
Jun 17 18:08:42 raspberrypi python[637]: 2022-06-17 18:08:42,925 INFO oven: kiln outside pid control window, max heating
Jun 17 18:08:44 raspberrypi python[637]: 2022-06-17 18:08:44,928 INFO oven: temp=1734.14, target=2007.76, error=273.62, pid=0.00, p=967.75, i=209.96, d=0.00, heat_on=2.00, heat_off=0.00, run_time=45394, total_time=52500, time_left=7105
Jun 17 18:08:44 raspberrypi python[637]: 2022-06-17 18:08:44,929 INFO oven: kiln must catch up, too cold, shifting schedule
Jun 17 18:08:44 raspberrypi python[637]: 2022-06-17 18:08:44,930 INFO oven: kiln outside pid control window, max heating
Jun 17 18:08:46 raspberrypi python[637]: 2022-06-17 18:08:46,932 INFO oven: temp=1734.47, target=2007.76, error=273.29, pid=0.00, p=966.59, i=209.96, d=0.00, heat_on=2.00, heat_off=0.00, run_time=45394, total_time=52500, time_left=7105
Jun 17 18:08:46 raspberrypi python[637]: 2022-06-17 18:08:46,933 INFO oven: kiln must catch up, too cold, shifting schedule
Jun 17 18:08:46 raspberrypi python[637]: 2022-06-17 18:08:46,934 INFO oven: kiln outside pid control window, max heating
```

rondoc, 2022-06-17:
Use a K type TC and get a Fluke temp reader. Plenty on ebay. You can
connect the Tc to both the Fluke and the pi via the max card. Make sure no
part of the TC is grounded or max chip won't work. The Fluke will read out
temp accurately as a check point for you .
Ronald.

JoeyG1973, 2022-06-17:
> Use a K type TC and get a Fluke temp reader. Plenty on ebay. You can connect the Tc to both the Fluke and the pi via the max card. Make sure no part of the TC is grounded or max chip won't work. The Fluke will read out temp accurately as a check point for you . Ronald.
> […](#)

yeah i got a k type from ebay.

JoeyG1973, 2022-06-17:
Thoughts on this from McMaster Carr?

https://www.mcmaster.com/thermocouples/thermocouple-type~k/maximum-temperature~2300-f/

JoeyG1973, 2022-06-17:
This is the log file and you can see it almost gets to 2000f but the TC died at that temp.

[crap.log](https://github.com/jbruce12000/kiln-controller/files/8931240/crap.log)

JoeyG1973, 2022-06-17:
Well I am pretty sure I killed the kiln.

[image]

rondoc, 2022-06-17:
Off topic... you can buy Kanthal wire , wind coils and replace. Not
difficult, or costly, but a bit time consuming.
Kanthal wire is good to cone10. 2400F.  The gage of wire to buy and length
or weight can be calculated or you might find a chart. You just need to
know the total power, watts , total length of winding and diameter of the
coils. It can get a bit more complex if you have several.pathways and/or
power level switching. If you need more help email me directly at
***@***.***  I am an EE but not a coder.

JoeyG1973, 2022-06-17:
I got the wire brand new a about 3 months ago from duralite which I just gave them the measurements and they make the wire custom.  They were the original providers for Norman Kilns when they were in business.  I have a Norman kiln.

I also looked at the temp probe as well...  It's galled on one side.

I think what happened is the probe failed and then the kiln went into full bore catchup and blasted right past 2400 and that's when the bricks failed and eventually the wire.

Now it looks like I need to replace some the bricks in the kiln because if you look carefully in the pictures, the brick has melted.

jbruce12000, 2022-06-18:
Everything is consistent with element failure. The kiln reaches a certain temp, and then starts to fall behind...even with the PID pushing 100% power to the elements.

jbruce12000, 2022-06-18:
I think this thread should be closed. Your initial tuning problem is solved.

closed 2022-06-18 by jbruce12000

JoeyG1973, 2022-06-18:
Agreed